earlephilhower/arduino-pico #247: Serial Monitor not following setup

LEBsci, 2021-07-12:
I was trying to use the SD card [listfiles](https://github.com/earlephilhower/arduino-pico/blob/master/libraries/SD/examples/listfiles/listfiles.ino) example (as is) on my Raspberry Pi Pico and I noticed I didn't get any message in the Serial Monitor. After some tests using this slightly modified Blink example:

```
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  Serial.begin(115200);
  Serial.println("test");
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
  Serial.println("ciclo");
}
```
 I noticed that the Serial Monitor was printing the string from the loop() cycle but not from the setup(). Furthermore, it doesn't matter if I don't specify the baud rate. I even deleted the _Serial.begin();_ line and I still get the print from the loop() at Baud rates from 9600 to 2000000. The LED also blinks but I don't get anything from the setup() in the Serial Monitor.

I'm using Arduino IDE 1.8.15 and the 1.9.2 version of this core for pico.

PontusO, 2021-07-12:
You probably only need to add a
`while(!Serial) 
    delay();`
directly after the Serial.begin statement.

LEBsci, 2021-07-12:
@PontusO I just did this and it works. Why is that line necessary? Thank you.

PontusO, 2021-07-12:
Great 👍. That line makes your program wait while the USB subsystem finishes arbitration between the PC and your Pico. It is the same for all arduino boards where the micro controller have the USB transceiver built in, so this is not a Pico specific thing.

PontusO, 2021-07-12:
Oh and worth of notice is that if you power your boards from an external power source and never connect a USB cable to your device it will hang waiting for a valid connection on that line.

closed 2021-07-12 by LEBsci