earlephilhower/arduino-pico #255: long computations result in hang.  watchdog?

manitou48, 2021-07-19:
I'm running IDE 1.9.2 with Arduino nano rp2040 connect board.  For benchmarks that take more than a few seconds of computation time, the sketches HANG.  I first noticed with CoreMark sketch
https://github.com/PaulStoffregen/CoreMark
I then modified a simple arithmetic benchmark, and if i increased the repetitions so the computation took more than 2 seconds, the sketch would hang?

is there a watchdog timeout or some IDE setting that I need ?

earlephilhower, 2021-07-19:
There's no WDT enabled in the IDE (although I do believe if you need one for a real app, the RP2040 has one in addition to all the other cool peripherals).

I've just run CoreMark and it did complete w/o incident while taking 14.63 seconds to complete:
````
CoreMark Performance Benchmark

CoreMark measures how quickly your processor can manage linked
lists, compute matrix multiply, and execute state machine code.

Iterations/Sec is the main benchmark result, higher numbers are better
Running.... (usually requires 12 to 20 seconds)

2K performance run parameters for coremark.
CoreMark Size    : 666
Total ticks      : 14627
Total time (secs): 14.63
Iterations/Sec   : 136.73
Iterations       : 2000
Compiler version : GCC10.3.0
Compiler flags   : (flags unknown)
Memory location  : STACK
seedcrc          : 0xE9F5
[0]crclist       : 0xE714
[0]crcmatrix     : 0x1FD7
[0]crcstate      : 0x8E3A
[0]crcfinal      : 0x4983
Correct operation validated. See README.md for run and reporting rules.
CoreMark 1.0 : 136.73 / GCC10.3.0 (flags unknown) / STACK
````

What exactly did you change?  One thing I see (and I have not dug into the code yet) is it says it's using the stack in some way.  We have a 4K stack in scratchpad RAM, so if you've got some kind of very deep recursion that increases as you increase whatever count you're changing, you can overflow the stack and become very unhappy....

earlephilhower, 2021-07-19:
Note, this was on a standard Rpi Pico board not the Arduino-brand Nano Connect RP2040.  The Arduino board itself might have some form of onboard, external WDT.  I don't think I've seen a schematic for it so can't comment.

I also overclocked to 250MHz and it automatically ran 4000 iterations giving the following results:
````
CoreMark Performance Benchmark

CoreMark measures how quickly your processor can manage linked
lists, compute matrix multiply, and execute state machine code.

Iterations/Sec is the main benchmark result, higher numbers are better
Running.... (usually requires 12 to 20 seconds)

2K performance run parameters for coremark.
CoreMark Size    : 666
Total ticks      : 14551
Total time (secs): 14.55
Iterations/Sec   : 274.90
Iterations       : 4000
Compiler version : GCC10.3.0
Compiler flags   : (flags unknown)
Memory location  : STACK
seedcrc          : 0xE9F5
[0]crclist       : 0xE714
[0]crcmatrix     : 0x1FD7
[0]crcstate      : 0x8E3A
[0]crcfinal      : 0x65C5
Correct operation validated. See README.md for run and reporting rules.
CoreMark 1.0 : 274.90 / GCC10.3.0 (flags unknown) / STACK
````

manitou48, 2021-07-19:
i get the same hangs on standard pico board -- actually just adding delay(5000) in computational benchmark loop causes hang.  delay(1000) is OK.

what do you get for CoreMark  at 125 MHz and -O3  (platform.txt)?

i must have something weird in all the arduino packages i have on my linux system

earlephilhower, 2021-07-19:
Here's the change:
````
earle@server:~/Arduino/hardware/pico/rp2040$ git diff
diff --git a/platform.txt b/platform.txt
index 27da167..34071e3 100644
--- a/platform.txt
+++ b/platform.txt
@@ -40,7 +40,7 @@ compiler.warning_flags.all=-Wall -Wextra -Werror=return-type -Wno-ignored-qualif
 
 compiler.defines={build.led} {build.usbstack_flags} -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSB_VID={build.vid} -DUSB_PID={build.pid} '-DUSB_MANUFACTURER={build.usb_manufacturer}' '-DUSB_PRODUCT={build.usb_product}'
 compiler.includes="-iprefix{runtime.platform.path}/" "@{runtime.platform.path}/lib/platform_inc.txt"
-compiler.flags=-Os -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions
+compiler.flags=-O3 -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions
 compiler.wrap="@{runtime.platform.path}/lib/platform_wrap.txt"
 compiler.libpico="{runtime.platform.path}/lib/libpico.a"
 compiler.libstdcpp="{runtime.platform.path}/lib/libstdc++.a"
````

And the output at 125MHz is fine (no changes to `git clone` of CoreMark):
````
CoreMark Performance Benchmark

CoreMark measures how quickly your processor can manage linked
lists, compute matrix multiply, and execute state machine code.

Iterations/Sec is the main benchmark result, higher numbers are better
Running.... (usually requires 12 to 20 seconds)

2K performance run parameters for coremark.
CoreMark Size    : 666
Total ticks      : 13166
Total time (secs): 13.17
Iterations/Sec   : 227.86
Iterations       : 3000
Compiler version : GCC10.3.0
Compiler flags   : (flags unknown)
Memory location  : STACK
seedcrc          : 0xE9F5
[0]crclist       : 0xE714
[0]crcmatrix     : 0x1FD7
[0]crcstate      : 0x8E3A
[0]crcfinal      : 0xCC42
Correct operation validated. See README.md for run and reporting rules.
CoreMark 1.0 : 227.86 / GCC10.3.0 (flags unknown) / STACK
````

Possibly you're seeing power issues that a different cable might solve?  I've had USB cables which were flaky (high gauge wire, high resistance)...things "work" until the processor needs high current and then they end up dropping so much voltage along the cable that it causes a brownout on the CPU.

manitou48, 2021-07-19:
If i plug the pico into another linux box, the sketch runs!  with -O3 i get 
CoreMark 1.0 : 226.65 / GCC10.3.0

so my linux boxes can get confused after lots of builds and runs with boards that swap USB between disk drive and serial port.  sigh.

Not your problem.  closing issue.  thanks for  your patience

closed 2021-07-19 by manitou48