earlephilhower/arduino-pico #266: Adafruit ILI9341

aroesz98, 2021-07-28:
Hi. I have one question. Why is your core slower about 2 times with Adafruit ILI9341 library then original arduino mbed core?
I'm using Bodmer optimized adafruit gfx and ili9341.
Your core benchmark:
Benchmark                Time (microseconds)
Screen fill              286588
Text                     57203
Lines                    187459
Horiz/Vert Lines         29527
Rectangles (outline)     17703
Rectangles (filled)      595953
Circles (filled)         92472
Circles (outline)        89547
Triangles (outline)      44872
Triangles (filled)       210015
Rounded rects (outline)  37068
Rounded rects (filled)   599763

And arduino mbed:
Benchmark                Time (microseconds)
Screen fill              113912
Text                     14038
Lines                    102271
Horiz/Vert Lines         10262
Rectangles (outline)     7121
Rectangles (filled)      236671
Circles (filled)         46604
Circles (outline)        43742
Triangles (outline)      22797
Triangles (filled)       100453
Rounded rects (outline)  18788
Rounded rects (filled)   244535

earlephilhower, 2021-07-29:
First of all, I'd check the upstream Adafruit library.  They've submitted quite a few updates to this repo and may have added optimized support for the ARDUINO_ARCH_RP2040 already, superseding Bodmer's original updates.  There was a new I2C call I remember was added that let them get significantly more performance from their library.

That said, it comes down to how the I2C control is handled.  I believe MBED does an async mode which lets things go faster, but which needs to twiddle bits in the HW directly to go (their library would do this, not your code).  For this core, I really tried to stick w/the plain SDK whenever possible for simplicity and to leverage any fixes they do (it's a 1-man show here!).  The SDK itself uses synchronous operations so there may be more dead time between transfers.  A logic analyzer capture would be handy to check.

I'm always happy to get PRs, so if you find a way to make I2C faster, please do throw a PR my way!

closed 2021-07-29 by earlephilhower

aroesz98, 2021-07-30:
Ok thanks. I tried on different clocks to run this benchmark but only 125MHz gives me good results (similar to arduino mbed). All over or under 125MHz SPI bus stuck on low frequency. Is there any solution for that? I need to use 200MHz clock to fast JPEG decoding. Regards :)

earlephilhower, 2021-07-30:
Very odd about that!  I use the plain SDK options to control I2C and they *should* take care of any peripheral clock issues when over/underclocking.  Do you have a scope capture that maybe we could look at and potentially post as an issue on the pico-sdk repo?

earlephilhower, 2021-07-30:
Actually, if you have scope traces of this core and the MBED one on the same I2C command (so they can be compared) that would be handy, too.  Let us check if the delta is dead time, clock period, etc.

aroesz98, 2021-07-30:
Unfortunatelyi haven't scope to check that. But there is other problem. When i'm using SPI on core 0 and i2c on core 1 then spi clock speed is slowing down. I can share you my project files and modified libraries to check them if you want. Maybe tomorrow i will have access to the oscilloscope.

earlephilhower, 2021-07-30:
The whole project may be hard to actually run for anyone but yourself.  If it expects any kind of an answer from connected devices (i.e and I2C read or write needs the slave to ack) it's not going to actually do the same time on a bare board.

If you can narrow down the case (i.e. say, if you do a 100khz I2C operation on core0 the supposed 1MHz SPI operation from core1 actually runs at 100khz, too) that would be very helpful here and upstream.

aroesz98, 2021-07-30:
I checked my program again but there's very strange. At first i compiled at 125MHz clocks and display runs fast but i2c doesn't work idk why. But when i compiled at 200MHz display runs slower but i2c works fine. At this point i'm confused. I don't know whats going on with that.

aroesz98, 2021-07-30:
Ok. That's my bad. I begin oled display in setup and setup1 so because of that something goes wrong. Now ili9341 and ss1306 are running well. But problem with spi clocks on overclocked cpu still exist. All setting over / under 125MHz are slowing down spi bus.

earlephilhower, 2021-07-30:
Just FYI, all clocks are going to approximate because there's not a unique clock generator for every peripheral.

So if you're w/in +/- 5% of desired clock rate then I would not sweat it and that's just what the Pico can support.  If you're asking for 1MHz and getting 50kHz, though, that's a real problem!

aroesz98, 2021-07-31:
So when i choose overclocked i.e. 150MHz then spi drivider doesn't stay same then 125MHz version? SPI doesn't get overclocked  when cpu clocks are higher? is there any way to "hard-choose" clocks for spi?

earlephilhower, 2021-07-31:
SPI takes a clock frequency as part of the `beginTransaction(SPISettings s)` call.  So, use that to specify the clock you want.  The Pico SDK will figure out the proper clock dividers/etc. to make it happen.

You can look at the RP2040 datasheet for more detailed info on clocking.

aroesz98, 2021-07-31:
Ok thanks for help. Regards