earlephilhower/arduino-pico #278: Serial1 - Board doesn't start

GregorR1, 2021-08-12:
Hi, I've been having some issues getting the Serial1 to work on my Pi Pico.

I had it working a while ago using this Board manager, but recently tried to upload a previously working sketch and my pico does not appear to be running the sketch.  The board does not connect to my PC and I have to use the BOOTSEL button to flash a different sketch to get the pico to reconnect.

I had a feeling this was due to the Serial1 as other sketches without this were working fine, so I did a quick test.

The following code works - the LED flashes and the board writes to the Serial monitor:

```
void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(9600);
  //Serial1.setRX(2);Serial1.setTX(1);Serial1.begin(57600);
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
Serial.println("hello");
//Serial1.println("hello1");
}
```

The following code fails - the LED does not flash and the pico will not connect to the PC.

```
void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(9600);
  Serial1.setRX(2);Serial1.setTX(1);Serial1.begin(57600);
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
Serial.println("hello");
Serial1.println("hello1");
}
```

I'm on V1.9.3 of the board manager.

I have also tried V1.9.0.

I will update this post as I try different versions.

Thanks.

earlephilhower, 2021-08-12:
The issue looks like one of incompatible pins.  The Serial1 (and all peripherals) have limited IOs they can connect to.  The `Serial1.setRX(2)` is failing and `panic()`ing with a message that might not come out, depending on how the machine was set up prior.  Check the RPI datasheet for the exact supported UART1 pins.

GregorR1, 2021-08-12:
Hi Earl,

Thanks for the quick response.  According to the pin diagram, these are the default pins for UART0 which is the UART connection referenced by Serial1.  
[image]

Could it be expecting the GPIO pin number instead of the physical pin number?  However, this doesn't explain why it worked previously.

earlephilhower, 2021-08-12:
It always wants the GPIO number.

So, GPIO4 and GPIO5 for Serial1 are OK;.   .setRX(5), .setTX(4)

GregorR1, 2021-08-12:
That seems to have been the issue.  Thanks for your help.

Still not sure how it worked before as I hadn't edited the sketches!  But at least it's working again now.

Thanks :)

earlephilhower, 2021-08-12:
Prior versions of the core would not panic on an invalid pinout choice.  So, the invalid assignments were no-ops.

closed 2021-08-12 by earlephilhower

remy, 2021-09-15:
I've run into the same problem as described in the subject, but I'm trying to use GPIO 26 for the TX. Am I right in thinking this is invalid and I can't just pick an arbitrary IO pin for serial and it *must* be one of the UART TX pins listed in the pinout?

earlephilhower, 2021-09-15:
Yes, they must be a port that the Pico is physically capable of connecting to the UART.  There are GPIO mux limitations that can't be worked around w/o using a full software-generated serial port.