earlephilhower/arduino-pico #289: Is pulseIn working?

carlesoriol, 2021-08-24:
3 times same while loop in pulse in. Is it ok?

extern "C" unsigned long pulseIn(uint8_t pin, uint8_t state, unsigned long timeout) {
    uint64_t start = time_us_64();
    uint64_t abort = start + timeout;
...
    // Wait for deassert, if needed
    while ((!!gpio_get(pin) != !state) && (time_us_64() < abort));
...
    // Wait for assert
    while ((!!gpio_get(pin) != !state) && (time_us_64() < abort));
...
    // Wait for deassert
    while ((!!gpio_get(pin) != !state) && (time_us_64() < abort));

slva21, 2021-08-29:
Ahh. Ultrasonic sensor wasn't working. Must the be reason 🙄

carlesoriol, 2021-08-30:
?

Missatge de Adebayo Popoola ***@***.***> del dia dg., 29
d’ag. 2021 a les 12:57:

> Ahh. Ultrasonic sensor wasn't working. Must the be reason 🙄
>
> or Android

PR #293 (2021-08-30): Fix pulseIn detection
Fixes #289
commit a8121f31 (2021-08-30): Fix pulseIn detection (#293)

Fixes #289
closed 2021-08-30 by earlephilhower