earlephilhower/arduino-pico #304: which multithreading environment is supported?

dsyleixa, 2021-09-10:
hello,
which multithreading environment is supported? 
Perhaps sth like for the ESP32 (RTOS, pthread, std::thread)?
#include <thread>
#include <freertos/task.h>
?

earlephilhower, 2021-09-10:
We don't use FreeRTOS here (it doesn't natively support multicore AFAIK).  There is the `setup1() ` `loop1()` option to run both cores in parallel on two different tasks, or you can look at #292 which is in progress.

Good luck!

closed 2021-09-10 by earlephilhower

dsyleixa, 2021-09-10:
thank you, but I don't find a documentation or code examples for multiple threads, neither in your link.

just to say,
I didn't claim that your lib might or would do it via RTOS.  
The mentioned  ESP MT libs  were just an example how they utilize pthread/std:thread. 
As the ESP32 is a double-core,  MT is also supported ffor both cores by pthread/std:thread, and of course the 2040 may have different pthread/std:thread immplementations.

dsyleixa, 2021-09-10:
thank you, but I don't find a documentation or code examples for multiple threads, neither in your link.

just to say,
I didn't claim that your lib might or would do it via RTOS.  
The mentioned  ESP MT libs  were just an example how they utilize pthread/std:thread. 
As the ESP32 is a double-core,  MT is also supported for both cores by pthread/std:thread, and of course the 2040 may have different pthread/std:thread implementations.

dsyleixa, 2021-09-11:
thank you, but I don't find a documentation or code examples for multiple threads, neither in your link.

just to say,
I didn't claim that your lib might or would do it via RTOS.  
The mentioned  ESP MT libs  were just an example how they utilize pthread/std::thread. 
As the ESP32 is a dual-core,  MT by RTOS is also supported for both cores via pthread/std::thread, and of course the 2040 may have different pthread/std:thread implementations (the main thing is that pre-emptive multithreading works with them).

(edited)