earlephilhower/arduino-pico #331: Why I can't set SPI1.

trkkesen26, 2021-09-30:
When I try to set SPI1 by using SPI.set... or SPI1.set... , I can't use it. Is there any reason that I did wrong.

trkkesen26, 2021-09-30:
`   SPI1.setRX(12);
    SPI1.setTX(11);
    SPI1.setSCK(10);
    SPI1.setCS(13);
    Serial.begin(9600);
    delay(5000);     //ı used like this and nothing happaned. 

    // Also I tried with without "1" I mean like SPI.setRX(12). After that the serial is locked `

earlephilhower, 2021-10-15:
There's something else going on in your code, I think, because I am able to set those pins w/o incident (on a generic RPi 2060 Pico board):
````
#include <SPI.h>

void setup() {
Serial.begin(115200);
delay(2000); Serial.printf("rx\n");
SPI1.setRX(12);
delay(2000); Serial.printf("tx\n");
SPI1.setTX(11);
delay(2000); Serial.printf("sck\n");
SPI1.setSCK(10);
delay(2000); Serial.printf("cs\n");
SPI1.setCS(13);
Serial.printf("done\n");
}

void loop() {
}
````

Gives output
````
rx
tx
sck
cs
done
````

Are you on a board, maybe, that doesn't support SPI1 or have it pinned out, maybe?

trkkesen26, 2021-11-29:
Sorry for that. You were right. Serial is blocked because of my bad managing serials.

closed 2021-11-29 by trkkesen26