earlephilhower/arduino-pico #333: Boot stage 2 option

RandomHacks-Git, 2021-10-08:
Hi, I made a board with the RP2040 and I'm trying to use the flash chip BY25Q16BS ([datasheet](https://datasheet.lcsc.com/lcsc/1912111437_BOYAMICRO-BY25Q16BSTIG-T_C382744.pdf)) can someone tell me which boot stage 2 option would be more adequate?

I'm having an issue where sometimes after power cycling a few times all the data stored with the eeprom libray is lost and I'm not sure if it is related to this setting.

RandomHacks-Git, 2021-10-09:
I think I fixed my issue now, in my code if there is a change on an analog pin I write to "eeprom", when turning of the microcontroller the capacitor I had on the analog pin was discharging slowly (slower than the vcc on rp2040) and therefore a change in the analog value was detected which triggered a write to "eeprom" and because there wasn't enough power to finish the write cycle it ended up corrupting the data.

All the boot stage 2 options seem to work with this flash except for "IS25LP080 QSPI/2" which fails to boot. Do you know which of the options is the fastest?

Another question I have, does EEPROM.put( ) (or EEPROM.commit( ) ??) read the data first to check if it is different than what is already written?

earlephilhower, 2021-10-11:
> All the boot stage 2 options seem to work with this flash except for "IS25LP080 QSPI/2" which fails to boot. Do you know which of the options is the fastest?

Any of the `QSPI /2` options will be about the same speed and fastest.  The clock rate and 4-bits really set it, not the specific flash quirks.
 
> Another question I have, does EEPROM.put( ) (or EEPROM.commit( ) ??) read the data first to check if it is different than what is already written?

Use the source, Luke. :)  https://github.com/earlephilhower/arduino-pico/blob/master/libraries/EEPROM/EEPROM.cpp

Yes, both are checked and if it's all the same, nothing is erased or written to flash.