earlephilhower/arduino-pico #341: Serial2 ports definition

0ba, 2021-11-01:
Hello, 
I have a custom RP2040 based design which defines Serial2 ports as follow :
#define PIN_SERIAL2_TX (4u)
#define PIN_SERIAL2_RX (5u)
instead of 
#define PIN_SERIAL2_TX (8u)
#define PIN_SERIAL2_RX (9u)
as defined by default in https://github.com/earlephilhower/arduino-pico/blob/master/variants/generic/pins_arduino.h

Is there another solution rather than locally overriding this?
Something like another board variant/option?

Thx for the good work!

earlephilhower, 2021-11-01:
One way would be to just create a new `SerialUART` in your `.ino` and use that instead of `Serial2`, like
````
SerialUART SerialPort(uart1, 8, 9);

void setuo() {
    SerialPort.begin(115200);
    ....
}
````

Otherwise, you could create a PR that uses an `ifdef` to create using the standard or overridden pins.  You can add `-DSERIAL2_OVERRIDE_RX=9` to your `platform.local` file, etc.

In the `SerialUART.cpp` you'd just do something like
````
#ifdef SERIAL2_OVERRIDE_RX
SerialUART Serial2(uart1, SERIAL2_OVERRIDE_TX, SERIAL2_OVERRIDE_RX);
#else
SerialUART Serial2(uart1, PIN_SERIAL2_TX, PIN_SERIAL2_RX);
#endif
````

0ba, 2021-11-02:
Thank you Earle for this quick answer!
Solution #1 works great.

O.

closed 2021-11-02 by 0ba