earlephilhower/arduino-pico #345: Low Power Modes

CyberCityCircuits, 2021-11-08:
Do you have any plans on implementing the native low power modes that the RP2040?  This would be deeply appreciated.

earlephilhower, 2021-11-10:
In the short term I'm unfortunately occupied with other things, but to access the `pico-extra` sleep functions should be possible with some simple modifications to the core.  (This is also for #344)

Basically, you will need to rebuild libpico.a to include the `pico_sleep` directory and add the proper include path to `platform_inc` to let the IDE fine the proper headers.

1. Add the pico_sleep line to the Cmakefile.txt:   https://github.com/earlephilhower/arduino-pico/blob/master/tools/libpico/CMakeLists.txt
2. Rebuild libpico.a by running make-libpico.sh  https://github.com/earlephilhower/arduino-pico/blob/master/tools/libpico/make-libpico.sh
3. Add the pico-extras/pico_sleep path to pl;atform_inc.txt:  https://github.com/earlephilhower/arduino-pico/blob/master/lib/platform_inc.txt  (the format is a little off but should be obvious)

Restarting the IDE should then give the ability to `#include <pico/sleep.h>` and call the functions exported there.  Note that I have not looked deeply at this and it is in the `extras` repo so it may not be 100% working from the RPi folks,yet.  I have a feeling you'll need to do things like save/restore all active clocks, set which clocks to disable in sleep mode, and make sure I/Os are set to appropriate drive strengths/inputs.

WildkatzGithub, 2021-12-02:
Hello, I tried to running make-libpico.sh with GitBash, but don't work.

$ ./make-libpico.sh
./make-libpico.sh: line 6: cd: ../../system/arm-none-eabi/bin: No such file or directory
./make-libpico.sh: line 11: cmake: command not found

earlephilhower, 2021-12-02:
You'll need to clone the git repo, run the get.py script to fetch the toolchain (your 1st missing exe), and also of course have cmake installed(the 2nd one).  It uses the pico-sdk build process, so all the pico-sdk prerequisites are needed.

For cmake, your distro's package manager should have it (`apt install cmake` or `yum install cmake` as the case may be...)

WildkatzGithub, 2021-12-02:
Thx for help! I have cloned the git repo, ran get.py and installed cmake.
Success & cmake was found by GitBash, but now it hangs around line 12:

./make-libpico.sh: line 12: make: command not found

I can't install the apt or yum package manager as I'm currently using only Windows 10.

Would it be better to use Linux for this?

earlephilhower, 2021-12-02:
> Would it be better to use Linux for this?

Short answer is, "Yes."  Long answer is, "Yeeeeeees." :)  WSL will work fine too.

Alternatively, you can see how they build the Pico SDK under Windows.  There is a readme in their project, but I never looked at it since I didn't have the need.  The build prerequisites they need are what the `makelibpico.sh` script requires, too.

WildkatzGithub, 2021-12-04:
Thank you for showing me these possibilities :)
At https://github.com/raspberrypi/pico-playground I read that the use of pico_sleep is not yet stable.

I took current measurements on the Pico yesterday.
The Pico needs 15 mA in a simple while loop at 125 MHz and only one core. Without external components and peripherals, of course.
If I now set the system clock to 10 MHz, it only needs 1 to 2 mA in the same loop. That is sufficient for my application.

With this I avoid the use of pico_sleep and the known problems like "frozen pico" or "printf () / puts () - not possible" after sleep / dormant.

alfonsoletizia1, 2021-12-09:
@WildkatzGithub how did you succeded in setting the clock to 10 MHz? I tried using platform io, but then it's unusable

WildkatzGithub, 2021-12-18:
Hello, you can set the clock to 10 MHz with this code in the Arduino IDE:

set_sys_clock_khz(10000, true);    // Set System clock to 10000 kHz

I've already tried smaller clocks, but 10 MHz was the lowest clock speed so far without the Pico getting stuck.
Please let me know if you can find a smaller clock.

By the way, if the clock is currently set to 10 MHz, the Pico can only be programmed using the button at this point.

sail81, 2022-01-23:
@WildkatzGithub thank you so much for your 10 MHz power saving hint. A great idea indeed. In my application I do some outdoor measurement every minute and send it via LoRa. In between I have 59s delay. The pico is driven by 3 rechargeable AA NiMH batteries. After I saw your post I created this function:

void power_save(uint16_t psinterval){
    LoRa.sleep();
      delay(2);
    set_sys_clock_khz(10000, false); // Set System clock to 10 MHz, sys_clock_khz(10000, true); did not work for me
      delay(2);
   vreg_set_voltage(VREG_VOLTAGE_0_95); // 0.85V did not work, 0.95V seems pretty stable
      delay(psinterval);
    vreg_set_voltage(VREG_VOLTAGE_DEFAULT); // corresponds to 1.10V, not sure if that is really required for 48 MHz 
      delay(2);
    set_sys_clock_48mhz(); // Set System clock back to 48 MHz to make LoRa work
      delay(2);
    LoRa.idle();
}

Now I am testing it since 3 days and it runs stably. I can already tell that the battery voltage drops much slower that before. I should have measured the power consumption before. I will do that once my test is completed. One could play with the voltages, too. Though, I have not yet an idea how much power saving would benefit from that. Let's see.

When checking lower clocks than 10 Mhz did you run more that a delay()?

earlephilhower, 2022-02-25:
Looks like this is closed.  Unfortunately the Pico SDK sleep modes are unstable per above, but setting a low system clock seems to help at the cost of peripheral(USB) issues.

closed 2022-02-25 by earlephilhower

JudgeBeeb, 2022-03-15:
Has anyone managed to get sleep_pico working in the Arduino IDE yet? whenever I try to build libpico.a I get the following errors:`ar: stdio.c.obj: not found in archive
ar: stdio_usb.c.obj: not found in archive
ar: stdio_usb_descriptors.c.obj: not found in archive`

earlephilhower, 2022-10-20:
On reboot the rom and app will set the core to 133mhz until your app changes it again.  So you could set the frequency in your setup() if desired and only a short period would be at the higher frequency.

Phoenix1747, 2022-10-23:
> Looks like this is closed. Unfortunately the Pico SDK sleep modes are unstable per above, but setting a low system clock seems to help at the cost of peripheral(USB) issues.

Is this still the case today? Are there any alternatives to using the native Pico SDK sleep functions that you know of?
Like for example the Adafruit SleepyDog lib: https://github.com/adafruit/Adafruit_SleepyDog

Phoenix1747, 2022-11-17:
@ninjampa The [Adafruit SleepyDog Arduino Library](https://github.com/adafruit/Adafruit_SleepyDog) now also supports the RP2040 architecture. You can try it if that's what you're looking for.

Phoenix1747, 2022-11-18:
Alright, I see no worries!

metrafonic, 2023-02-16:
@ninjampa did you find any solution for getting pico-extras sleep.h functionality or similar?

earlephilhower, 2023-02-21:
Did you recompile the libpico.a using the `make-libpico.sh` script before building?  The addition of the pico-sleep target in the CMakefile should cause it to include the new source.  If that did not work properly, then I'd suggest just cut-n-pasting the function in question to your own code and working from there.

Dj-EKI, 2023-02-22:
I ran make-libpico.sh. But what do you mean by building? Should I edit another CMake file? If so, which ones and what do I have to do? As I said, I only did the 3 steps you described. Since it didn't work, I finally ran make-libpico.sh again.

Dj-EKI, 2023-02-24:
I had programmed my Pico with Micropython (Python). But this script language was too slow for me, so I'm trying Arduino now. Deepsleep worked with Python. But I also used Python extras. I implemented it myself in the rp2 port.

I copied xosc.c, rosc.c, sleep.c to rp2 folder.
I copied hardware/rosc.h, hardware/xosc.h, pico/sleep.h to rp2 folder.
Modified modmachine.c.

```
#include "pico/sleep.h"
#include "hardware/regs/io_bank0.h"
#include "hardware/xosc.h"
#include "hardware/rosc.h"

static dormant_source_t _dormant_source;

#if MICROPY_PY_MACHINE

#define RP2_RESET_PWRON (1)
#define RP2_RESET_WDT (3)

static void _go_dormant(void) {
    assert(dormant_source_valid(_dormant_source));

    if (_dormant_source == DORMANT_SOURCE_XOSC) {
        xosc_dormant();
    } else {
        // WARNING: This stops the rosc until woken up by an irq
        rosc_write(&rosc_hw->dormant, ROSC_DORMANT_VALUE_DORMANT);
        // Wait for it to become stable once woken up
        while(!(rosc_hw->status & ROSC_STATUS_STABLE_BITS));
    }
}

....
....
....

STATIC mp_obj_t machine_deepsleep(size_t n_args, const mp_obj_t *args) {
    
    bool edge = mp_obj_get_int(args[1]);
    bool high = mp_obj_get_int(args[2]);
    uint gpio_pin = mp_obj_get_int(args[0]);
    
    bool low = !high;
    bool level = !edge;

    // Configure the appropriate IRQ at IO bank 0
    assert(gpio_pin < NUM_BANK0_GPIOS);

    uint32_t event = 0;

    if (level && low) event = IO_BANK0_DORMANT_WAKE_INTE0_GPIO0_LEVEL_LOW_BITS;
    if (level && high) event = IO_BANK0_DORMANT_WAKE_INTE0_GPIO0_LEVEL_HIGH_BITS;
    if (edge && high) event = IO_BANK0_DORMANT_WAKE_INTE0_GPIO0_EDGE_HIGH_BITS;
    if (edge && low) event = IO_BANK0_DORMANT_WAKE_INTE0_GPIO0_EDGE_LOW_BITS;

    gpio_set_dormant_irq_enabled(gpio_pin, event, true);

    _go_dormant();
    // Execution stops here until woken up

    // Clear the irq so we can go back to dormant mode again if we want
    gpio_acknowledge_irq(gpio_pin, event);
    //machine_lightsleep(n_args, args);
    //return machine_reset();
    return mp_const_none;
}
MP_DEFINE_CONST_FUN_OBJ_VAR_BETWEEN(machine_deepsleep_obj, 0, 3, machine_deepsleep);
```

I'm using this in Python. I would also like to have _write_bits and _read_bits in Arduino, but I don't know how to implement that.
```
# Clock
CLOCKS_BASE = 0x40008000
CLK_REF_CTRL = 0x30
CLK_SYS_CTRL = 0x3C
CLK_PERI_CTRL = 0x48
CLK_USB_CTRL = 0x54
CLK_ADC_CTRL = 0x60
CLK_RTC_CTRL = 0x6C
PLL_SYS_BASE = 0x40028000
PLL_USB_BASE = 0x4002C000
PLL_PWR = 0x4

ROSC_BASE = 0x40060000

@micropython.asm_thumb
def _read_bits(r0):
    ldr(r0, [r0, 0])

@micropython.asm_thumb
def _write_bits(r0, r1):
    str(r1, [r0, 0])

DORMANT_PIN = 21
p1 = Pin(DORMANT_PIN, Pin.IN)

_write_bits(CLOCKS_BASE + CLK_REF_CTRL, 0x02)#
_write_bits(CLOCKS_BASE + CLK_SYS_CTRL, 0x00)#
#_write_bits(CLOCKS_BASE + CLK_USB_CTRL, 0x00)
_write_bits(CLOCKS_BASE + CLK_ADC_CTRL, 0x00)#
_write_bits(CLOCKS_BASE + CLK_RTC_CTRL, 0x60)#
_write_bits(CLOCKS_BASE + CLK_PERI_CTRL, 0x00)#
#_write_bits(PLL_USB_BASE + PLL_PWR, 0x2D)
#_write_bits(ROSC_BASE, 0xD1EAA0)

deepsleep(DORMANT_PIN,1,0)

_write_bits(CLOCKS_BASE + CLK_REF_CTRL, 0x02)#
_write_bits(CLOCKS_BASE + CLK_SYS_CTRL, 0x01)#
#_write_bits(CLOCKS_BASE + CLK_USB_CTRL, 0x800)
_write_bits(CLOCKS_BASE + CLK_ADC_CTRL, 0x800)#
_write_bits(CLOCKS_BASE + CLK_RTC_CTRL, 0x800)#
_write_bits(CLOCKS_BASE + CLK_PERI_CTRL, 0x800)#
#_write_bits(PLL_USB_BASE + PLL_PWR, 0x04)
#write_bits(ROSC_BASE, 0xFFFAA0)
```

This script works wonderfully. The Pico is woken up by an Arduino Nano via pin 21.
If I use the same script in Modmachin.h then the pico goes to sleep but can't be woken up. Although that's the same script. I do not understand that.

ninjampa, 2023-02-27:
> @ninjampa did you find any solution for getting pico-extras sleep.h functionality or similar?

Hello @metrafonic,

Well, there is not a supported way of entering a real (regarding power saving) sleep mode. My application at the time was an embedded battery-solar solution (something around uA), so I had to drop the RP2040 in the design. 

I just noticed that @Phoenix1747 updated the [Adafruit SleepyDog Arduino Library](https://github.com/adafruit/Adafruit_SleepyDog) that had a delay function and now has a proper sleep_ms, but I didn't use it, can you test it and put the results in here?

If the RP is mandatory in low power mode in your project, I suggest looking for a solution [like this one](https://www.youtube.com/watch?v=05o5QOZkpwg&t=184s&ab_channel=LowLevelLearning).

ninjampa, 2023-02-27:
> I ran make-libpico.sh. But what do you mean by building? Should I edit another CMake file? If so, which ones and what do I have to do? As I said, I only did the 3 steps you described. Since it didn't work, I finally ran make-libpico.sh again.

@Dj-EKI, do you have the logs from make-libpico.sh? Back in the day I had the same problem, and the reason was because make-libpico.sh was not compiling as it should.

Phoenix1747, 2023-02-27:
@ninjampa Relating to the [Adafruit SleepyDog Arduino Library](https://github.com/adafruit/Adafruit_SleepyDog), the lib doesn't actually enable any deep sleep modes as you might have already read here:

https://github.com/earlephilhower/arduino-pico/issues/345#issuecomment-1320010455

That's good, but it doesn't shut down anything. It's still consuming a whole lot of power (relatively) in this mode.

ninjampa, 2023-02-27:
@Phoenix1747 Thank you, it was my own comment and I forgot about it.

Phoenix1747, 2023-02-27:
Oh yeah, true. Didn't even notice that 😄

Dj-EKI, 2023-02-27:
@ninjampa I don't know where the log files are, but I found them here. http://dj-eki.de/Dateien/CMakeFiles.7z

Dj-EKI, 2023-03-03:
I managed to implement pico-extras in arduino.
I didn't know that I also had to enter something in the cmake file.

1. Copy pico-extras into the hardware folder of rp2040.
%USERPROFILE%\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.0.0
2. Copy these commands into platform_inc.txt. (Arduino15\packages\rp2040\hardware\rp2040\3.0.0\lib\platform_inc.txt)
```
-iwithprefixbefore/pico-extras/src/rp2_common/pico_sleep/include
-iwithprefixbefore/pico-extras/src/rp2_common/hardware_rosc/include
```
3. Enter the following codes in CMakeLists.txt. (Arduino15\packages\rp2040\hardware\rp2040\3.0.0\tools\libpico\CMakeLists.txt)
Do not enter twice what is already available.
```
cmake_minimum_required(VERSION 3.12)

set(PICO_EXTRAS_PATH ../../../pico-extras)

include(pico_sdk_import.cmake)
include($ENV{PICO_EXTRAS_PATH}../../pico-extras/external/pico_extras_import.cmake)
```
in target_link_libraries:
```
hardware_xosc
hardware_rosc
hardware_sleep
pico_sleep
```
4. setup_default_uart() does not work with Arduino. That has something to do with "Serial". That's why it has to be commented out.
Arduino15\packages\rp2040\hardware\rp2040\3.0.0\pico-extras\src\rp2_common\pico_sleep\sleep.c
```
...
    // Assuming both xosc and rosc are running at the moment
    if (dormant_source == DORMANT_SOURCE_XOSC) {
        // Can disable rosc
        rosc_disable();
    } else {
        // Can disable xosc
        xosc_disable();
    }

    // Reconfigure uart with new clocks
    //setup_default_uart();
}
```

5. Run make-libpico.sh. Can also be done with Linux.  (Arduino15\packages\rp2040\hardware\rp2040\3.0.0\tools\libpico\make-libpico.sh)
6. Enter dormant mode with this command.
```
sleep_run_from_xosc();
sleep_goto_dormant_until_pin(21,1,0);
```
There are other commands too. See playground.
But sleep_run_from_xosc(); is very important to reduce the current.
After waking up you have to adjust the clock rate again.

SIeeepy, 2023-03-10:
> 5. Run make-libpico.sh. Can also be done with Linux.

I really want to use the rp2040 low power modes but I don't understand this step. Can someone explain how I can run the make-libpico.sh file on windows?

earlephilhower, 2023-03-10:
Does the Windows install of the Pico SDK still install a full MiinGW?  If so, then I think you can just run the bash script there.  Make sure you're using a full `git` clone of the tree, and use `python3 tools/get.py` to install the binaries so the build script can find them.

SIeeepy, 2023-03-11:
Thanks. I'm familiar with Arduino but I struggle with Raspi/Linux/Python subjects and terminology.

I've learned now that sh files are Linux related and also managed to install Git Bash and Python3 and to finally run the make-libpico.sh. But now I've reached the hurdle that already stopped @WildkatzGithub, the error message "./make-libpico.sh: line 12: make: command not found".  

@Dj-EKI's wording "Run make-libpico.sh. Can also be done with Linux" gave me hope that there was a way in windows but maybe I misunderstood.

Dj-EKI, 2023-03-11:
It is not easy to equip Windows with Linux components.
Install Chocolatey.
And use this command `choco install make`
The make command was not found.  it is an important component for Linux.  but you will be missing more, so I would recommend you to download Ubuntu from Windows Store.
In Ubuntu U can use this command to get all components.
```
sudo apt update
sudo apt install cmake gcc-arm-none-eabi libnewlib-arm-none-eabi build-essential libstdc++-arm-none-eabi-newlib
```

earlephilhower, 2023-03-11:
You can also use `MinGW` which is what I think the RPI team originally had folks do.  But running it all under WSL is probably easier at @Dj-EKI says.

SIeeepy, 2023-03-12:
Now it works!! The RP2040 is sleeping peacefully. Thanks @earlephilhower & @Dj-EKI !

If someone has the same issues:
1. Install a Linux distribution like Ubuntu via Microsoft Store (e.g. Ubuntu 22.04.02)
2. Open Ubuntu in admin mode and setup user name and a password
3. Run a command to install cmake, python3 and gcc-arm-none-eabi
>sudo apt update && sudo apt install gcc-arm-none-eabi cmake python3 -y
 
4. Guide Ubuntu to the libpico folder an run the make-libpico.sh file 
>cd /mnt/c/Users/[YOUR USERNAME]/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/3.0.0/tools/libpico && ./make-libpico.sh

(edit your directory and user name)

On a side note I'm still not getting near the low sleep power consumption of 180µA promised in the [RP2040 datasheet](https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf)  (p. 622). I measured 1.2mA during dormant sleep. It's 20x better then the aktive 23mA consumption before, but why is it still 7x higher than spec? Is this a hardware issue or could I switch off more software-wise? 

Hardware: I'm using the tiny [Waveshare RP2040 Zero](https://www.waveshare.com/wiki/RP2040-Zero). [The schematics](https://www.waveshare.com/w/upload/4/4c/RP2040_Zero.pdf) don't indicate power hungry components, the [rt9013-33 linear regulator](https://www.richtek.com/assets/product_file/RT9013/DS9013-10.pdf) seems fine.

Here's my test code, maybe I'm missing something (a button press wakes the rp2040 up for 1 sec)

```
#include <pico/stdlib.h>
#include <pico/sleep.h>
#include <hardware/rosc.h>
#include <SoftwareSerial.h>

SoftwareSerial mySerial(0, 1); // RX, TX

int Button_Pin = 8;
bool awake = false;

void setup()
{
  pinMode(Button_Pin, INPUT_PULLUP);
  mySerial.begin(9600);
  delay(1000);
  mySerial.println("Start...");
  delay(1000);
}

void loop()
{
  if (!awake)
  {
    sleep_run_from_xosc();
    sleep_goto_dormant_until_pin(Button_Pin, false, false); // Pin, Edge, HIGH
    awake = true;
  }

  else if (awake)
  {
    Get_Up();
    mySerial.println("I have things to do and I will never slee...");
    delay(1000);
    awake = false;
  }
}

void Get_Up()
{
  rosc_write(&rosc_hw->ctrl, ROSC_CTRL_ENABLE_BITS);
  clocks_init();
  set_sys_clock_khz(133000, true);
}
```

julianatgit, 2023-03-14:
Is there any way to check weather the "make-libpico.sh" bash ran successfully or not?

I followed the instructions (from [Dj-EKI](https://github.com/earlephilhower/arduino-pico/issues/345#issuecomment-1453916464) ) and the sleep.h is found, but the functions (sleep.c) are not.
With my very limited knowledge i think that something went wrong during the execution of the "make-libpico.sh" bash, but i have no idea how to figure that out.

Im using the Pico setup for Windows (https://github.com/raspberrypi/pico-setup-windows/releases) and executed everything in the "Pico - Developer Command Prompt".

> c:/users/WIN_USER/appdata/local/arduino15/packages/rp2040/tools/pqt-gcc/1.5.0-b-c7bab52/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld.exe: C:\Users\WIN_USER\AppData\Local\Temp\arduino\sketches\90E80519620805E9FF0E2CC4AAE29730\sketch\SleepTest.ino.cpp.o: in function `_Z18recover_from_sleepjjj':
C:\Users\WIN_USER\Desktop\gsm-car-control\ArduinoIDE\SleepTest/SleepTest.ino:56: undefined reference to `stdio_init_all'
c:/users/WIN_USER/appdata/local/arduino15/packages/rp2040/tools/pqt-gcc/1.5.0-b-c7bab52/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld.exe: C:\Users\WIN_USER\AppData\Local\Temp\arduino\sketches\90E80519620805E9FF0E2CC4AAE29730\sketch\SleepTest.ino.cpp.o: in function `sleep_run_from_xosc':
c:\users\WIN_USER\appdata\local\arduino15\packages\rp2040\hardware\rp2040\3.1.0\pico-extras\src\rp2_common\pico_sleep\include\pico/sleep.h:50: undefined reference to `sleep_run_from_dormant_source'
c:/users/WIN_USER/appdata/local/arduino15/packages/rp2040/tools/pqt-gcc/1.5.0-b-c7bab52/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld.exe: C:\Users\WIN_USER\AppData\Local\Temp\arduino\sketches\90E80519620805E9FF0E2CC4AAE29730\sketch\SleepTest.ino.cpp.o: in function `rtc_sleep':
C:\Users\WIN_USER\Desktop\gsm-car-control\ArduinoIDE\SleepTest/SleepTest.ino:40: undefined reference to `sleep_goto_sleep_until'
collect2.exe: error: ld returned 1 exit status

earlephilhower, 2023-03-14:
Make sure you're running your git copy, not the Boards Manager install.

The script assumes you're running from the git version of  the core and installs the rebuilt libraries into that path, not the path that the Arduino IDE uses for Boards Manager.

I believe CMake will give very clear error messages and the script will abort if there is a compile error.

Also, `stdio_init_all` is not legal but being called in your sketch, so drop that too.

viduq, 2023-03-26:
Is it planned to implement this to the upstream arduino-pico? If yes, when approximately?

yaswanthsk04, 2023-06-17:
Is this feature already implemented? if no, when will be available?

viduq, 2023-06-26:
Maybe open a new issue. Don't think we'll get an answer here, since this is closed.

Phoenix1747, 2023-06-26:
> Is this feature already implemented? if no, when will be available?

https://github.com/earlephilhower/arduino-pico/discussions/1544#discussioncomment-6205182

viduq, 2023-06-26:
Thank you, didn't see this comment until now.

therman89, 2023-09-07:
Hi,
I ran the get.py script  from the git repo which downloaded and according to the output, it extracted the files. Then I tried to run the make-libpico.sh script, but it throws an error that the Pi Pico SDK is missing from arduino-pico/pico-sdk folder. Did I miss something with the install?
I'm using Linux Mint.

`tibi@tibi-VivoBook:~/arduino-pico/tools/libpico$ bash make-libpico.sh 
Using PICO_SDK_PATH from environment ('/home/tibi/arduino-pico/pico-sdk')
CMake Error at pico_sdk_import.cmake:57 (message):
  Directory '/home/tibi/arduino-pico/pico-sdk' does not appear to contain the
  Raspberry Pi Pico SDK
Call Stack (most recent call first):
  CMakeLists.txt:3 (include)

-- Configuring incomplete, errors occurred!
`

Sokoloft, 2023-10-20:
Been messing with this all day with no success. I keep getting the undefined reference error and cant fix it. I've built it on my linux laptop using the make-libpico.sh script and it ran without error. Moved it back over to my windows desktop and I still have this undefined reference error.

xrevert, 2024-06-11:
It took me so many days to get it to compile. Worked with Ming64. but I am getting this error now:

`c:/users/myuser/appdata/local/arduino15/packages/rp2040/tools/pqt-gcc/2.2.0-d04e724/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/bin/ld.exe: C:\Users\myuser\AppData\Local\Temp\arduino\sketches\6A7C4EDC103DDEFA1BCC2E875EAC0882\sketch\RP2040_story.ino.ino.cpp.o: in function `loop':
c:\users\myuser\appdata\local\arduino15\packages\rp2040\hardware\rp2040\3.9.2\pico-extras\src\rp2_common\pico_sleep\include\pico/sleep.h:50: undefined reference to `sleep_run_from_dormant_source'
c:/users/myuser/appdata/local/arduino15/packages/rp2040/tools/pqt-gcc/2.2.0-d04e724/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/bin/ld.exe: c:\users\myuser\appdata\local\arduino15\packages\rp2040\hardware\rp2040\3.9.2\pico-extras\src\rp2_common\pico_sleep\include\pico/sleep.h:89: undefined reference to `sleep_goto_dormant_until_pin'
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1`

Does this mean my build was not successful?

toseef-ahmd, 2024-07-29:
Hi. Is there anyone who has solved the RP2040 not waking up from dormant mode? I'm trying to get it back on key interrupt, but it doesn't wake up.

I have pico-sdk and pico-extras installed, as this thread suggests. I am able to use sleep.h functionality. 

But the mcu does not want to wake up once it goes to dormant mode. 

Anyone? please!

DerJantob, 2024-08-07:
Had the same issue, but it worked with this:

```
sleep_run_from_xosc();
sleep_goto_dormant_until_pin(15, true, true);  // Pin, Edge, HIGH
rosc_write(&rosc_hw->ctrl, ROSC_CTRL_ENABLE_BITS);
clocks_init();
set_sys_clock_khz(133000, true);
```

Humancell, 2024-09-19:
Hello,

I apologize if this is beating a dead horse hoever I am unclear about the status of this topic.  With the recent RP2350 introduction I'm curious if it still the case that the "standard" board support does not support Deep Sleep?

This is such a critical feature for so many low power/battery applications.  We're working on several solar powered remote applications, and have had to move to the ESP32 for these.

It seems from this thread there is a well known solution to manually add Deep Sleep, but it has not yet been rolled into the "standard" library, and that a custom build must be created?  Is that accurate?

Will this same manual process work for the new RP2350?

What contribution is being looked for to make this already included?

Thank you for any insights in advance!

NuclearPhoenixx, 2024-09-19:
> I apologize if this is beating a dead horse however I am unclear about the status of this topic. With the recent RP2350 introduction I'm curious if it still the case that the "standard" board support does not support Deep Sleep?

Especially regarding the RP2350, IIRC there are also additional low-power/sleep modes available now. So that would make that even more interesting!

earlephilhower, 2024-09-19:
I'm happy to entertain PRs here, especially since the RP2350 has much better low power support, but I'm not really able to consistently measure very low sleep mode currents to actually test things.  If you have some additions, please do send them in!

seamusdemora, 2024-11-17:
@SIeeepy : 
> On a side note I'm still not getting near the low sleep power consumption of 180µA promised in the [RP2040 datasheet](https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf) (p. 622). I [...]

AFAIK, **no one** has been able to replicate those (claimed) results. And "The Raspberries" provided no code to support this claim. There seems to be some *dissonance* in the ranks at RPi - esp wrt Pico - & even more esp wrt low-power examples. I've speculated that the RPi "talent pool" may be a bit shallow in Pico - demands of a growing business & all that I'll guess. And since I'm *opining* I'll add this: 

>Seems to me that the most important aspect of any uC is its _**ability to get by on tiny amounts of power**_. This low-power capability is what sets the uC apart from the many "embedded" Linux platforms that consume large quantities of energy.  

Anyway... I've so far been unable to get any low-power example working on my Pico :(  so my hat is off to you! But in closing, I'd like to refer you to [this comment by Earle P](https://github.com/earlephilhower/arduino-pico/issues/2528#issuecomment-2466976999)... in my experience (mostly hardware; software only v. recently) his cmt makes a lot of sense!

maxgerhardt, 2024-11-17:
Per link above, I'm now working on a library to simplify all this low power stuff. I'm down to 760µA at 5V fed into VSYS! (so, 0.7mA, matching the Pico datasheet a bit better.

[image]

seamusdemora, 2024-11-18:
@maxgerhardt  
> Per [link above](https://github.com/earlephilhower/arduino-pico/pull/2620), I'm now working on a library to simplify all this low power stuff. I'm down to 760µA at 5V fed into VSYS! (so, 0.7mA, matching the Pico datasheet a bit better.

AWESOME  :)) Looking forward to trying this!!

DrKRR, 2024-12-05:
Hello Maxgerhardt
I have also worked on Raspberry Pi Pico 2 and Raspberry Pi Pico and got good resuts with your inspiration

seamusdemora, 2025-03-04:
Hello again @maxgerhardt !  It has now been a while since you declared you were `working on a library to simplify all this low power stuff`.  I don't want to sound like a nag, but I just wanted to know if it remains your intention to publish this library in this repo?

jonaskello, 2025-03-21:
I was able to use `sleep_goto_dormant_until_pin()` from pico extras by using this as a starting point: https://github.com/matthias-bs/arduino-pico-sleep/tree/main?tab=readme-ov-file.  You need to copy the two files there into your project. I then replaced the contents of `sleep.c` with the latest version from pico extras: https://github.com/raspberrypi/pico-extras/blob/master/src/rp2_common/pico_sleep/sleep.c. The final code I used was like this:

```
  // Go to sleep until we see a low level on wakeup_pin
  sleep_run_from_xosc();
  sleep_goto_dormant_until_pin(wakeup_pin, 0, 0);

  // Recover from sleep
  sleep_power_up();

  // Reset
  watchdog_reboot(0, SRAM_END, 0);
  for (;;) {
    __wfi();
  }
```

Not sure if it actually helps with power management as I haven't done any measurements but at least this was more than I was able to do directly using this core.

seamusdemora, 2025-03-23:
@jonaskello 
> I was able to use `sleep_goto_dormant_until_pin()` from pico extras by using this as a starting point: https://github.com/matthias-bs/arduino-pico-sleep/tree/main?tab=readme-ov-file. You need to copy the `sleep.*` and `rosc.*` there into your project. I then replaced the contents of `sleep.c` with the latest version from pico extras: https://github.com/raspberrypi/pico-extras/blob/master/src/rp2_common/pico_sleep/sleep.c. 

Thanks for that... it seems we've gotten all we're going to get from @maxgerhardt on this library, so as soon as I finish my current project, I'm going to try this. Hope you'll be around to answer some questions!

seamusdemora, 2025-11-12:
Rather than repeat my opinions here, I'll just post [this link](https://github.com/earlephilhower/arduino-pico/pull/2620#issuecomment-3524043865) to my latest (and possibly final) comment to this thread.