earlephilhower/arduino-pico #348: Hardware RTC not accessable

drakejest, 2021-11-17:
I could not access the hardware [Harrdware RTC api](https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__rtc.html#gacf9024748b942a7ae375cf75951afa9c). 

```
void setup() {
rtc_init();
}

void loop() {
}
```
_error: 'rtc_init' was not declared in this scope_

even if  i  include `#include "hardware/rtc.h"`,  It could not find the library. `fatal error: hardware/rtc.h: No such file or directory`

Is this intentional? the hardware rtc is already in the main sdk so i would assume that i should be able to access it. The fact that the Time example did not use hardware rtc seems to indicate that it is indeed not supported.

Can anybody confirm so that i wont have to keep pursing this?

Thanks for the great work.

khoih-prog, 2021-11-17:
Check `Stackexchange's` [**RP2040 RTC alarm or similar function**](https://arduino.stackexchange.com/questions/86665/rp2040-rtc-alarm-or-similar-function) and [**RP2040_RTC** library](https://github.com/khoih-prog/RP2040_RTC)

drakejest, 2021-11-18:
> 
> 
> Check `Stackexchange's` [**RP2040 RTC alarm or similar function**](https://arduino.stackexchange.com/questions/86665/rp2040-rtc-alarm-or-similar-function) and [**RP2040_RTC** library](https://github.com/khoih-prog/RP2040_RTC)

@khoih-prog I noticed that you dont have .cpp files and the headers only have the function prototype? what is going on with that?

drakejest, 2021-11-18:
I played with @khoih-prog library and made this example

```
extern "C"    // does not work without extern C
{
  typedef void(* rtc_callback_t )(void);

  void rtc_init           (void);
  bool rtc_set_datetime   (datetime_t *t);
  bool rtc_get_datetime   (datetime_t *t);
  void rtc_set_alarm      (datetime_t *t, rtc_callback_t user_callback);
  void rtc_disable_alarm  (void);
  void rtc_enable_alarm   (void);

  bool rtc_alarm_repeats(datetime_t *t);

  void datetime_to_str(char *buf, uint buf_size, const datetime_t *t);
}

datetime_t t = {
  .year  = 2020,
  .month = 06,
  .day   = 05,
  .dotw  = 5, // 0 is Sunday, so 5 is Friday
  .hour  = 15,
  .min   = 45,
  .sec   = 00
};

void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  Serial.begin(9600);
  while(!Serial);
  Serial.println("Serial");

  rtc_init();
  rtc_set_datetime(&t);
}

// the loop function runs over and over again forever
void loop() {
  char datetime_buf[256];
  char *datetime_str = &datetime_buf[0];                // wait for a second
  rtc_get_datetime(&t);
  datetime_to_str(datetime_str, sizeof(datetime_buf), &t);
  Serial.println(datetime_str);
  sleep_ms(1000);
}
```

Output
```
Friday 5 June 15:46:30 2020
Friday 5 June 15:46:31 2020
Friday 5 June 15:46:32 2020
Friday 5 June 15:46:33 2020
Friday 5 June 15:46:34 2020

```

Are the headers for the rtc just missing? That is the only logical conclusion i can come up with that it works with with only prototypes

earlephilhower, 2021-11-28:
The headers are there, but the include path does not have the RTC directory so Arduino can't find it.  It's a trivial fix, let me do a PR...

PR #353 (2021-11-28): Add RTC headers to path and pico_utils to link
Fixes #348

Add the RTC header path in the SDK to the Arduino build path.  Also add
pico_utils to the libpico.a for the datetime_to_str call.

Note that the pico_sdk has a bug in the datetime.h header so you need to
manually bracket the include file as "C":

extern "C" {
    #include "pico/util/datetime.h"
}
commit 1afbbaba (2021-11-28): Add RTC headers to path and pico_utils to link (#353)

Fixes #348

Add the RTC header path in the SDK to the Arduino build path.  Also add
pico_utils to the libpico.a for the datetime_to_str call.

Note that the pico_sdk has a bug in the datetime.h header so you need to
manually bracket the include file as "C":

extern "C" {
    #include "pico/util/datetime.h"
}
closed 2021-11-28 by earlephilhower