earlephilhower/arduino-pico #349: Upload error after good compile on windows and linux

nsilveri, 2021-11-21:
Hello,
every time i try to upload i get this error, i tried with windows and linux on the same computer and i get the same error, i tried on another computer and everything works fine.
How can I solve? 

[image]
[image]

centerorbit, 2022-01-08:
I discovered that if you have Arduino IDE installed via Flatpak (on Linux) you will receive an error on line 250 in get_drives. Whereas if you install Arduino IDE via the download from their website, you don't get such error and upload works.

I believe it's likely a Flatpak permissions issue.

earlephilhower, 2022-01-19:
Good point @centerorbit .  Also see #420 for some other users' experience.  Permissions under snap and flatpak can get messy!

closed 2022-01-19 by earlephilhower