earlephilhower/arduino-pico #351: SPI Issue on Arduino Nano RP2040

alfonsoletizia1, 2021-11-25:
Hi, 

I'm trying to use the library to read/write on a flash memory using the SPI. It seems that the spi pins are not setted correctly, but even if i use the set methods the results does not change. 

The flash library is the SerialFlash.h, the SPI settings i use are: 
  SPI.setTX(PIN_SPI0_MOSI);    
  SPI.setRX(PIN_SPI0_MISO);    
  SPI.setSCK(PIN_SPI0_SCK);
  SPI.setCS(PIN_SPI0_SS);

I've also tried in using the SDSF, but i cannot initialize the sd card. 

Are there any known issue about the use of the SPI on the arduino nano?

earlephilhower, 2021-11-28:
Without a full MCVE it's hard to give much help here, but the one thing you need to be aware of is that the pin change can only be done *before* `SPI.begin()` is called.  So place those calls at the top of `setup()` or in an object constructor who's calling `SPI.begin()`.

alfonsoletizia1, 2021-11-30:
Thanks of your advice, i didn't looked at the pin number defined here with respect to the pico pinout.

closed 2021-11-30 by alfonsoletizia1