earlephilhower/arduino-pico #361: ARM CMSIS function `NVIC_SystemReset()` is not available

jandrassy, 2021-12-03:
I want to support PR2040 in my ArduinoOTA library and it of course requires self reset of the MCU. For other ARM based MCU I use NVIC_SystemReset .

NVIC_SystemReset is a standard function for an ARM CPU self reset and it is available in other ARM MCU based cores (SAMD, nRF5).

In this core it is in pico-sdk\src\rp2_common\cmsis\stub\CMSIS\Core\Include\core_cm0plus.h` which is included among other required includes in `pico-sdk\src\rp2_common\cmsis\stub\CMSIS\Device\RaspberryPi\RP2040\Include\RP2040.h`. The paths for these includes  are not listed in platform_inc.txt.

When I add
```
-iwithprefixbefore/pico-sdk/src/rp2_common/cmsis/stub/CMSIS/Core/Include
-iwithprefixbefore/pico-sdk/src/rp2_common/cmsis/stub/CMSIS/Device/RaspberryPi/RP2040/Include
```
to platform_inc.txt I still can't include the right RP2040.h, because a .h with the same name exists in /core. Only after I renamed the RP2040.h in /core I could build a sketch with NVIC_SystemReset .

Or is there some other way for the self-reset?

earlephilhower, 2021-12-03:
> I want to support PR2040 in my ArduinoOTA library and it of course requires self reset of the MCU. For other ARM based MCU I use NVIC_SystemReset .
...
> In this core it is in pico-sdk\src\rp2_common\cmsis\stub\CMSIS\Core\Include\core_cm0plus.h`which is included among other required includes in`pico-sdk\src\rp2_common\cmsis\stub\CMSIS\Device\RaspberryPi\RP2040\Include\RP2040.h`. The paths for these includes are not listed in platform_inc.txt.
...
> to platform_inc.txt I still can't include the right RP2040.h, because a .h with the same name exists in /core. Only after I renamed the RP2040.h in /core I could build a sketch with NVIC_SystemReset .

Looks like I chose an inopportune name there.  I think this is a bug and I should rename `RP2040.h` as well as add those new includes.

Also, I believe the CMSIS directory from the pico-sdk will need to be included in the `libpico.a` file (you need to call a function, not an inline from the header, right?).

Let me throw a PR out there with the rename/updated libs.

earlephilhower, 2021-12-03:
@jandrassy can you please undo your local changes and try #362 and report back?  Thx!

jandrassy, 2021-12-03:
it builds and the reset works. thank you.

on the other platforms CMSIS is included within Arduino.h, here I have to extra include the RP2040.h.  It is not a problem for my library, only something to consider for the core.

earlephilhower, 2021-12-03:
Thanks for the update.  I added the CMSIS RP2040.h to Arduino.h and it'll be in the next point release.

closed 2021-12-03 by earlephilhower

jandrassy, 2021-12-13:
@earlephilhower my ArduinoOTA library git master now supports RP2040 with this core.

PR #362 (2021-12-03): Support CMSIS and rename conflicting header file
Fixes #361