earlephilhower/arduino-pico #369: variants are missing LED_BUILTIN define

bperrybap, 2021-12-09:
The variants are not creating the LED pin define LED_BUILTIN
For historical reasons this must be a define.
There was much discussion over this over the years on the mailing list and in the end due to how things evolved through IDE versions and across platforms, this symbol must exist as a define.
You can still make it const int if that is desirable but there still must be a cpp defined symbol with the same if you take this approach.
This is because sketch and library code needs to be able to compile time detect the existence of a LCD control pin.
i.e. 
Either something like this:
```
#define LED_BUILTIN PIN_LED
```
or
```
const int LED_BUILTIN = PIN_LED;
#define LED_BUILTIN LED_BUILTIN
```

defining a self referencing symbol is a convenient NOP that allows the creation of a symbol that matches a variable.
And yes this usage is allow as the C standard specifies that macros must not automatically expand self references.

jandrassy, 2021-12-09:
there is #include "../generic/common.h"

bperrybap, 2021-12-09:
oops.
I just saw that and was about to close the issue.
My apologies.