earlephilhower/arduino-pico #373: Problem with WiFiNINA library for Arduino Nano Connect (`__has_include` directive)

jandrassy, 2021-12-15:
I found a strange problem for Arduino Nano Connect
these lines in nina_pins.h completely disable the WiFiNINA library. 
```
#if __has_include("WiFiNINA.h")
#  define NINA_ATTRIBUTE
#else
#  define NINA_ATTRIBUTE __attribute__ ((error("Please include WiFiNINA.h to use this pin")))
#endif
```
The compile messages of examples compilation are as if the library were not included.
if I comment out all these lines except of `#  define NINA_ATTRIBUTE`, the examples compile.

I know that this file is copied from Arduino mbed core for the Nano Connect.
is `__has_include` enabled in Pico core? it is non standard

earlephilhower, 2021-12-15:
`__has_include` is a GCC thing and it is supported w/the 10.x version of GCC we're using.  The `SdFat` library uses it to exclude its `File` implementation which clashes w/our own, for example.

For the `#define` issue, what are you building?  Do you have a suggestion to fix?  I don't personally have the Arduino Nano Connect, so I'm kind of working by proxy to support it.

Thx!

jandrassy, 2021-12-15:
I have [a first user](https://forum.arduino.cc/t/status-on-local-ota/913276/10) for the ArduinoOTA on RP2040. I don't have the Nano Connect I only try to build. The Arduino Nano Connect has ublox Nina module (esp32) with Arduino nina-fw as WiFi adapter  and it is used with the Arduino WiFiNINA library.
The error occurs even with the basic WiFiWebClient example of the library.

The nina-fw allows control of io pins of the WiFi module. For some reasons the functions are declared at the end of nina_pins.h and the purpose of the define is to make them invalid if WiFiNINA is not in sketch. I guess the implementation of those functions is in WiFiNINA library.

khoih-prog, 2021-12-31:
You can use this new [**WiFiNINA_Generic** releases v1.8.14-2](https://github.com/khoih-prog/WiFiNINA_Generic/releases/tag/v1.8.14-2), without having to modify anything.

Have a look at [WiFiWebClientRepeating on ARDUINO_NANO_RP2040_CONNECT](https://github.com/khoih-prog/WiFiNINA_Generic#3-wifiwebclientrepeating-on-arduino_nano_rp2040_connect)

---

### Releases v1.8.14-2

1. Add support to `Nano_RP2040_Connect` using `arduino-pico` core
2. Update `Packages' Patches`

earlephilhower, 2022-01-01:
Looks like a dup of #376, which @khoih-prog has fixed with a new release of his library, so closing this one.  If there is a real core issue, please let us know...

jandrassy, 2022-01-02:
@earlephilhower this is a different issue

khoih-prog, 2022-01-02:
Sorry, will post a new PR soon to fix this for WiFiNINA.

khoih-prog, 2022-01-02:
@jandrassy 

Please test the new [**PR** #403](https://github.com/earlephilhower/arduino-pico/pull/403)

I'm able now to compile and run OK with WiFiNINA for Nano_RP2040_Connect

jandrassy, 2022-01-02:
@khoih-prog, so throwing the foolproof guard away of curse makes it compile, but I see no reason why it shouldn't work. It is some strange compiler or preprocessor problem. That is of course not a problem you should solve.

jandrassy, 2022-01-02:
this is the example for the RGB LED pins (the LED is on the NINA module as are A4 to A7)
```

#include <WiFiNINA.h>

void setup() {
  pinMode(LEDR, OUTPUT);
  pinMode(LEDG, OUTPUT);
  pinMode(LEDB, OUTPUT);}

void loop() {
  digitalWrite(LEDR,HIGH); // Turn On RED LED
   delay(250);
   digitalWrite(LEDR,LOW); // Turn Off RED LED

   digitalWrite(LEDG,HIGH); // Turn On GREEN LED
   delay(250);
   digitalWrite(LEDG,LOW); // Turn Off GREEN LED

   digitalWrite(LEDB,HIGH); // Turn On BLUE LED
   delay(250);
   digitalWrite(LEDB,LOW); // Turn Off BLUE LED
}
```

khoih-prog, 2022-01-02:
@earlephilhower 

Can you check why there is no (__has_include) defined? Some compiler directive is missing???

That's why we can't use `#if __has_include("WiFiNINA.h")`

Use this to check

```
#if defined (__has_include)
  #warning Not __has_include defined
#endif
```

@jandrassy 

If we remove all NINA_ATTRIBUTE-related from your PR #404, it's working OK using your sketch, LED are blinking OK.

The issue now is (__has_include) is not defined and usable.

earlephilhower, 2022-01-02:
`__has_include` is not a `#define`, it's a preprocessor built-in macro.  You can't check for `ifdef(__has_include)`, youe need to check for a specific GCC extension (I think).

````

void setup() {
  Serial.begin();
  Serial1.begin(1200);
  s.begin(1200, SERIAL_8N1);
  delay(5000);
#if __has_include("stdio.h")
  Serial.printf("__has_include(stdio.h) = true\n");
#else
  Serial.printf("__has_include(stdio.h) = false\n");
#endif
}
````
gives
````
__has_include(stdio.h) = true
````

jandrassy, 2022-01-02:
in Eclipse Sloeber I don't have the error (it uses a makefile). 
the error is only in Arduino IDE. `-I` for WiFiNINA library is missing on compiler command line.

earlephilhower, 2022-01-02:
Err, so I did read the link to the GCC 10.1 docs and it seems they do fake it even though it's not a real `#define`.

But, it still works fine following their examples. So, I think you have another problem somewhere...
````
void setup() {
  Serial.begin();
  Serial1.begin(1200);
  s.begin(1200, SERIAL_8N1);
  delay(5000);
#if defined __has_include
#if __has_include("stdio.h")
  Serial.printf("__has_include(stdio.h) = true\n");
#else
  Serial.printf("__has_include(stdio.h) = false\n");
#endif
#endif
Serial.printf("xxx\n");
}
void loop() { }
````
gives
````
12:23:23.176 -> __has_include(stdio.h) = true
12:23:23.176 -> xxx
````

earlephilhower, 2022-01-02:
> in Eclipse Sloeber I don't have the error (it uses a makefile). the error is only in Arduino IDE. `-I` for WiFiNINA library is missing on compiler command line. so it points to arduino-cli. something confuses it.

Library includes are added by the IDE itself, not the core.  So, maybe the __has_include is not understood by the arduino-cli and so it does not search for the appropriate header path to include while building?

khoih-prog, 2022-01-02:
I just check and `__has_include()` is working correctly inside WiFiNINA ibrary and elsewhere, but inside `nina_pins.h`,
My interpretation is that it's not correct to put those `__has_include()` checks too early inside `nina_pins.h`.

This seems to be a kludge to me. I haven't found out why `__has_include()` is working for `Arduino-mbed` core inside `nina_pins.h`

If we move

```

#if __has_include("WiFiNINA.h")
  #warning __has_include("WiFiNINA.h")
  #  define NINA_ATTRIBUTE
#else
  #warning Not __has_include("WiFiNINA.h")
  #  define NINA_ATTRIBUTE __attribute__ ((error("Please include WiFiNINA.h to use this pin")))
#endif

void      NINA_ATTRIBUTE pinMode     (NinaPin pin, PinMode mode);
PinStatus NINA_ATTRIBUTE digitalRead (NinaPin pin);
void      NINA_ATTRIBUTE digitalWrite(NinaPin pin, PinStatus value);
int       NINA_ATTRIBUTE analogRead  (NinaPin pin);
void      NINA_ATTRIBUTE analogWrite (NinaPin pin, int value);
```

from `nina_pins.h` to the end of [wifi_drv.h](https://github.com/arduino-libraries/WiFiNINA/blob/master/src/utility/wifi_drv.h), everything will be OK, and we don't even need to check `#if __has_include("WiFiNINA.h")`

Just a quick thought for a direction to solve the issue.

earlephilhower, 2022-01-02:
Maybe there's a saner way of going about this?  Even if there is an include `WiFiNINA.h` there's no guarantee it's actually been included somewhere beforehand, no?  Why not check for something actually defined in the header instead of just seeing it could *possibly* have been included?

khoih-prog, 2022-01-02:
> Maybe there's a saner way of going about this?

That's what I'm thinking. These pins are good only for WiFiNINA + Nano_RP2040_Connect. Putting then inside WiFiNINA is better because if `WiFiNINA.h` is not included, you certainly can't use those pins.

jandrassy, 2022-01-03:
We can't change what Arduino does and putting __has_include WiFiNINA into WiFiNINA misses the purpose.

the __has_include WiFiNINA is only an extra move by Arduino for foolproof. it is not necessary for the functionality.
so we can stay by the current already merged removal of this 'guard'

jandrassy, 2022-01-04:
reopened. I thought the fix was already merged, but it was not.

commit 52a22e51 (2022-01-01): Fix Nano_RP2040_Connect pinout issue (#400)

Fix issues with [WiFiNINA](https://github.com/arduino-libraries/WiFiNINA) and [WiFiNINA_Generic](https://github.com/khoih-prog/WiFiNINA_Generic) libraries

Fixes #373
Fixes #375 
Fixes #376
commit 9408b7e2 (2022-01-09): Fix WiFiNINA issue for Arduino Nano Connect (#403)

Fixes #373
closed 2022-01-09 by earlephilhower