earlephilhower/arduino-pico #374: 'USBD_MAX_POWER_MA' was not declared in this scope

jandrassy, 2021-12-15:
current master doesn't compile
Arduino/hardware/pico/rp2040/cores/rp2040/RP2040USB.cpp:209:118: error: 'USBD_MAX_POWER_MA' was not declared in this scope

In SDK I found USBD_MAX_POWER_MA only in src/rp2_common/pico_stdio_usb/stdio_usb_descriptors.c

it looks like `#define USBD_MAX_POWER_MA (250)` is missing in RP2040USB.cpp

PontusO, 2021-12-15:
Exactly what are you building? USBD_MAX_POWER_MA has been moved out to the Arduino configuration framework and is now included from boards.txt. An old boards.txt could possibly cause this.

jandrassy, 2021-12-15:
yes, sorry. I blame Sloeber (Eclipse plugin). it makes copies of boards.txt and platform.txt and doesn't update them for manually installed package

closed 2021-12-15 by jandrassy