earlephilhower/arduino-pico #375: how i can run Example "ConnectWithWPA"

kendo55, 2021-12-16:
how i can run Example "ConnectWithWPA" 

1. I included the "WiFiNINA" library 
over Sketch -> include Library -> manage Libraries
and
she was installed in <Sketch_folder>/libraries/WiFiNINA

2. if I compile "ConnectWithWPA" Example those Errors are reportet:
------------------------------------------------------------------------------------
ConnectWithWPA:18:14: error: 'WL_IDLE_STATUS' was not declared in this scope
   18 | int status = WL_IDLE_STATUS;     // the WiFi radio's status
      |              ^~~~~~~~~~~~~~
D:\projekte\RP2040\sketchbook\ConnectWithWPA\ConnectWithWPA.ino: In function 'void setup()':
ConnectWithWPA:28:7: error: 'WiFi' was not declared in this scope
   28 |   if (WiFi.status() == WL_NO_MODULE) {
      |       ^~~~
ConnectWithWPA:28:24: error: 'WL_NO_MODULE' was not declared in this scope
   28 |   if (WiFi.status() == WL_NO_MODULE) {
      |                        ^~~~~~~~~~~~
etc...
--------------------------------------------------------------------------

3. I tried to move the Library to 
"D:\arduino-1.8.16_RP2040_SDK\portable\packages\rp2040\hardware\rp2040\1.9.8\libraries"
like in ESP32 environment..
but no sucess! 

4. I removed the root Lib:  D:\arduino-1.8.16_RP2040_SDK\libraries\WiFi

no sucess!

jandrassy, 2021-12-16:
https://github.com/earlephilhower/arduino-pico/issues/373

Claupio, 2021-12-31:
Renaming the file WiFiNINA.h to WiFiNINA2.h and obviusly change the include in the example; for me it works (compile) but the tcp and udp doesn't work

jandrassy, 2021-12-31:
> Renaming the file WiFiNINA.h to WiFiNINA2.h and obviusly change the include in the example; for me it works (compile) but the tcp and udp doesn't work

does it work with the Arduino core for Nano Connect?

Claupio, 2021-12-31:
it works with this core but I don't know if this works also in the official arduino core

khoih-prog, 2021-12-31:
You can use this new [**WiFiNINA_Generic** releases v1.8.14-2](https://github.com/khoih-prog/WiFiNINA_Generic/releases/tag/v1.8.14-2), without having to modify anything.

Have a look at [WiFiWebClientRepeating on ARDUINO_NANO_RP2040_CONNECT](https://github.com/khoih-prog/WiFiNINA_Generic#3-wifiwebclientrepeating-on-arduino_nano_rp2040_connect)

---

### Releases v1.8.14-2

1. Add support to `Nano_RP2040_Connect` using `arduino-pico` core
2. Update `Packages' Patches`

commit 52a22e51 (2022-01-01): Fix Nano_RP2040_Connect pinout issue (#400)

Fix issues with [WiFiNINA](https://github.com/arduino-libraries/WiFiNINA) and [WiFiNINA_Generic](https://github.com/khoih-prog/WiFiNINA_Generic) libraries

Fixes #373
Fixes #375 
Fixes #376
closed 2022-01-01 by earlephilhower