earlephilhower/arduino-pico #376: UPD not running with WiFiNINA library (Example WiFiUdpSendReceiveString)

kendo55, 2021-12-17:
Board: Nano RP2040 Connect
Lib: WiFiNINA 1.8.13 fist trie, then I changed to this "Version" [https://github.com/ocrdu/WiFiNINA](url)

First of all I followed this hint #373 --> and the connection to the AP runs! Thanks to @jandrassy

But the UDP Sever is not working! (Example: WiFiUdpSendReceiveString)

on the other environment, with the original Arduino Mbed OS Nana Bords Manager  
--> -> it's going as it has been since centuries :-)
on AVR...STMxx ESP8266 32 etc....

jandrassy, 2021-12-18:
why do you use ocrdu's library?

kendo55, 2021-12-18:
i used ocrdu's library WiFiNINA --> this "Version:" [https://github.com/ocrdu/WiFiNINA](url)

jandrassy, 2021-12-18:
> i used ocrdu's library WiFiNINA --> this "Version:" [https://github.com/ocrdu/WiFiNINA](url)

why not the official one by Arduino?

kendo55, 2021-12-18:
- fist I trie it with the original WiFiNINA 1.8.13
- when with ocrdu's 1.8.12 with roll-back https://github.com/ocrdu/WiFiNINA
of https://github.com/arduino-libraries/WiFiNINA/pull/53 
because of https://github.com/arduino-libraries/WiFiNINA/issues/87

--> in hope, the udp-Example would work

kendo55, 2021-12-18:
for the websocket implementation I will try to use ocrdu's Lib:
https://github.com/ocrdu/NINA-Websocket
but for the UDP implementation it woul by nice to stay on the "real arduino way"

khoih-prog, 2021-12-31:
I confirm that there is certain issue with WiFiNINA UDP with this core.

1. The debug terminal using [**ArduinoCore-mbed core v2.6.1**](https://github.com/arduino/ArduinoCore-mbed) => **OK**

```
Start WiFiUdpNtpClient on MBED NANO_RP2040_CONNECT
WiFiNINA_Generic v1.8.14-2
Attempting to connect to SSID: HueNet1
Connected to WiFi
SSID: HueNet1
IP Address: 192.168.2.86
Signal strength (RSSI):-16 dBm

Starting connection to server...
Packet received
Seconds since Jan 1 1900 = 3849962133
Unix time = 1640973333
The UTC time is 17:55:33
Packet received
Seconds since Jan 1 1900 = 3849962143
Unix time = 1640973343
The UTC time is 17:55:43
Packet received
Seconds since Jan 1 1900 = 3849962154
Unix time = 1640973354
The UTC time is 17:55:54
```

---

2. The debug terminal using this [**arduino-pico core v1.9.11**](https://github.com/earlephilhower/arduino-pico) => **Not OK**

```
Start WiFiUdpNtpClient on ARDUINO_NANO_RP2040_CONNECT
WiFiNINA_Generic v1.8.14-2
Attempting to connect to SSID: HueNet1
Connected to WiFi
SSID: HueNet1
IP Address: 192.168.2.86
Signal strength (RSSI):-17 dBm

Starting connection to server...  <======= Just staying here forever
```

Will have closer look to see what the issue is.

khoih-prog, 2021-12-31:
I confirm that there is some issue with WiFiNINA UDP and this core

1. NANO_RP2040_CONNECT using [**ArduinoCore-mbed core v2.6.1**](https://github.com/arduino/ArduinoCore-mbed) => **OK**

```
Start WiFiUdpNtpClient on MBED NANO_RP2040_CONNECT
WiFiNINA_Generic v1.8.14-2
Attempting to connect to SSID: HueNet1
Connected to WiFi
SSID: HueNet1
IP Address: 192.168.2.86
Signal strength (RSSI):-17 dBm

Starting connection to server...
Packet received
Seconds since Jan 1 1900 = 3849962580
Unix time = 1640973780
The UTC time is 18:03:00
Packet received
Seconds since Jan 1 1900 = 3849962591
Unix time = 1640973791
The UTC time is 18:03:11
Packet received
Seconds since Jan 1 1900 = 3849962601
Unix time = 1640973801
The UTC time is 18:03:21
Packet received
Seconds since Jan 1 1900 = 3849962612
Unix time = 1640973812
The UTC time is 18:03:32
Packet received
Seconds since Jan 1 1900 = 3849962622
Unix time = 1640973822
The UTC time is 18:03:42
Packet received
```

---

2. NANO_RP2040_CONNECT using [**arduino-pico core v1.9.11**](https://github.com/earlephilhower/arduino-pico) => **Not OK**

```
Start WiFiUdpNtpClient on ARDUINO_NANO_RP2040_CONNECT
WiFiNINA_Generic v1.8.14-2
Attempting to connect to SSID: HueNet1
Connected to WiFi
SSID: HueNet1
IP Address: 192.168.2.86
Signal strength (RSSI):-17 dBm

Starting connection to server...     <==== Staying here forever
```

Will spend time to find out why. Any hint will be appreciated.

khoih-prog, 2021-12-31:
OK, found the issue and fixed it. The real problem is the definitions in this core's [**pins_arduino.h**](https://github.com/earlephilhower/arduino-pico/blob/master/variants/arduino_nano_connect/pins_arduino.h) is not compatible with those in [ArduinoCore-mbed core **pins_arduino.h**](https://github.com/arduino/ArduinoCore-mbed/blob/master/variants/NANO_RP2040_CONNECT/pins_arduino.h)

```
static const uint8_t D20 = (2u);
...
static const uint8_t D24 = (3u);
```

while in [ArduinoCore-mbed core **pins_arduino.h**](https://github.com/arduino/ArduinoCore-mbed/blob/master/variants/NANO_RP2040_CONNECT/pins_arduino.h)

```
#define NINA_RESETN		(24u)
...
#define NINA_GPIO0		(20u) // real GPIO0 on NINA
```

So we need this in [WiFiNINA_Pinout_Generic.h](https://github.com/khoih-prog/WiFiNINA_Generic/blob/master/src/WiFiNINA_Pinout_Generic.h)

```
#if defined(ARDUINO_ARDUINO_NANO_RP2040_CONNECT)
    // Fix UDP issue for arduino-pico core
    #define NINA_GPIOIRQ    (D20)   // actual 2, NINA_GPIO0
    #define NINA_RESETN     (D24)   // actual 3, NINA_RESETN
#endif
```

Will post a new release shortly.

khoih-prog, 2021-12-31:
The terminal after the fix => **OK** now

```
Start WiFiUdpNtpClient on ARDUINO_NANO_RP2040_CONNECT
WiFiNINA_Generic v1.8.14-3
Attempting to connect to SSID: HueNet1
Connected to WiFi
SSID: HueNet1
IP Address: 192.168.2.86
Signal strength (RSSI):-16 dBm

Starting connection to server...
Packet received
Seconds since Jan 1 1900 = 3849967073
Unix time = 1640978273
The UTC time is 19:17:53
Packet received
Seconds since Jan 1 1900 = 3849967084
Unix time = 1640978284
The UTC time is 19:18:04

```

khoih-prog, 2021-12-31:
You can try this new [**WiFiNINA_Generic** releases v1.8.14-3](https://github.com/khoih-prog/WiFiNINA_Generic/releases/tag/v1.8.14-3) which fixed the UDP issue.

---

### Releases v1.8.14-3

1. Fix issue with UDP for Nano_RP2040_Connect using arduino-pico core. Check [UDP not running with WiFiNINA library (Example WiFiUdpSendReceiveString) #376](https://github.com/earlephilhower/arduino-pico/issues/376)

kendo55, 2021-12-31:
Thanks!
and happy new year
from Switzerland!

khoih-prog, 2021-12-31:
You're very welcome.
Happy New Year from Canada !

jandrassy, 2022-01-01:
@khoih-prog shouldn't we fix pins_arduino.h for Nano Connect in this core?

khoih-prog, 2022-01-01:
That's the best but so intrusive choice, so I use this faster, simpler way and just fix the library.

If there are more issue creating by this Nano Connect's `pins_arduino.h`, we might have to modify it. But as long as we know it, I don't think there'll be any more.

It's up to @earlephilhower to decide.

earlephilhower, 2022-01-01:
@khoih-prog @jandrassy If we need to change the core here, no problem.  But I have to admit I have no idea what UPD is or what workaround you're doing.  Can you please put in a PR, or at least point me to the patch commits you did as workarounds in your own lib?

jandrassy, 2022-01-01:
@earlephilhower it is a typo for UDP. but that is not important. the problem is with defines for the pins connected on PCB to the networking chip on Nano Connect (an esp32).
@khoih-prog explains the problem in this comment:
https://github.com/earlephilhower/arduino-pico/issues/376#issuecomment-1003438535

khoih-prog, 2022-01-01:
So, these are the pros and cons when changing the core

### Pros

- Can use original WiFiNINA Library and possibly other WiFiNINA-related libraries written for `ArduinoCore-mbed` core
- Easier to port / adapt hardware-dependent libraries from `ArduinoCore-mbed` core
- Cleaner solution

### Cons

- This is **breaking change** for the Nano_RP2040_Connect boards using this core
- Have to retest all libraries and softwares written so far for this core

---

There is also one `incompatibility` here at this core's [**boards.txt**#L3811](https://github.com/earlephilhower/arduino-pico/blob/master/boards.txt#L3811)

```
arduino_nano_connect.build.board=ARDUINO_NANO_RP2040_CONNECT
```

while in `ArduinoCore-mbed` core [**boards.txt**#L253](https://github.com/arduino/ArduinoCore-mbed/blob/master/boards.txt#L253)

```
nanorp2040connect.build.variant=NANO_RP2040_CONNECT
```

That's why [WiFiNINA_Pinout_Generic.h#L107-L108](https://github.com/khoih-prog/WiFiNINA_Generic/blob/master/src/WiFiNINA_Pinout_Generic.h#L107-L108) has to just add a strange board def as follows to fix [#375](https://github.com/earlephilhower/arduino-pico/issues/375) and [#373](https://github.com/earlephilhower/arduino-pico/issues/373)

```
// Double ARDUINO_ due to -DARDUINO_{build.board} in platform.txt
defined(ARDUINO_ARDUINO_NANO_RP2040_CONNECT) 
```

```
#if ( defined(ARDUINO_NANO_RP2040_CONNECT) || defined(ARDUINO_SAMD_NANO_33_IOT) || defined(ARDUINO_SAMD_MKRWIFI1010) || \
      defined(ARDUINO_SAMD_MKRVIDOR4000) || defined(ARDUINO_ARDUINO_NANO_RP2040_CONNECT))
```

khoih-prog, 2022-01-01:
Ah, I found the less intrusive fix by just a simple change in the Nano Connect's pins_arduino.h. Tested OK with WiFiNINA library.

Will post a PR soon.

khoih-prog, 2022-01-01:
The PR has been posted as [**Fix Nano_RP2040_Connect issue** #400](https://github.com/earlephilhower/arduino-pico/pull/400)

commit 52a22e51 (2022-01-01): Fix Nano_RP2040_Connect pinout issue (#400)

Fix issues with [WiFiNINA](https://github.com/arduino-libraries/WiFiNINA) and [WiFiNINA_Generic](https://github.com/khoih-prog/WiFiNINA_Generic) libraries

Fixes #373
Fixes #375 
Fixes #376
closed 2022-01-01 by earlephilhower