earlephilhower/arduino-pico #380: build.mcu mentionned twice in boards.txt

rei-vilo, 2021-12-23:
The `build.mcu` variable is mentioned twice in [boards.txt](https://github.com/earlephilhower/arduino-pico/blob/master/boards.txt).

Example for `rpipico` board

https://github.com/earlephilhower/arduino-pico/blob/7120a1508c09f8d409014552d4b7009106a58623/boards.txt#L20
https://github.com/earlephilhower/arduino-pico/blob/7120a1508c09f8d409014552d4b7009106a58623/boards.txt#L32

`build.mcu` is used for libraries distributed as [pre-compiled binaries](https://arduino.github.io/arduino-cli/0.20/library-specification/#precompiled-binaries). This duplicate entry prevents from using that feature.

rei-vilo, 2021-12-23:
See PR #381

earlephilhower, 2021-12-23:
Thanks for the PR!  I've combined your changes with the proper script changes and things should be working fine now w/binary libs.

closed 2021-12-23 by earlephilhower

rei-vilo, 2021-12-23:
You're welcome!

PR #382 (2021-12-23): Remove redundant/wrong second board.mcu in boards.txt
Fixes #380

Boards.txt had the setting board.mcu twice.  The first one was correct,
while the second one wasn't and shouldn't have been there.  Remove it.
commit 48877984 (2021-12-23): Remove redundant board.mcu in boards.txt, fix precompiled libs (#382)

* Remove redundant/wrong second board.mcu in boards.txt

Fixes #380

Boards.txt had the setting board.mcu twice.  The first one was correct,
while the second one wasn't and shouldn't have been there.  Remove it.

* Include fix from @rei-vilo in #381