earlephilhower/arduino-pico #392: spi.beginTransaction() takes a long time to complete

Bodmer, 2021-12-26:
A line like this using this board package:
```
spi.beginTransaction(SPISettings(SPI_FREQUENCY, MSBFIRST, TFT_SPI_MODE));
```
takes ~50us to complete. Using the Arduino Mbed OS RP2040 package completes in <1us.

It appears that the function is slowed down for some reason, maybe by debug code?

Bodmer, 2021-12-26:
Here is a test sketch that demonstrates the problem. This slows down SPI based sketches significantly.
```
// Arduino RP2040 package   ~0.49us
// Community pacakge      ~ 69.10us
#include <SPI.h>
void setup() {
  Serial.begin(115200);
  SPI.begin();
}

void loop() {
  uint32_t dt = micros();
  for (int i = 0; i < 1000; i++) {
    SPI.beginTransaction(SPISettings(70000000, MSBFIRST, 0));
    SPI.endTransaction();
  }
  Serial.println(micros() - dt);
}
```

earlephilhower, 2021-12-27:
@Bodmer please try the PR #394 and report back, since you seem to have a setup ready to test.

It should make performance basically identical to the MBED one, but I need someone to verify that the change I did still results in working SPI comms...

Bodmer, 2021-12-27:
Yes, performance is now the same.

Thank you.

earlephilhower, 2021-12-27:
> Yes, performance is now the same.

And, just to be 100% sure, it still works, right? :)

Bodmer, 2021-12-27:
Yes, it works for me. I have two SPI devices on the same bus that operate at different SPI clock frequencies and the beginTransaction() and endTransaction() functions operate as expected, changing the SPI frequency when each device is accessed.

Bodmer, 2021-12-27:
Performance is now comparable to the Mbed package. Drawing 10,000 pixels to a TFT, first using beginTransaction() and endTransaction(), second without.
```
1.9.9 Philhower package!
drawPixel = 1486ms
Locked drawPixel = 75ms

Arduino MBED!
drawPixel = 113ms
Locked drawPixel = 77ms

FastSPI Philhower package!
drawPixel = 110ms
Locked drawPixel = 75ms
```

earlephilhower, 2021-12-27:
Thanks for the confirmation!  This will be in the next release, coming soon (serial stuff I want to get in, too).

PR #394 (2021-12-27): Speed up SPI by not reinitting if the paramters are the same as before
Leave the SPI port initted after a transaction, and when a new one comes in
check if it is the same settings we're already running.  If so, do nothing.
If not, deinit and reinit.  In general the settings will be identical, so
this will speed things up massively.

Fixes #392
commit f46d7cc8 (2021-12-27): Speed up SPI by not reinitting if the paramters are the same as before (#394)

Leave the SPI port initted after a transaction, and when a new one comes in
check if it is the same settings we're already running.  If so, do nothing.
If not, deinit and reinit.  In general the settings will be identical, so
this will speed things up massively.

Fixes #392
closed 2021-12-27 by earlephilhower