earlephilhower/arduino-pico #396: PlatformIO's uf2 breaks, whereas Arduino IDE's fine. What's the catch?

Day0Dreamer, 2021-12-29:
_Good day. I have been sitting on a mystery for two days and I need your help._

## Preface:
I love JetBrains and try to use their IDEs everywhere I go. So it's CLion, and Platform IO or raw C++ SDK. Having searched for a CAN Bus library, and finding only Arduino core based, I feel forced to use ever glitching PlatformIO, but such is life.

## Two Pillars (the problem):
I've managed to write a code, that compiles without changes both in Arduino IDE and in CLion+PlatformIO, but:
* Arduino IDE - spits out the perfect uf2 and elf files, which I can't debug _(I didn't find a way to use internal debugger.)_
* CLion IDE - spits out compiled files that hang after configuring the CAN class right on can.available() request _(the debugger just says "SIGNAL 0")_

## Tried ways to mend the issue:
1. Set the Arduino IDE to arduino-pico core (version 1.9.9)
2. Set the PlatformIO to arduino-pico core (latest, using `...platform_packages = maxgerhardt/framework-arduinopico@https://github.com/maxgerhardt/arduino-pico.git...`)
3. Sniping build flags from the Arduino IDE and forcing them upon Platformio.ini
4. Snatching Arduino's .elfs and uploading them through CLion's picoprobe (worked, and hence proved the way of uploading way is not the issue)
5. Tried to match the various settings from Arduino IDE, like Debug Level, Debug Port, Flash Size, CPU Speed, Optimize, USB Stack between IDEs

Alas, everything to no avail. Something is different between the two compiling setups, and I've no idea what it could be.

I've seen people ask for files, so here are some :)

Also, I'd be grateful for any help. This is so weird...

Full logs: 
* Arduino IDE https://pastebin.com/raw/9ZT8Aw5T
* CLion https://pastebin.com/raw/tnJi9q6z

platformio.ini
```[platformio]
default_envs = pico_standard

[env]
monitor_speed = 115200

[env:pico_earlephilhower]
lib_deps = pierremolinaro/ACAN2515@^2.0.9
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower
board_build.filesystem_size = 1M
lib_ldf_mode = chain+
platform_packages = 
	maxgerhardt/framework-arduinopico@https://github.com/maxgerhardt/arduino-pico.git
	maxgerhardt/toolchain-pico@https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.3.1-a/x86_64-w64-mingw32.arm-none-eabi-7855b0c.210706.zip
```

main.cpp
```#ifndef ARDUINO_ARCH_RP2040
#error "Select a Raspberry Pi Pico board"
#endif

#include "Arduino.h"
#include <ACAN2515.h>

static const byte MCP2515_SCK  = PIN_SPI0_SCK ; // SCK input of MCP2515
static const byte MCP2515_MOSI = PIN_SPI0_MOSI ; // SDI input of MCP2515
static const byte MCP2515_MISO = PIN_SPI0_MISO ; // SDO output of MCP2517

static const byte MCP2515_CS  = PIN_SPI0_SS ;  // CS input of MCP2515 (adapt to your design)
static const byte MCP2515_INT = 20 ;  // INT output of MCP2515 (adapt to your design)

ACAN2515 can (MCP2515_CS, SPI, MCP2515_INT) ;

static const uint32_t QUARTZ_FREQUENCY = 20UL * 1000UL * 1000UL ; // 20 MHz

void setup () {
    //--- Switch on builtin led
    pinMode (LED_BUILTIN, OUTPUT) ;
    digitalWrite (LED_BUILTIN, HIGH) ;
[... 158 lines cut ...]
Archiving .pio\build\pico_earlephilhower\libFrameworkArduino.a
Indexing .pio\build\pico_earlephilhower\libFrameworkArduino.a
Linking .pio\build\pico_earlephilhower\firmware.elf
Generating UF2 image
elf2uf2 ".pio\build\pico_earlephilhower\firmware.elf" ".pio\build\pico_earlephilhower\firmware.uf2"
Retrieving maximum program size .pio\build\pico_earlephilhower\firmware.elf
Flash size: 2.00MB
Sketch size: 1.00MB
Filesystem size: 1.00MB
Maximium size: 1044480 Flash Length: 1044480 EEPROM Start: 270528512 Filesystem start 269479936 Filesystem end 270528512
Checking size .pio\build\pico_earlephilhower\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   1.5% (used 3920 bytes from 270336 bytes)
Flash: [          ]   4.9% (used 51212 bytes from 1044480 bytes)
Building .pio\build\pico_earlephilhower\firmware.bin
========================= [SUCCESS] Took 4.01 seconds =========================

Environment          Status    Duration
-------------------  --------  ------------
pico_earlephilhower  SUCCESS   00:00:04.010
========================= 1 succeeded in 00:00:04.010 =========================
Built target Debug

Build finished
```

earlephilhower, 2021-12-30:
This is just guesswork, since I don't use Platform.IO, but here's what I would do:
* Triple check the compile command lines for a core file and a file in your library folder.  Order isn't generally important.
* You have 2 ELF files.  They will not be binary-identical (ELF has things like paths and timestamps), but you can extract the `.text` and `.data` sections and compare.  You can also disassemble .text, too, and look at that.  Assuming you don't have `__DATE__` or the like, of course

Good luck!

Day0Dreamer, 2021-12-30:
Thanks, alas, I am at a loss. I don't seem to understand the logs from arduino ide, or make any sense of cryptic `.text` and `.data` fields inside the .elfs

I suppose it's the time to let go, and possibly wait for an update to something...

earlephilhower, 2021-12-30:
Sorry, looks like there's nothing to be done here, then.  Closing, but if you do find a core issue please file a bug.  Thx!

closed 2021-12-30 by earlephilhower