earlephilhower/arduino-pico #399: 'boards = ' only works with 'pico' and 'nanorp2040connect' when using Platformio

rousslabs, 2021-12-31:
When trying to uploading It seems as though only the boards 'pico' and 'nanorp2040connect' are supported. When changing to anything else, like the 'adafruit_feather' there is an error.

```
> Executing task in folder test with arduino-pico: platformio run --target upload <

Processing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi.git; board: adafruit_feather; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Error: Unknown board ID 'adafruit_feather'
The terminal process "platformio 'run', '--target', 'upload'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

earlephilhower, 2022-01-19:
@maxgerhardt is this something you need to work on in the PIO packaging?

maxgerhardt, 2022-01-19:
Correct, there needs to be a board JSON for each board listed in the core's `boards.txt`. The best way to solve this would be to expand the existing https://github.com/earlephilhower/arduino-pico/blob/master/tools/makeboards.py script that already produces this file to additionally produce the all board JSON files. I did something similiar for my GigaDevice 32 platform in [board_generator.py](https://github.com/CommunityGD32Cores/platform-gd32/blob/main/misc/scripts/board_generator.py#L274-L337).

earlephilhower, 2022-01-19:
@maxgerhardt got it.  Let me take a look this weekend.  It should be almost trivial to add it in to the makeboards script like you said.

rousslabs, 2022-01-21:
That's the ticket! I temporarily forked the repos and did a quick blinky blinky test. Dropped in an "[adafruit_feather_rp2040.json](boards/adafruit_feather_RP2040.json)" board definition and successfully compiled/uploaded to the board.

Screenshot:
[image]

maxgerhardt, 2022-01-21:
Okay I see you've created https://github.com/rousslabs/platform-raspberrypi/blob/develop/boards/adafruit_feather_RP2040.json which sets the variant to `adafruit_feather_RP2040` but since this folder name does not exist here in [`variants`](https://github.com/earlephilhower/arduino-pico/tree/master/variants), you renamed the `adafruit_feather` folder to be called `adafruit_feather_RP2040` in https://github.com/rousslabs/arduino-pico/commit/15046addccf061bdc0c99ff41f921643d50483ed. Works too, but you could also directly use the regular arduino-pico repo and reference `adafruit_feather` as your `variant`.

earlephilhower, 2022-01-22:
@maxgerhardt , Can you please review #426 and see if that's the format you're looking for?  I'm unsure about the first IDs as well as if I've correctly changed "arduino" to "earlephilhower" in the right spots...

closed 2022-01-28 by earlephilhower

maxgerhardt, 2022-01-28:
I'm intending to do a PR later for the documentation in regards to how select the now generated boards, i.e, set `board = ..` to the filename (without extension) available in https://github.com/maxgerhardt/platform-raspberrypi/tree/develop/boards so that this becomes clearer.

PR #426 (2022-01-28): Generate Platform.IO board definitions
As part of the boards.txt generation, also make necessary files for
Platform.IO.

Fixes #399
commit 4f27ec44 (2022-01-28): Generate Platform.IO board definitions (#426)

As part of the boards.txt generation, also make necessary files for
Platform.IO.

Fixes #399