earlephilhower/arduino-pico #402: Multicore with picoProbe.

StarflyerAlien, 2022-01-02:
The multicore setup/loop + setup1/loop1 works fine if you load directly onto a pico. However, if you load it through a picoprobe, the second core does not start. If you then power cycle the target pico, then the both cores will start correctly.

maxgerhardt, 2022-06-09:
When using PlatformIO to upload, it seems to restart all cores correctly.

[image]

StarflyerAlien, 2022-06-10:
That's useful to know. The problem I had was when directly uploading from the arduino app. Is that fixed or is it still faulty?

Best Regards,
Michael Leitch-Devlin

-------------------
No electrons were harmed by the sending of this email.
However, large numbers of electrons were slightly disturbed.
Electronic counselling has been offered.
-------------------

[grafik]<https://user-images.githubusercontent.com/26485477/172942136-a7fa5df5-91f7-4fed-88ba-8e4a8acd5f0e.png>

commit fb05d969 (2022-07-14): Add delay to 2nd core start for Picoprobe (#676)

No idea why, but when a Picoprobe upload and reset is used, the 2nd core
does not start w/o a __wfe() call on the first one before launch.

Fixes #674
Fixes #402