earlephilhower/arduino-pico #420: uf2conv.py fails on Linux Mint 20.2

cbrewer314, 2022-01-16:
I'm attempting to build RP2040 code using the Arduino framework by following the instructions at https://www.tomshardware.com/how-to/program-raspberry-pi-pico-with-arduino-ide .

When I attempt to build an executable I receive the following error:

Traceback (most recent call last):
Resetting /dev/ttyACM0
Converting to uf2, output size: 131584, start address: 0x2000
  File "/home/chris/.arduino15/packages/rp2040/hardware/rp2040/1.9.13/tools/uf2conv.py", line 380, in <module>
    main()
  File "/home/chris/.arduino15/packages/rp2040/hardware/rp2040/1.9.13/tools/uf2conv.py", line 360, in main
    drives = get_drives()
  File "/home/chris/.arduino15/packages/rp2040/hardware/rp2040/1.9.13/tools/uf2conv.py", line 250, in get_drives
    for d in os.listdir(rootpath):
FileNotFoundError: [Errno 2] No such file or directory: '/opt/media'
FileNotFoundError: [Errno 2] No such file or directory: '/opt/media'

For what it's worth this version of Linux Mint provides Arduino IDE version 1.8.19 and Python 3.8.10, both installed via the synaptic package manager.

I did look at get_drives(), line 240, before posting this issue. It first checks for /media, which *does* exist on my system (a normal directory with permissions drwxr-xr-x, root/root), but os.path.isdir(rootpath) returns false.

Also, /media/chris (which I think it should be ultimately using?) exists, but is an empty directory on my system, and I'm not sure what uf2conv.py expects to find there.

Andy2No, 2022-01-16:
Does it make a difference if you create the directory /opt/media ?

earlephilhower, 2022-01-16:
@cbrewer314 this boils down to the different way different distros handle removable media.

For upload, what happens is we reboot into UF2 upload mode which converts the Pico to a fake mass storage device for upload.  I think the problem here seems to be that Mint isn't auto-mounting, whereas Ubuntu has always "just worked" out of the box.

Does Mint auto-mount?  If so, where?  And if not, is there a userland tool that can be called to make the mount happen?

Might also be an issue in the Python.  The MSFT upstream didn't look at /opt IIRC, so we might need to protect some `isdir` calls or something in the code.

earlephilhower, 2022-01-16:
Here's the relevant section of code.  No mounted filesystems were found so the listdir call is barfing:

https://github.com/earlephilhower/arduino-pico/blob/d43dcbcf8c8b044538da4412c3dca6264a876cca/tools/uf2conv.py#L239-L251

Also, very basic thing, try replugging in the Pico whie holding bootsel.  The 1st time (or if the Pico is really hung) you need to manually enable UF2 mode because the Pico doesn't know about the special Arduino "reboot to bootloader" ping.

jandrassy, 2022-01-16:
I have Linux Mint 20.1 and no problem with upload to Pico.
but I think I had to make something with udev rules

cbrewer314, 2022-01-17:
First, thank you very much for prompt replies from all of you: it's quite encouraging. With the provided information I'm beginning to understand how this is supposed to work.

Secondly, I do indeed see RPI-RP2 mount in /media/chris when I plug in the pico board while holding the bootsel button.

Third, I added a test line to uf2conf.py, and it barfs at /media. I'm certain this is the source of the problem. At first I thought it might be a permissions issue, but:
/media access is drwxr-xr-x, root/root
/media/chris is drwxr-x---+ chris
/media/Chris/RPI-RP2 is drwxr-xr-x chris
...and all of that seems pretty straightforward.

I'm at a loss to explain why that test is failing. Maybe it's some detail about how os.path.isdir() works? I'm not a python maven but I'll do some research.

As an ugly, ugly workaround I can try creating links for /opt/media to see if that makes the system happy. I'll report my results later today.

earlephilhower, 2022-01-17:
@cbrewer314 so you're saying that `os.path.isdir("/media")` is `False` even though there is a directory there w/proper permissions and everything?

I think Mint disables it, but are you using a snap or flatpak or any other kind of pre-packaged Arduino install?  I know on Ubuntu and Windows, the snaps/WinStore versions often have really tight access restrictions and can't access lots of the filesystem.  If somehow the IDE is running in a container/cgroup or something w/o permission to see `/media` I can see how that test would fail.

So, maybe as an easy test, can you use a downloaded tarball from arduino.cc to run the IDE and see if things still fail?

cbrewer314, 2022-01-17:
Not only am I saying that os.path.isdir("/media") returns False, I am saying that uf2conv.py, within it's given execution context, cannot see a number of files.

If I execute os.listdir("/") within uf2conv, I don't see boot, cdrom, lib32, lib32x, mnt, root, snap, srv, or timeshift.

If I execute that same line in an external test program, I see all of these.

Yes, I used snap to install Qucs Spice, a circuit simulator, but that was done after uf2conf had already failed and I tried working on something else for a while. 😁

So it seems to be something in the execution environment?
For what it's worth, running my test program under both python2 and python3 returns identical results, so it's probably not executing under the 'wrong' interpreter.

Finally, the Arduino IDE came straight from synaptic: the only thing I have ever used snap for was Qucs.

cbrewer314, 2022-01-17:
Oh, and yes, snap is disabled by default in Linux Mint.

cbrewer314, 2022-01-17:
I can't try the tarball test at this time as my internet  is down: I'm replying on my cellphone at the moment.
I will try it once my service is restored.

cbrewer314, 2022-01-17:
Oh! Sorry! I just realized that I had actually installed Arduino IDE 1.8.19 from an Arduino tarball, not from a synaptic repository.
So it's not that.

earlephilhower, 2022-01-17:
I guess right now we only know that Python works, but Python's behavior changes depending on whether you run it from a terminal vs. having the Arduino IDE call it...

I'd also look at the kernel.log/syslog files.  Possibly there's an apparmor or equivalent thing going on when python under whatever process calls the uf2 script?  Might be a "DENIED" or something showing that would help pinpoint.

cbrewer314, 2022-01-17:
I don't administrate linux very often, so I don't know what to look for, or what logs might be useful. I did look at dmesg,but didn't see anything there resulting from running the toolchain.
I'm pretty sure I'm not running anything like apparmor (at least not *deliberately*), but I have no idea what the IDE is doing.I'll try to look into this. By any chance can you point me at documentation regarding the Arduino toolchain execution environment? 

I did see something odd, though.Using sudo I created the /opt directory and added a softlink to /media within it, and rebooted the pico while holding the bootsel button. Saw the drive at /opt/media/chris/RPI-RP2, tried upload, received an error message telling me it couldn't find the serial port. I was so excited I forgot to write down the error message and went straight to rebooting w/o holding the bootsel button, tried uploading it again- and now uf2conv apparently can't see /opt/media anymore?
I can't reproduce that first result- maybe I'm hallucinating...

cbrewer314, 2022-01-18:
I removed the previous 'pico' installation and the Arduino IDE and reinstalled everything from scratch with Arduino 1.8.19, and now everything seems to be working. I suspect that I improperly ran the the pico_setup.sh script while trying to figure out how to make it run without errors (for those playing along at home, linux users not on the Raspberry Pi and not using VSCODE should execute it as:`SKIP_VSCODE=1 SKIP_UART=1 ./pico_setup.sh`).

So it was indeed a user error. I'm closing the issue at this point, and thank you for your patience.

closed 2022-01-18 by cbrewer314

Gabriel38g, 2022-07-25:
I can confirm the problem is most likely the Flatpak version of Arduino IDE.
After getting the FileNotFound ... /opt/media error, 
I uninstalled the Flatpak version of Arduino IDE using Software Manager and downloaded the tarball (arduino-1.8.19-linux64.tar.xz) from the Arduino website and installed it using the provided install.sh. Now everything works great except you have to hold the bootsel button and reconnect every time you want to upload. (I think this is because RPI pico has no 'reset' button, not because of the software.) Thanks for the fix. :+1:

AnachronisticPenguin, 2023-01-08:
If your Arduino IDE is installed by Flatpak you can resolve this error by configuring Flatpak to give the application access to the host file system. I had the same error mentioned in this issue on Pop!_OS 22.10 and this command resolved it for Arduino IDE v2:

`flatpak override --user --filesystem=host cc.arduino.IDE2`

Older IDE versions (e.g., 1.8.x or older) have a different application name:

`flatpak override --user --filesystem=host cc.arduino.arduinoide`

earlephilhower, 2023-01-09:
@AnachronisticPenguin this is good information.  Want to add a small section to the readme or docs with it?