earlephilhower/arduino-pico #435: Link error when using pico-sdk setup_default_uart()

benjylxwang, 2022-01-25:
Hi,

I'm running into a compile error when trying to use setup_default_uart() from "pico/stdlib.h".

Error is:

    ...Arduino15\packages\rp2040\hardware\rp2040\1.9.14/lib/libpico.a(stdio_uart.c.obj): in function `stdio_uart_init_full':
    /home/earle/Arduino/hardware/pico/rp2040/pico-sdk/src/rp2_common/pico_stdio_uart/stdio_uart.c:76: undefined reference to `stdio_set_driver_enabled'
    collect2.exe: error: ld returned 1 exit status
    exit status 1
    Error compiling for board Raspberry Pi Pico.

Here's the minimal example I'm getting it in: 

    #include "pico/stdlib.h"

    void setup() {
        setup_default_uart();
    }
    
    void loop() {
    }

Not sure if this is just a bug or if it's not supported?
Thanks

earlephilhower, 2022-01-25:
That's not supported.  The UARTs are under normal Arduino-style `Serial1/2` control now and the stdio package is very different than what the pico-sdk uses.

closed 2022-01-25 by earlephilhower

Andy2No, 2022-01-25:
> Not sure if this is just a bug or if it's not supported? Thanks

You need to use Serial.begin() in setup() and Serial.print() etc, in loop()

The default (Serial.begin()) is serial over USB, to connect to Arduino's serial monitor window as a console.  If you want to use a physical UART, e.g. for RS232 or 5 pin DIN MIDI, you have to use Serial1 or Serial 2, instead, and you may want to specify which pins to use, because it's configurable from a small number of choices - or just use the default pin numbers.

IIRC, Serial1 relates to UART0 and Serial2 is for UART1, as they appear in the Raspberry Pi Pico and RP2040 datasheets.

benjylxwang, 2022-01-25:
Ah I see. Perhaps I'm coming at this from the wrong direction. I'm trying to change the clock to use the ROSC and all the examples from the pico side I found had it restarting the UART with that command. Would it just be a matter of calling Serial.end() and Serial.begin() before and after changing the clocks to keep it functioning?

earlephilhower, 2022-01-25:
Theoretically that should work, @benjylxwang .  I can't say that I've tried it, but the code is there to support it.  `Serial1.begin()` and `Serial1.end()` would be the calls, of course.