earlephilhower/arduino-pico #436: Tone library glitch

jhmaloney, 2022-01-25:
I'm seeing some strange behavior in the tone() function when I attempt to change the output frequency on a pin. Here's an example:

```
tone(10, 200);
delay(250);
tone(10, 250);
```

Expected behavior: the frequency starts at 200 Hz then changes to 250 Hz after 250 milliseconds. After that, the output should be a nice, symmetric 250 Hz square wave.

Actual behavior observed with an oscilloscope:

[image]

Note the narrow pulses. It behaves as if there are two instances of the state machine outputting to the pin at different frequencies.

After doing some debugging, the issue appears to be that the find() operation is failing to find the existing tone entry for that pin:

```
    auto entry = _toneMap.find(pin);
    Tone *newTone;
    if (entry == _toneMap.end()) { // this is always true, even when there was already an entry for the given pin
```

As a result, it creates a new tone instance and associated state machine.

The code looks correct, but I don't think it is behaving as intended. Unfortunately, I'm not enough of C++ expert to know why find() isn't returning the existing entry.

Andy2No, 2022-01-25:
> ```
> tone(10, 200);
> delay(250);
> tone(10, 250);
> ```

It might be best if you post the entire sketch, assuming it's fairly short.

Is there a delay after the second tone()?  If not, it immediately does the first one again, creating a short glitch, before switching back to 200Hz.

jhmaloney, 2022-01-25:
Those lines are in setup(). loop() is empty.

earlephilhower, 2022-01-25:
@jhmaloney I am unable to reproduce anything like what you're showing.  Can you post your exact `setup`?
````
void setup() {
  tone(10, 200);
  delay(250);
  tone(10, 250);
}
void loop() { }
````
Gives me this output on my logic analyzer:

Right at startup after the 1st tone but before the 2nd:
[image]

And at the transition:
[image]

And a few seconds later:
[image]

So, I see no hiccups and no beat frequencies.  I also instrumented the `find` and it is correctly processing things and writing a new value to the tone register vs/starting a new tone on the same pin.

jhmaloney, 2022-01-25:
There's a complete Arduino sketch below.

New information: The test program works as expected when compiled with the Arduino IDE but manifests the problem I am reporting when built with Platformio. I normally use Platformio with [Max Gerhardt's platformio support](https://github.com/maxgerhardt/platform-raspberrypi). I assumed they were the same except for the platformio parts but it now appears that's not the case.

```
#include "Arduino.h"

void setup() {
	tone(10, 200);
	delay(250);
	tone(10, 250);
}

void loop() { }
```

jhmaloney, 2022-01-25:
Ahh, my comment crossed with yours. Thanks for the super quick response, and sorry for bothering you with something that's working perfectly in your code.

earlephilhower, 2022-01-25:
I think it might just be that you ended up with an old version of the core.  He's just doing the magic scripting to make the code and toolchain work w/PIO.  IIRC, I started out using the CPU and timer interrupts for `Tone()` and that may have had issues, but for many releases now I'm using the PIO(programmable IO, not platformIO!) hardware to do it.  So, if you can, try and pull the git master head into wherever platform.io sticks the sources.

Good luck!

closed 2022-01-25 by earlephilhower

jhmaloney, 2022-01-25:
Max's version is using your PIO code and his Tone.cpp is identical to yours. This could be a bug in the C++ library in his fork, strange as that sounds. It doesn't make sense that find() isn't finding the existing entry. In any case, I will report it to him and you can close this issue.

BTW, thank you so much for creating and maintaining this Arduino framework for the Pico board! I'm the lead developer for MicroBlocks (https://microblocks.fun). We have an interpreter similar to MicroPython, except the source language is visual blocks. It's about 12k lines and uses a lot of low-level features, from Tone and servos to NeoPixels, I2C, SPI, and UARTs. I just started using your framework last week and already nearly everything is working. In fact, I just got persistance and our file system working thanks to your LittleFS library. I started out using the official Arduino library, but I don't like the fact that it is built on mbed. MicroBlocks built with your framework uses substantially less Flash, less RAM, and -- most important to me -- is more understandable when I need to dig into the code. Thanks for all the hard work you've put into it and continue to put into supporting it.

earlephilhower, 2022-01-26:
Microblocks looks pretty neat!  If you get anywhere w/the debug, let me know.  I would still try and replace the core sources with git head, if possible, because if that fixes things then it's just a matter of a new release.  Max and I are working to get a real PIO release going, so it's coming (slowly)...

jhmaloney, 2022-01-26:
Will keep you posted. Glad you and Max are working on a real PIO release. Meanwhile, I can use the Arduino IDE to build with your version and Platformio to build with Max's.

jhmaloney, 2022-01-28:
Max found the problem. The compiler was linking with the wrong version of libstdc++. In that version, a number of library functions were failing, including the map find() function. He fixed it.