earlephilhower/arduino-pico #439: Timezone supported? libraries/rp2040/examples/Time/Time.ino

kendo55, 2022-01-26:
in your example   ....libraries/rp2040/examples/Time/Time.ino   I tried to use the timezone.
But where was no effect, with the struct timezone definition i found in C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\arm-none-eabi\include\sys\time.h

```
```
`#include <time.h>
#include <sys/time.h>

void setup() {
  Serial.begin(115200);
  delay(5000);
  
//configTime(TZ_Europe_Zurich, "pool.ntp.org");

  struct timezone tz;
  tz.tz_minuteswest = -120;
  tz.tz_dsttime = DST_MET;  // 

  struct timeval tv;
  tv.tv_sec = 1611198855; // Jan 21, 2021  3:14:15AM ...RPi Pico Release;
  tv.tv_usec = 0;         // https://www.unixtimestamp.com/
  
  settimeofday(&tv, &tz);   // DST_MET
  //settimeofday(&tv, nullptr);  
}

void loop() {
  time_t now;
  char buff[80];

  time(&now);
  strftime(buff, sizeof(buff), "%c", localtime(&now));
  Serial.println(buff);
  delay(1000);
}
`
```
```
==> Here I found your function settimeofday() and I tried to use the timezone. 

`// packages\rp2040\hardware\rp2040\1.9.8\cores\rp2040\posix.cpp
    
    int settimeofday(const struct timeval *tv, const struct timezone *tz) {
        (void) tz; 
        uint64_t now_us = to_us_since_boot(get_absolute_time());
        if (tv) {
            uint64_t newnow_us;
            newnow_us = tv->tv_sec * 1000000L;
            newnow_us += tv->tv_usec;
            __timedelta_us = newnow_us - now_us;
        }
        return 0;
    }`

`//   C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\arm-none-eabi\include\sys\time.h
 
      struct timezone {
      int tz_minuteswest; /* minutes west of Greenwich */
      int tz_dsttime; /* type of dst correction */
      };
      #define DST_NONE  0 /* not on dst */
      #define DST_USA   1 /* USA style dst */
      #define DST_AUST  2 /* Australian style dst */
      #define DST_WET   3 /* Western European dst */
      #define DST_MET   4 /* Middle European dst */
      #define DST_EET   5 /* Eastern European dst */
      #define DST_CAN   6 /* Canada */
`

khoih-prog, 2022-01-26:
You can try my [**Timezone_Generic** library](https://github.com/khoih-prog/Timezone_Generic), which supports this core, and this [**tzTest** example](https://github.com/khoih-prog/Timezone_Generic/tree/main/examples/tzTest)

earlephilhower, 2022-01-28:
This is expected behavior because that field hasn't been valid in a long time...

> The use of the timezone structure is obsolete; the tz argument should normally be specified as NULL. (See NOTES below.) 

https://linux.die.net/man/2/settimeofday

I can't really comment on the `GNU Arm Embedded Toolchain` as it is not used at all in this core and I don't know what exactly they are doing for things like libc and other things.

Please look at `tzset` and associated calls/environment variables for timezone handling.

closed 2022-01-28 by earlephilhower