earlephilhower/arduino-pico #447: Using pgm_read_word returns unexpected results

caternuson, 2022-01-29:
This seems related to issue #79 and PR #81.

Here is the test sketch:
```cpp
static const PROGMEM uint16_t wordtable[] = {
  0b1000000011110111,
  0b1011001100110011,
  0b1000000000000001,
  0b1000000110000001,
};

void setup() {
  Serial.begin(9600);
  while(!Serial);
  Serial.println("PROGMEM Test.");

  Serial.print("0b"); Serial.println(pgm_read_word(wordtable + 0), BIN);
  Serial.print("0b"); Serial.println(pgm_read_word(wordtable + 1), BIN);
  Serial.print("0b"); Serial.println(pgm_read_word(wordtable + 2), BIN);
  Serial.print("0b"); Serial.println(pgm_read_word(wordtable + 3), BIN);
}

void loop() {
}
```

This returns:
```
PROGMEM Test.
0b1000000011110111
0b11001110000000
0b1011001100110011
0b110110011
```

Changing the preproc logic to `#if 1` to engage the original `pgm_read_` macros produces expected results:
```
PROGMEM Test.
0b1000000011110111
0b1011001100110011
0b1000000000000001
0b1000000110000001
```

earlephilhower, 2022-01-29:
@caternuson thanks for the report and test sketch.  Can you try #449 and verify?  My own testing looks good w/it.

caternuson, 2022-01-29:
Thanks for the quick PR fix! Just tested and it produces expected output:
```
PROGMEM Test.
0b1000000011110111
0b1011001100110011
0b1000000000000001
0b1000000110000001
```

PR #449 (2022-01-29): Fix pgm_read_ macros with inline functions
Fixes #447
commit d8835df9 (2022-01-29): Fix pgm_read_ macros with inline functions (#449)

Fixes #447
closed 2022-01-29 by earlephilhower