earlephilhower/arduino-pico #450: SPI1 MOSI not set correctly

nuwaveit, 2022-01-29:
https://github.com/earlephilhower/arduino-pico/blob/66e48660917c7fa3dac2e97434c58c476e621e8a/variants/rpipico/pins_arduino.h#L24

It should be 11, not 15.  15 is the "physical pin number", but DO1 is GP11, so this should be 11.

earlephilhower, 2022-01-30:
I'm not quite sure about that.  Can you please link to the pinout diagram you're using?  The current pin is valid as far as I can see.  Also, the SPI1 clock would also be moved if we changed the defaults, no?

nuwaveit, 2022-01-30:
Thanks for the reply.  I'm using Adafruits pin card:
https://www.adafruit.com/product/4901

As it so happens, SPI1 clock is correct as 14.  But it's coincidence I
think that SCK1 is GP14 as well as GP10, which happens to be listed as
#14.  As far as I can tell, Arduino is using the GP numbers for the pins,
and I did successfully use an SD Card on SPI in Arduino with it wired to
the GP numbered pins 18, 19, 20.  I didn't have luck with my ili9341 tft on
SPI1 in arduino, even with forking your repo and changing the MOSI pin to
11, but I think its a deeper issue with Adafruit GFX library not fully
supporting rp2040 in hardware SPI mode.  The tft works great in
Circuitpython with the same setup, using clock=GP14 and MOSI=GP11, while
also using the SD on GP18 GP19 GP20.  For both, I am using busio.SPI() to
query the hardware SPI interfaces.

earlephilhower, 2022-01-30:
Thanks for the link,  Those pinouts are for CircuitPython as far as I can see, and not really the same as the Rpi published sheet:
https://datasheets.raspberrypi.com/pico/Pico-R3-A4-Pinout.pdf

If you look at the defines:
````
#define PIN_SPI1_MISO  (12u)
#define PIN_SPI1_MOSI  (15u)
#define PIN_SPI1_SCK   (14u)
#define PIN_SPI1_SS    (13u)
````
they do match the appropriate pins on the RPi Pico datasheet.  They also mirror the SPI0 pins (i.e. the bottom 4 data pins on either side of the PCB).

You don't actually need to change any defines if you've wired things differently, just use the `SPI1.setRX`/etc calls (see https://arduino-pico.readthedocs.io/en/latest/spi.html ) before SPI1.begin().  You can do it 1st thing in your `setup()` routine.

nuwaveit, 2022-01-30:
Oh?  I didn't try that...hmmm...

But I do also have NeoPixels connected to GP5, and in Arduino I define the
pin as 5, and they do work.  Same for the SD_CS, I have it on GP17, and in
Arduino it works with SD.begin(17);

But anyway, thanks, you're library is awesome, and I got it to work in
PlatformIO as well, which is a huge help.

Thanks!!

closed 2022-02-01 by earlephilhower