earlephilhower/arduino-pico #455: Could the EEPROM emulation limit be increased or removed?

technoblogy, 2022-02-03:
The core includes EEPROM emulation developed from the ESP8266 version, and in EEPROM.begin() limits the size to 4096 bytes:
````text
    if ((size <= 0) || (size > 4096)) {
        size = 4096;
    }
````
Is there any reason for this limit? I would like to be able to provide a command to save the workspace from my uLisp interpreter to EEPROM, which is 180Kbytes. There's plenty of flash available.

Under the Arduino Mbed OS RP2040 Boards core I was able to allocate a block device, using FlashIAPBlockDevice, which allowed this sort of size.

technoblogy, 2022-02-03:
Sorry, I think I've seen what the problem is. EEPROM allocates a 4096 RAM sector buffer on the heap, and this would have to increase to the size of the whole Lisp workspace for the change I'm suggesting, which is impractical. I would need to rewrite the code to erase the flash, and then write the Lisp workspace directly to the flash 256 bytes at a time.

earlephilhower, 2022-02-03:
Expanding EEPROM isn't easily possibly as you found out and doesn't make a lot of sense.  Its use case is for old code which ran on the 8-bit AVRs (who did have a real EEPROM) to let them port more easily.  LittleFS is the way to go for anything written in the last 20 years and not complicated at all.  No need to worry about flash at all.

````
File ws = LittleFS.open("workspace.bin", "w");
if (!ws) { error... }
auto len = ws.write(workspace, 160*1024);
if (len != 160*1024) { error... }
ws.close();
````
and
````
File ws = LittleFS.open("workspace.bin", "r");
if (!ws) { error... }
auto len = ws.read(workspace, 160*1024);
if (len != 160*1024) { error... }
ws.close();
````

are pretty much all you need do.  There are examples in the tree you can look at for more info, or the docs.

closed 2022-02-03 by earlephilhower

technoblogy, 2022-02-03:
Thanks a lot. I'll see if I can get that working!

earlephilhower, 2022-02-03:
Also, please make sure you make space for the FS using the `Tools->Flash Size` IDE menu.  You'll also want a `LittleFS.begin()` in your `setup()` somewhere, too, of course.

technoblogy, 2022-02-03:
Thanks - it's working nicely, and was as simple as you described.