earlephilhower/arduino-pico #458: Allow changing I2C clock speed after begin() is called

jhmaloney, 2022-02-04:
While I'm not an Arduino expert, on most of the 32-bit Arduino platforms I use if you want to change the I2C clock speed to 400 kbps "fast mode", you do so after calling begin():

```
	Wire.begin();
	Wire.setClock(400000); // set i2c fast mode
```

However, in this framework it must be done in the reverse order:

```
	Wire.setClock(400000); // set i2c fast mode
	Wire.begin();
```

After Wire.begin() is called, calls to Wire.setClock() have no effect. I discovered this because my I2C OLED display updates were taking much longer on the Pico than on other boards. When I checked with a 'scope, the I2C clock was running at the default speed of ~100 kbps.

I've worked around this with an #ifdef but I wanted to mention it since it might take folks by surprise when porting code from other Arduino frameworks.

earlephilhower, 2022-02-05:
Thanks for the very concise bug report!

#459  should fix this.  Just need to call the right API if we're already running.

closed 2022-02-05 by earlephilhower

jhmaloney, 2022-02-05:
That was quick. Glad it was an easy fix. Thanks!

PR #459 (2022-02-05): Allow changing I2C clock speed while running
Fixes #458
commit c4046602 (2022-02-05): Allow changing I2C clock speed while running (#459)

Fixes #458