earlephilhower/arduino-pico #460: Add support for analogReadResolution and default to 10-bits

jhmaloney, 2022-02-05:
Feature request, not urgent.

Although the [Arduino docs](https://www.arduino.cc/reference/en/language/functions/zero-due-mkr-family/analogreadresolution) say that the analogWriteResolution() function is for the Zero, Due, and Mkr board families, it is supported by many other Arduino frameworks (e.g. nRF5x, ESP32, ESP8266). It is useful when porting Arduino code from other boards. Most frameworks default to 10-bit resolution for backward compatibility with AVR-based Arduinos. Parameter range is 1 to 31 bits. The analogRead() function shifts the result right or left as needed to match the hardware ADC resolution to the desired resolution.

technoblogy, 2022-02-05:
I support this (I assume you meant **analogReadResolution()** in the first sentence), for consistency with other cores.

jhmaloney, 2022-02-05:
Yes, I did. Good catch! I've corrected it above.

closed 2022-02-05 by earlephilhower

jhmaloney, 2022-02-05:
Terrific! Thank you!

commit 8f4c2ca8 (2022-02-05): Add support for analogReadResolution, default 10b

Most other boards and the MBED RP2040 support analogReadResolution which
just shifts read data around as needed, with a default of only 10b of
resolution.  The Pico ADC technically supports 12b, but only has about
8b of real data after noise, so you're not really losing anything in the
general case.

Fixes #460
PR #461 (2022-02-05): Add support for analogReadResolution, default 10b
Most other boards and the MBED RP2040 support analogReadResolution which
just shifts read data around as needed, with a default of only 10b of
resolution.  The Pico ADC technically supports 12b, but only has about
8b of real data after noise, so you're not really losing anything in the
general case.

Fixes #460
commit bbae702e (2022-02-05): Add support for analogReadResolution, default 10b (#461)

Most other boards and the MBED RP2040 support analogReadResolution which
just shifts read data around as needed, with a default of only 10b of
resolution.  The Pico ADC technically supports 12b, but only has about
8b of real data after noise, so you're not really losing anything in the
general case.

Fixes #460