earlephilhower/arduino-pico #464: SerialUART::readBytes() not working

giampiero7, 2022-02-07:
Hi,
In version 1.9.3 it used to work correctly.
Now, version 1.10.0, it always returns 0, while available() returns a positive value.
If relevant, in my case I have a timeout set to 0.
I'll post here again if I get more details.

giampiero7, 2022-02-07:
The issue seems related to the timeout being set to 0, in which case, read() too is not working.

earlephilhower, 2022-02-07:
Great debug.  On inspection, the code definitely immediately times out w/timeout==0 instead of waiting forever.  I'll thrown in a PR right now...`

earlephilhower, 2022-02-07:
D'oh, too early still.  Updated the PR.  If you could undo your workaround and try the simple patch and report back, it would be much appreciated!

jandrassy, 2022-02-07:
peek() and read() should not use Stream._timeout at all

earlephilhower, 2022-02-07:
> peek() and read() should not use Stream._timeout at all

Ah, got it, thanks!  I read the top part of 
https://www.arduino.cc/reference/en/language/functions/communication/serial/settimeout/

But didn't notice the list of affected functions does not include `peek` or `read`.  

> Notes and Warnings
> Serial functions that use  the timeout value set via Serial.setTimeout():

    Serial.find()
    Serial.findUntil()
    Serial.parseInt()
    Serial.parseFloat()
    Serial.readBytes()
    Serial.readBytesUntil()
    Serial.readString()
    Serial.readStringUntil()

Well, one more push and even simpler code coming...

giampiero7, 2022-03-11:
Hi @earlephilhower,
After the modifications from 1.10.0 to 1.11.0 it looks like SerialUART became much slower, unusable in situations where timing is critical, like with ModBus.

During a ModBus request of 8 byte, on 1.10.0 I would receive those 8 bytes in a single readBytes() (available() would return 8).
On 1.11.0 it gets split in chunks of 4 bytes, and the delay between the two chunks is greater than the maximum delay allowed on ModBus.

Any clue what might have happened?

Thanks!

earlephilhower, 2022-03-11:
Use `Serial1->setPollingMode()` to go back to the old way.  The current default is IRQ driven which the hardware waits for either 4 characters read or 4 characters of idle-time before generating the IRQ to cause a flush to the Arduino buffers.

earlephilhower, 2022-03-12:
@giampiero7 can you please undo the `setPollingMode()` call and try the PR in #511?

It should combine the best of IRQ and polling mode and remove any delay in the app receiving data that was already read by the HW FIFO.  Simple tests show it working as expected and returning data to the app immediately on reception, and not waiting for 4 bytes to come in.  That should fix your ModBus issues (and others where the delay is fatal).

earlephilhower, 2022-03-12:
Testing w/ GP0 tied to GP1 (Serial1->Serial2 loopback) [and kids, please do not copy this test's  poor coding style at home!]
````
void setup() {
  Serial.begin(115200);
  Serial1.setPollingMode(false);
  Serial1.begin(110);
  delay(5000);
  Serial.printf("go\n");
}

void loop() {
  Serial1.printf("hello\n");
  int lastavail = -1;
  int avail;
  while ((avail = Serial1.available()) < 6) {
    if (avail != lastavail) {
      Serial.printf("Avail: %d\n", avail);
      lastavail=avail;
    }
  }
  Serial.printf("read: ");
  while (Serial1.available()) {
    Serial.printf("%c", Serial1.read());
  }
  Serial.printf("\n");
  delay(1000);
}
````

Now gives
````

Avail: 0
Avail: 1
Avail: 2
Avail: 3
Avail: 4
Avail: 5
read: hello
````
Whereas before it gave
````
Avail: 0
Avail: 4
read: hello
````

Avail: 0
Avail: 1
Avail: 2
Avail: 3
Avail: 4
Avail: 5
read: hello
````

earlephilhower, 2022-03-14:
Thanks for the report, @giampiero7 .

While it's possible the code has an issue in the hybrid IRQ case, the polling mode code path is completely unchanged except for the acquiring and releasing of a mutex.  In the polling case, the "IRQ" code (which is called by the main app, not an IRQ) is the *only* one to touch that mutex, so there should be 0 behavior changes in that mode  So, could you please double-check there are no other changes to the app or core other than applying the patch in #511?

Also, I believe the patch won't ever hit the 4-byte-time timeout if you're calling `::available()` or `::read()` fast enough.  So I'm really perplexed here.

On my end, I will write a small loopback test that captures the time each character becomes available which should make it easy to spot if there is an issue every 4 chars.  What is the baud rate and the # of chars per message you are using?  I couldn't find anything concrete in the Wikipedia ModBus entry.

giampiero7, 2022-03-14:
Hi @earlephilhower,
In the polling case, I think the behaviour in my test was caused by this little addition:
https://github.com/earlephilhower/arduino-pico/pull/511/files#diff-ec985fdfa3b46c8af6d7b1d05c85f55cc0b1c75dfd86ddaa16c0fc8a5b01197eR130

In my messy test I had a double `begin()`, so I think `setPollingMode(true)`, which was called between the two `begin()`, was ignored.
The 4-bytes issue might have been linked to this too (?)
As a side note, maybe it would be better to call end() inside begin() instead of just returning if the serial was already running?

So, ignore my previous report. The behaviour now seems ok, although in 1.10.0 available() would return 8 upon a single request, while with #511 I get 1 byte per loop, even with polling mode. So there seems to be some difference. Possibly this is not relevant, just to let you know.

While the above seems resolved, with #511 I'm having some issues when running both cores while reading from UART (from one of the cores). The RP gets stuck.
In my test case many things are going on so I wasn't able to pinpoint the root cause, I'm just giving you a heads up in case it is enough for you to spot a potential issue. I'll update you once I get a clearer idea.

earlephilhower, 2022-03-15:
@giampiero7 I have updated the PR to address the mutex init issue and multi-begin like you said.  It's also more paranoid about when the end() is called to make sure there isn't any use-after-free by a stray IRQ handler (which *might* have caused a hang but seems very unlikely to do so since the free'd memory would have to be used for something else within a few clock cycles).

Is the `Serial1.begin()` on the same core as the `Serial1.read()` calls?  If not, could you try moving the `begin()` call to the core that's using it and see if the hangs go away.  If the IRQ and read() are on the same core, the protection method that will work is different (disabling IRQs) than if they're on different cores (the mutex that's tried to acquire in the IRQ...but if the IRQ can't immediately get the mutex it just aborts which means it should not hang, ever).

giampiero7, 2022-03-16:
Thank you, @earlephilhower.
This fixes the multi-begin and the original issue about the delay in IRQ mode. So I would consider this issue closed.

About the multi-core issue, it doesn't seem strictly related to #511, probably its timings changes just highlighted an unrelated issue. In case I'll open a new issue with details.

Thanks and have a great day!

commit 1a3f5531 (2022-02-07): Fix SerialUART::read/peek when timeout==0

Fixes #464

Return data if present if timeout==0 for SerialUART::read/peek, or -1 if not
immediately available.
PR #466 (2022-02-07): Remove timeout for SerialUART/SerialPIO::read/peek
Fixes #464 and other incompatibilities

Remove the timeout check from ::read and ::peek on the SerialUART/PIO
classes  (SerialUSB already ignores it).  See documentation
https://www.arduino.cc/reference/en/language/functions/communication/serial/settimeout/
and thanks to @jandrassy's explanation
https://github.com/earlephilhower/arduino-pico/issues/464#issuecomment-1031657044
commit 4cd65d15 (2022-02-07): Remove timeout for SerialUART/SerialPIO::read/peek (#466)

Fixes #464 and other incompatibilities

Remove the timeout check from ::read and ::peek on the SerialUART/PIO
classes  (SerialUSB already ignores it).  See documentation
https://www.arduino.cc/reference/en/language/functions/communication/serial/settimeout/
and thanks to @jandrassy's explanation
https://github.com/earlephilhower/arduino-pico/issues/464#issuecomment-1031657044
PR #511 (2022-03-16): Avoid "chunkiness" of UART FIFO availability
The UART FIFO will generate an IRQ to transfer data into the SerialUART
FIFOs either every 4 received bytes, or every 4 idle byte times.  This
causes the ::available count to report "0" until either of those two
cases happen, causing a potentially delay in data becoming available to
the app.

Change the code to pull data from the HW FIFO on a read/available/peek.
Use a non-blocking mutex and IRQ disabling to safely empty the FIFO from
user space.  The mutex added to the IRQ is non-blocking and will be
a single CAS the vast majority of the time, so i [... cut]