earlephilhower/arduino-pico #474: Compiles code successfully for Pico, but crashes when trying to program ("cannot find the file specified")

heliophagus, 2022-02-09:
Greetings,

This is my 2nd attempt at getting this build of the Arduino environment for the RPi Pico to program the Pico - alas, with complete lack of success. I'm running 64-bit Win10 & Arduino 1.8.19. Removing the Philhower package & re-installing it has made no difference. Any help would be GREATLY appreciated!

Here are the errors I get when trying to upload:

> Resetting COM7
> Traceback (most recent call last):
> Converting to uf2, output size: 137216, start address: 0x2000
>   File "C:\Users\xxxx\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\1.10.0/tools/uf2conv.py", line 380, in <module>
>     main()
>   File "C:\Users\xxxx\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\1.10.0/tools/uf2conv.py", line 360, in main
>     drives = get_drives()
>   File "C:\Users\xxxx\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\1.10.0/tools/uf2conv.py", line 233, in get_drives
>     "FileSystem,", "DriveType"])
>   File "D:\obj\Windows-Release\37win32_Release\msi_python\zip_win32\subprocess.py", line 395, in check_output
>   File "D:\obj\Windows-Release\37win32_Release\msi_python\zip_win32\subprocess.py", line 472, in run
>   File "D:\obj\Windows-Release\37win32_Release\msi_python\zip_win32\subprocess.py", line 775, in __init__
>   File "D:\obj\Windows-Release\37win32_Release\msi_python\zip_win32\subprocess.py", line 1178, in _execute_child
> FileNotFoundError: [WinError 2] The system cannot find the file specified
> FileNotFoundError: [WinError 2] The system cannot find the file specified

I do not have a folder called "obj" on my D drive (and yes, "hidden" folders are visible in my system). Very strange. 

I'm at a complete impasse. Please help. Thanks in advance!

Andy2No, 2022-02-09:
Judging by the AppData path, it's trying to use version 1.10.0 of arduino-pico.   The current release is 1.9.15.

It seems there may be some limitations, not necessarily relevant to this, if you installed the Windows App Store version of Arduino.  I read that the Teensy package can't be added to it, which suggests permissions problems.  The solution to that was to install it from the Arduino website instead.  If you can't install over the one from the App Store, or it still doesn't work, you'll need to uninstall that one.

You should probably detail what you've done to get this far.  You can install arduino-pico from the Boards Manager, on the Boards menu, or install it manually following instructions in the Readme for this repository. I described the process in discussion #199 (about half way down the page), but it's fairly simple.

I stick to using the Boards Manager, but there can be problems caused by MalwareBytes Anti Malware - I've detailed some workarounds, in #133, and they may apply to other anti malware / antivirus software.  

MBAM tends to lock files that the boards manager needs to overwrite, causing an error message during an update using the Boards Manager, because it can't replace those files.  One simple thing to try, when that happens, is to just reboot, then try it again before doing anything else.

earlephilhower, 2022-02-09:
That line of the script is trying to run `wmic` which is a standard Windows management tool.  Can you try opening up a command prompt and running:
````
wmic PATH Win32_LogicalDisk get DeviceID VolumeName FileSystem DriveType
````
If that gives an error (copy it here) you'll have to figure out how to install it on your system.  I've never met a Windows install w/o it, though.  The line of code in question (and the UF2 upload tool) actually comes from Microsoft Corp. directly. :)

And @Andy2No's comments about the App Store version are correct, it's borked in many instances (but we really are on release 1.10.0).  So don't use the App Store one, use the Arduino.cc MSI installer.

Andy2No, 2022-02-09:
I somehow missed the update notice, @earlephilhower :)  I also didn't see your post until after I'd edited that one to correct myself, after I checked the Releases page.

heliophagus, 2022-02-09:
Aha! Thanks so much for replying. I'm using a standard Arduino install directly from the site (not via the Windows app store). And indeed, when I try to run

> wmic PATH Win32_LogicalDisk get DeviceID VolumeName FileSystem DriveType

I get the following message: "Windows cannot find 'WMIC'". 

And this is not surprising, because Microsoft says ( https://docs.microsoft.com/en-us/windows/win32/wmisdk/wmic ):

> The WMI command-line (WMIC) utility is deprecated as of Windows 10, version 21H1, and as of the 21H1 semi-annual channel release of Windows Server. This utility is superseded by Windows PowerShell for WMI (see [Chapter 7—Working with WMI](https://docs.microsoft.com/en-us/powershell/scripting/learn/ps101/07-working-with-wmi)). This deprecation applies only to the WMI command-line (WMIC) utility; Windows Management Instrumentation (WMI) itself is not affected.

But I do find WMEC in C:\Windows\System32\wbem.....

Probably a PATH snafu. I'll dig further & update when I have more info.

Again, thanks, your help means a lot.

heliophagus, 2022-02-09:
And thanks to @Andy2No as well, of course! :)

earlephilhower, 2022-02-09:
Reopening.  I can add a fallback to call powershell w/the appropriate command simply enough.  Should be less than 5 lines of Python...(famous last words).