earlephilhower/arduino-pico #475: Challenger RP2040 Wifi

atrueresistance, 2022-02-09:
Using the latest arduino ide with the earlephilhower library as mentioned in the iLabs documentation. 

There are samples for some of the iLabs 2040 Challenger Wifi, but none that I see directly mention wifi. 

----
[Website wifi documentation](https://ilabs.se/product/challenger-2040-wifi/) initialization example is sparse.

> To simplify the hardware aspect of the WiFi modem we have included a small helper class in the Arduino IDE that is always available to you as a developer. This new class replaces the previous examples on how to do a proper reset and sync.
>So this is basically all you need to do before starting the WiFi manager.
```
if (Challenger2040WiFi.reset())
    Serial.println(F("WiFi Chip reset OK !"));
else
    Serial.println(F("Could not reset WiFi chip !"));
```
>After these step the modem is initialized and is ready to accept your commands.

>The reset() function of the built in Challenger2040WiFi library will perform a HW reset of the WiFi modem and wait for it to return the “ready” prompt. After this a WiFi manager can be started and start communicating with the modem.

>Here at iLabs we use the WiFiEspAT library and initializing the stack is as easy as
```
WiFi.init(Serial2);
```

---
So that basically ends up with the following error because the helper class can't be found. Does anyone know where this helper class can be found? 
`'Challenger2040WiFi' was not declared in this scope.`

Here is my attempt, sorry I'm new to hardware. If I comment out the Challenger if statement I get "Communication with WiFi module failed!"
```
#include <WiFiEspAT.h>
#define ESP_PORT Serial2
// Emulate Serial1 on pins 6/7 if not present
#if defined(ARDUINO_ARCH_AVR) && !defined(HAVE_HWSERIAL1)
#include <SoftwareSerial.h>
#define AT_BAUD_RATE 9600
#else
#define AT_BAUD_RATE 115200
#endif

WiFiServer server(80);

void setup() {

  Serial.begin(115200);
  while (!Serial);

  Serial1.begin(AT_BAUD_RATE);
  if (Challenger2040WiFi.reset())
    Serial.println(F("WiFi Chip reset OK !"));
  else
    Serial.println(F("Could not reset WiFi chip !"));
  WiFi.init(Serial2);

  if (WiFi.status() == WL_NO_MODULE) {
    Serial.println("Communication with WiFi module failed!");
    // don't continue
    while (true);
  }

  // waiting for connection to Wifi network set with the SetupWiFiConnection sketch
  Serial.println("Waiting for connection to WiFi");
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.print('.');
  }
  Serial.println();

  server.begin();

  IPAddress ip = WiFi.localIP();
  Serial.println();
  Serial.println("Connected to WiFi network.");
  Serial.print("To access the server, enter \"http://");
  Serial.print(ip);
  Serial.println("/\" in web browser.");
}
```

earlephilhower, 2022-02-09:
Have you looked in the Arduino Library Manager?  That's how they'd normally ship a class like that.  Same for the WiFiEspAT, should be a library from them, maybe in the Library Manager, maybe in their GitHub.

PontusO, 2022-02-09:
I can see that I missed an important line there.

Can you try adding #include <ChallengerWiFi.h> at the beginning of your code. That should fix it.

I'll update the instructions tomorrow.
/Pontus

atrueresistance, 2022-02-10:
Thanks @PontusO that seemed to solve the not found issue. Is the `#define ESP_PORT Serial2` the correct way to define for the init `WiFi.init(Serial2);`?

The reason I ask is I pulled up the "ScanNetworksAdvanced" under "basic" for WiFiEspAT examples and it doesn't show any networks, but I know for a fact there are 7 in range.  

```
WiFi Chip reset OK !
MAC: 60:01:94:81:3B:51

Scanning available networks...
** Scan Networks **
number of available networks: 0
```

jandrassy, 2022-02-10:
so this is not the right repo for WiFiEspAT problem, but maybe we can solve it fast.
@atrueresistance turn on logging in the library https://github.com/jandrassy/WiFiEspAT#logging

PontusO, 2022-02-10:
For tech support of the board you can also jump directly to our discord https://discord.gg/RR6HsPyq

jandrassy, 2022-02-10:
@PontusO I see you ar from ILABS? the WiFiEspAT library supports hw reset. `init` has a second optional parameter for the reset pin number.

PontusO, 2022-02-10:
@jandrassy Yes, I am from iLabs. I am aware that the WiFiEspAT library supports hw reset and everyone is encouraged to use that function for compatibility. However our support library also supports putting the ESP in flash download mode to support uploading new firmware to the device. I.e. doing this:
`  Challenger2040WiFi.flashReset()`
would put the ESP in programming mode. Of course thus can be accomplished by setting the mode pin properly, this is just a support library that comes with the board which makes the module easier to understand and use.

earlephilhower, 2022-02-12:
Looks like all is well here.  Closing

closed 2022-02-12 by earlephilhower