earlephilhower/arduino-pico #477: Including SPI and/or EEPROM seems to force Adafruit TinyUSB stack

Doridian, 2022-02-10:
I have started using this in PlatformIO using the steps documented on the readthedocs website.
Everything seems to be working fine, except that as soon as I use SPI, it forces me to swap to the Adafruit TinyUSB stack.

As can be seen by the dependency graph below, it seems SPI depends on it somehow? As does EEPROM?
```
Scanning dependencies...
Dependency Graph
|-- <SPI> 1.0
|   |-- <Adafruit TinyUSB Library> 1.9.2
|-- <Adafruit TinyUSB Library> 1.9.2
|-- <EEPROM> 1.0
|   |-- <Adafruit TinyUSB Library> 1.9.2
```

Did I set up something wrong or is that library required to operate EEPROM or SPI?

earlephilhower, 2022-02-10:
The TinyUSB include is `#ifdef`'d out of normal compilation.  Looks like the PIO preprocessor isn't honoring that.

https://github.com/earlephilhower/arduino-pico/blob/ba92377ebabcd219910bf6a98d6e058b9c0b12c4/libraries/EEPROM/EEPROM.cpp#L28-L35

In theory, nothing should link to the TinyUSB library even if it compiled, w/o that `USE_TINYUSB` so unless you're finding it is (and crashing/etc.), I would not worry about it.

Doridian, 2022-02-10:
> The TinyUSB include is `#ifdef`'d out of normal compilation. Looks like the PIO preprocessor isn't honoring that. [...]

The problem I have is the following is causing it to not even compile unless I explicitly allow it to use the TinyUSB stack (by adding said define to platformio.ini)
```
In file included from C:\Users\Doridian\.platformio\packages\framework-arduinopico\libraries\Adafruit_TinyUSB_Arduino\src\arduino\Adafruit_TinyUSB_API.cpp:30:
C:\Users\Doridian\.platformio\packages\framework-arduinopico\libraries\Adafruit_TinyUSB_Arduino\src/Adafruit_TinyUSB.h:31:2: error: #error TinyUSB is not selected, please select it in "Tools->Menu->USB Stack"
   31 | #error TinyUSB is not selected, please select it in "Tools->Menu->USB Stack"
      |  ^~~~~
```

mtfurlan, 2022-02-19:
Setting `lib_ldf_mode = chain+` in the `platformio.ini` didn't work for me, but `lib_ignore = Adafruit TinyUSB Library` did.

earlephilhower, 2022-03-29:
Fixed with #513

closed 2022-03-29 by earlephilhower