earlephilhower/arduino-pico #478: Included OpenOCD doesn't run

Spegs21, 2022-02-10:
The included OpenOCD executable doesn't run in a Win10 x64 environment. It produces no output when invoked from the command line. This causes use of a Picoprobe target to fail uploading.

Spegs21, 2022-02-10:
When installed by board manager the executable doesn't work but when I downloaded the zip from the json file and manually copied everything over it works. Really strange...

earlephilhower, 2022-02-10:
Tracked in #456 .  OpenOCD on Windows is kind of a mess.  There was a libusb bug which might be biting you with the timeouts, and the packaged Windows version had a DLL issue because `mingw64` was updated when the fixed libusb.dll was built.

Spegs21, 2022-02-10:
> Tracked in #456 . OpenOCD on Windows is kind of a mess. There was a libusb bug which might be biting you with the timeouts, and the packaged Windows version had a DLL issue because `mingw64` was updated when the fixed libusb.dll was built.

@earlephilhower I built the RPi fork of OpenOCD in WSL2 and it seems to work well enough. The problem I had with it is if I build a normal Pico image and attempt to debug it, the breakpoints never hit. This is just like @SirSydom described in #456. So I tried your packaged version and ran into this issue. Do you think the breakpoint issue is related to the OpenOCD build or is their some wrong with the way the image is built? It seems like addresses are wrong. It is supposed to break on main and instead breaks on some random function.

SirSydom, 2022-02-11:
> Do you think the breakpoint issue is related to the OpenOCD

As I pointed out here https://github.com/earlephilhower/arduino-pico/discussions/457, since the problem only occurs with the gdb from this core, and not with a gdb from another source I think this issue (not hitting breakpoints) is gdb related, not openocd.

Spegs21, 2022-02-11:
@earlephilhower You are indeed correct! I switched my `gdbPath` in VS Code to the `arm-none-eabi-gdb.exe` provided by the Adafruit SAMD core and it functions correctly.

earlephilhower, 2022-02-11:
Good debugging all around.  I've cross-compiled a GDB for Win32 with manually bilt expat libs and was able to get the BPs working as expected.  I'll integrate it into the p-q-t toolchain `Makefile` and we should be good.

closed 2022-02-12 by earlephilhower

SirSydom, 2022-02-13:
sorry to bring that up again, but

```
> C:\Users\dosp\AppData\Local\Arduino15\packages\rp2040\tools\pqt-openocd\1.3.3-a-ed6d983\share>C:/Users/dosp/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/1.11.0/../../../tools/pqt-openocd/1.3.3-a-ed6d983/bin/openocd.exe  "-s" "C:/Users/dosp/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/1.11.0/../../../tools/pqt-openocd/1.3.3-a-ed6d983/share/openocd/scripts" "-f" "interface/picoprobe.cfg" "-f" "target/rp2040.cfg"
Open On-Chip Debugger 0.11.0-g610f137 (2022-02-11-14:08)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'swd'
adapter speed: 5000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: Failed to open or find the device
Error: Can't find a picoprobe device! Please check device connections and permissions.
```

I cannot connect to my picoprobe with your binary. Any idea? libusb driver version installed on my system (1.2.6) ?

With the same probe it works with the digikey openocd binary..

```
C:\Users\dosp>"D:/Data/Eigene Dokumente/Projekte/EIB/OpenKNX/Debugging_RP2040/openocd/bin/openocd.exe" "-c" "gdb_port 50000" "-c" "tcl_port 50001" "-c" "telnet_port 50002" "-s" "D:/Data/Eigene Dokumente/Projekte/EIB/OpenKNX/Debugging_RP2040/openocd/share/openocd/scripts" "-f" "interface/picoprobe.cfg" "-f" "target/rp2040.cfg"
Open On-Chip Debugger 0.10.0+dev-geb22ace (2021-04-12-11:02)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'swd'
Warn : Transport "swd" was already selected
adapter speed: 5000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Info : clock speed 5000 kHz
```

earlephilhower, 2022-02-13:
What version of Windows and x32 or x64 are you using?  And are you a full user (ie. local admin)?

What you ran looks like the 32b version:
> Open On-Chip Debugger 0.11.0-g610f137 (2022-02-11-14:08)

Whereas my run was 64b under Win7x64:
> Open On-Chip Debugger 0.11.0-g610f137 (2022-02-11-13:57)

I don't think you can have a 32b libusb access the 64b drivers.

SirSydom, 2022-02-13:
Win10 Pro x64 12H1 Build 19043.1526
Yes, I'm a local admin.

The openocd.exe is a 32/64 bit binary, right?

I think I have 32b drivers ? the picoprobe is listed in the device manager as "libusb-win32" device.

earlephilhower, 2022-02-13:
Very odd!  For some reason your Arduino IDE is picking the wrong bittedness for the toolchain.  Or, alternatively, this repo and the ESP8266 one I work on have been building Winx64 toolchains which were never used because the Arduino IDE always grabs 32b under Windows.

The libusb-win32.sys driver is whatever bittedness your kernel is running.

The openocd.exe you're running is a PE32 and, yes, will run on x32 or x64, but the ioctls to the 64b libusb.sys are not going to work.   The openocd I ran is a PE64 and only runs on x64 and can speak to the 64b libusb.sys.

Can you manually try replacing the files in your pqt-openocd/.../bin w/the ones in this link's /bin dir for a sanity test:
https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.3.3-a/x86_64-w64-mingw32.openocd-e3428fadb.220212.zip

Also, is GDB now successfully connecting w/whatever version of OCD is running?

earlephilhower, 2022-02-13:
I just found the Digikey app you're using.  I could not agree with him more!  :sob: 
> As with the previous tutorial, building some of the Pico tools on Windows is frustratingly difficult (as of right now).

Also, the file he's built is 64-bit only, so it can talk w/the libusb.sys driver...
````
earle@amd:~/Downloads/1/src$ file openocd.exe 
openocd.exe: PE32+ executable (console) x86-64, for MS Windows
````

SirSydom, 2022-02-13:
I really appreciate your work on this core!
Your support is great.

> Also, is GDB now successfully connecting w/whatever version of OCD is running?
yes, it is working with the digikey-binary of openocd, and with youre sanity test one.

> Can you manually try replacing the files in your pqt-openocd/.../bin w/the ones in this link's /bin dir for a sanity test:
> https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.3.3-a/x86_64-w64-mingw32.openocd-e3428fadb.220212.zip
yes, just a quick test with my blink app, but this one is working fine! Breakpoints are hitting and also expression like variable names are correctly translated to memory adresses.

earlephilhower, 2022-02-13:
Thanks for the info.  I'm going to cut my losses and just ship the x64 binary in the Win x32 package since a) it seems Arduino is always using x32 no matter what, and b) anyone still using Win x32 in 2022 has other, bigger problems than debugging. One more release, but only 1 tool update.

SirSydom, 2022-02-13:
I don't know if that has anything to do with "Arduino" as I called openocd from cmd.exe (for testing purpose and the outputs quoted here) and from Visual Studio Code with CortexDebug Plugin

earlephilhower, 2022-02-14:
No worries.  It's just a weird Arduino thing.  I've tried on my own Arduino install and it also always gives x32 for Windows.  Only thing is it will make gcc a little slower using x32 instead of x64 (register pressure) in most cases.

I did check and the openOCD installed now via the ide (vs get.py) runs OK as it really is the 64-bit version (you'll see a `build64.sh` script I used to generate it in the install dirs!).

SirSydom, 2022-02-14:
I installed 1.12.0 via board manager, adapted my paths and now its working like a charm! Thank you!

PR #482 (2022-02-12): Update GDB and OpenOCD to work together, finally
GDB for non-Linux systems was built w/o expat which caused odd behavior
under Windows and other systems (i.e. breakpoints not working, etc.)
New toolchain manually builds cross-compiled libexpat and ensures it is
used, fixing the issue.

Windows OpenOCD binaries now come from manually built and tested copies
(using a real Windows system).

Fixes #478
Fixes #457
Fixes #456
and probably others...
commit 1b61a866 (2022-02-12): Update GDB and OpenOCD to work together, finally (#482)

GDB for non-Linux systems was built w/o expat which caused odd behavior
under Windows and other systems (i.e. breakpoints not working, etc.)
New toolchain manually builds cross-compiled libexpat and ensures it is
used, fixing the issue.

Windows OpenOCD binaries now come from manually built and tested copies
(using a real Windows system).

Fixes #478
Fixes #457
Fixes #456
and probably others...