earlephilhower/arduino-pico #501: USB Device not detected by host when using platformIO

NalcoDesigns, 2022-02-23:
**Issue :** 
Device is not detected by Host computer after uploading from platform io

**What I have tried**
- Many of the Adafruit TinyUSB library examples (HID,MIDI,.. none did work)
- Compiling on 3 different computers (didn't help)
- Including libraries or not in the platformio.ini file

### what works :

- When compiled on PIO : The rest of the code seems to run fine, just not the USB part
- **When compiled on Arduino IDE : USB comunication works fine (everything works)**

**Example code :**
```c++
#include <Arduino.h>
#include <Adafruit_TinyUSB.h>
#include <MIDI.h>

Adafruit_USBD_MIDI usb_midi;
MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI_USB);

void setup(void) {
  TinyUSB_Device_Init(0);

  MIDI_USB.begin(MIDI_CHANNEL_OMNI);
  while( !TinyUSBDevice.mounted() ) delay(1); // obviously hangs there if not commented
}

void loop(){
//device should be detected at this point
};
```
**platformio.ini file :**
```ini
[env:pico_W64]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower
board_build.filesystem_size = 0m
platform_packages = 
	maxgerhardt/framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git
	maxgerhardt/toolchain-pico@https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.3.3-a/x86_64-w64-mingw32.arm-none-eabi-ed6d983.220212.zip
build_flags = -DUSE_TINYUSB
lib_deps = 
	adafruit/Adafruit TinyUSB Library@^1.9.4; tried with and without
```

Nalco

NalcoDesigns, 2022-02-24:
I'm adding this, it could be a clue
[image]

earlephilhower, 2022-02-24:
Please make sure you're using the absolute latest version of the Platform.IO and core files.  An earlier version of the build script had it including an out-of-date stdc++ library which caused very strange behavior (things like `std::map<>` never finding any set values, etc.).

NalcoDesigns, 2022-02-25:
It was up to date, but i tried reinstalling it, has the exact same behavior

Platform IO : up to date
Core : up to date 
Pico toolchain : up to date (1.3.3-a)

NalcoDesigns, 2022-02-26:
It seems that USB Serial usb is being forced until i use `build_flags = -DPIO_FRAMEWORK_ARDUINO_NO_USB`

functionpointer, 2022-03-12:
I found that the TinyUSB library has the problem, while the built-in usb works fine.

Platformio for some reason automatically includes the TinyUSB library if you use Wire/I2C or SPI.
You can stop it from doing that by adding
```
lib_ignore = 
    Adafruit TinyUSB Library
```
to the `platformio.ini`

Of course, this workaround is only helpful if you don't need any TinyUSB features.

NalcoDesigns, 2022-03-13:
> I found that the TinyUSB library has the problem, while the built-in usb works fine.
>
> Platformio for some reason automatically includes the TinyUSB library if you use Wire/I2C or SPI. You can stop [...]

Yes, the workaround  works if you dont need tinyusb, but i need usb midi functionality built into Adafruit tinyusb, i will try to make a dirty port on my end but .. i would like a cleaner implementation for sure

godario, 2022-04-21:
Hi! I'm using the tinyUSB lib and I got the same issue. Compiling the code with the Arduino IDE all works fine, while using PIO the debugger never enters inside the `TinyUSB_Device_Init(0);` function and then the USB device is never initialized. Have you managed to solve? Thanks.

NalcoDesigns, 2022-04-21:
The issue is still present, when i look up at `TinyUSB_Device_Init()` it inits a Usb-Serial :
```c++
Adafruit_USBD_Device TinyUSBDevice;

void TinyUSB_Device_Init(uint8_t rhport) {
  // Init USB Device controller and stack
  TinyUSBDevice.begin(rhport);
}
```

```c++
bool Adafruit_USBD_Device::begin(uint8_t rhport) {
...
  SerialTinyUSB.begin(115200);
...
}
```

For me, since i'm doing MIDI - USB, here is working code for Arduino IDE:

```c++
#include <Adafruit_TinyUSB.h>
#include "MIDI.h"
Adafruit_USBD_MIDI usb_midi;
MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI_USB);

void setup(){
    MIDI_USB.begin(MIDI_CHANNEL_OMNI);
}
void loop(){}
```

the `begin()` method ends up calling the right function (not TinyUSB_Device_Init in this case)
### However

I'm still stuck with the MIDI USB not working on platformIO.

when the usb midi begin occurs, it does a `TinyUSBDevice.addInterface()` which return an error (just a false)

I don't know how to debug this further

NalcoDesigns, 2022-04-23:
It seems to be an issue withing the tinyUSB `TinyUSBDevice.addInterface()` method : 

When parsing interface descriptors, nothing happens (it should init TUSB_DESC_INTERFACE and TUSB_DESC_ENDPOINT descriptors)

godario, 2022-04-28:
Hi, I just solved upgrading the Adafruit_TinyUSB_Arduino library to the last version and using the workaround explained [here](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/issues/117).

maxgerhardt, 2022-04-28:
Ah yes, the good old `lib_archive = no` -- I will try to repro + fix. We can add a library.json to the TinyUSB lib that sets this setting equivalently to always work without platformio.ini changes.

JonnyHaystack, 2022-04-28:
> Hi, I just solved upgrading the Adafruit_TinyUSB_Arduino library to the last version and using the workaround explained [here](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/issues/117).

Just to confirm, this workaround seems to be doing the job perfectly for me, for now 😄

maxgerhardt, 2022-04-29:
Have you tried adding additionally
```
lib_deps = https://github.com/adafruit/Adafruit_TinyUSB_Arduino/archive/refs/heads/master.zip
```

Or 
```
build_flags =
   -DCFG_TUD_MIDI=1
   -DCFG_TUD_MIDI_RX_BUFSIZE=128
   -DCFG_TUD_MIDI_TX_BUFSIZE=127
```

NalcoDesigns, 2022-04-29:
Doing First and/or second leads to the same result :
[image]

skybrian, 2022-04-29:
Just to confirm and in case it's helpful to others, with TinyUSB, both serial and MIDI are working for me. Here is my platformio.ini file. (I'm on a Mac.)

```
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower
board_build.filesystem_size = 0m
build_flags = -DUSE_TINYUSB
lib_archive=no
platform_packages =
	maxgerhardt/framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git
	maxgerhardt/toolchain-pico@https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.3.3-a/x86_64-apple-darwin14.arm-none-eabi-ed6d983.220212.tar.gz
lib_deps =
	pfeerick/elapsedMillis@1.0.6
	fortyseveneffects/MIDI Library@5.0.2
```

JonnyHaystack, 2022-04-30:
It seems like I'm getting massive performance issues from having **_any_** USB stack enabled. It is much worse with the Adafruit TinyUSB than with the basic Pico SDK USB stack, though. I was having no problems whatsoever when I had `-DUSE_TINYUSB` but before applying the workarounds from above. The execution time of even very simple low level reads and writes is much longer now than it should be, and it's somewhat inconsistent and is causing some real issues.

Also running these lines on startup makes things even worse yet again:
```
this->usb_hid.setPollInterval(1);
this->usb_hid.setReportDescriptor(desc_hid_report, sizeof(desc_hid_report));
```

This is without even having USB connected, as well. I can see the timing differences very clearly using a logic analyzer. Is there a way to completely disable all the USB functionality at runtime? I tried `Serial.end()`, `tud_disconnect()`, `USBDevice.detach()`, and `noInterrupts()` and none of those helped. I feel like I shouldn't have to do this anyway... I just don't understand why having a USB stack would cause a huge performance impact when USB isn't even connected.

The Pico is being powered via VBUS if that makes any difference.

For context, I am trying to do timing sensitive communication using the RP2040's PIO (programmable IO) and the USB stack is somehow adding massive delays to the system code (not the code running on the state machine of course) which manages the PIO program, and thus causing communication failures and connection dropouts. Something that took 16-18us before is now taking upwards of 66us, and even 100us if the `usb_hid.setPollInterval()` and `usb_hid.setReportDescriptor()` were called.

earlephilhower, 2022-04-30:
The USB stack is IRQ driven at a 1khz timer (plus any IRQs that are caused by the upstream USB host).  Not sure I've seen people have macroscopic performance problems, but even so there's probably a way around it.

Just run your app on core1.  Use `setup1()` and `loop1()` and none of your CPU will go to the USB stack.  (The way the IRQs work on the Pico is that each core has its own vector/enable table and USB is started (and will remain) only on core0).

JonnyHaystack, 2022-05-01:
@earlephilhower I tried running it on core1, but it didn't work at all. Then I disabled the USB stack again and it worked. So basically it was even worse than running it on core0. Besides, I will need core1 for other things later on so that's not a viable workaround. If there was a way to disable the interrupts maybe I could work with that.

This performance impact seems way too severe to me for idling USB anyway. Basic things such as reading GPIO pins are taking at least twice as long as they do with the USB stack disabled. It's as if the entire system has had its clock speed reduced.

earlephilhower, 2022-05-01:
You could be hitting cache spills and/or flash contention due to the large USB code overhead.  Try marking your time-sensitive functions to move them to RAM:

````
void __not_in_flash_func(fn)(p1, p2, ...) { ... }
````

JonnyHaystack, 2022-05-01:
I'll give that a shot, thanks

JonnyHaystack, 2022-05-01:
So far I'm seeing significant improvements from using that! 😄 By the way, would there be any reason for me to use `__no_inline_not_in_flash_func` instead of just `__not_in_flash_func`? I'm not sure what the point of the former is.

earlephilhower, 2022-05-01:
I haven't had to look into it, but I suppose it just also as the `<noinline` attribute too.  In some older GCCs the compiler would inline functions that had the segment attribute set, causing the section attribute to get lost.

earlephilhower, 2022-05-01:
Also, you can book hoo to 133mhz (in spec) or higher through the menu to hit timing targets

JonnyHaystack, 2022-05-01:
Even after moving all timing sensitive functions to RAM, although it has improved things, it's still not enough. Is there really no way I can just disable the USB stack at runtime rather than compile time? Like disable the interrupts/interrupt timer?

maxgerhardt, 2022-05-01:
Since it's an ARM chip you should be able to tell your NVIC to disable the timer and USB interrupts [here](https://github.com/raspberrypi/pico-sdk/blob/bfcbefafc5d2a210551a4d9d80b4303d4ae0adf7/src/rp2_common/cmsis/stub/CMSIS/Device/RaspberryPi/RP2040/Include/RP2040.h#L40-L77) (`core_cm0plus.h`). 

```cpp
NVIC_DisableIRQ(USBCTRL_IRQ_IRQn);
```

plus whatever you need to do do remove the timer alarm in [here](https://github.com/earlephilhower/arduino-pico/blob/bda630e419cb221cb48131ff325aa1716afe1df9/cores/rp2040/RP2040USB.cpp#L302-L305). (Multiple alarms might be multiplexed on one Timer?)

But I think as soon as you disable those IRQs, the computer will loose the USB connection after a short period of time.

JonnyHaystack, 2022-05-01:
Thanks Max, I will try that. Yeah, it's not even connected to a computer so I don't want it to be wasting time on USB interrupts while it's plugged into something else and it doesn't need to maintain any USB connection. It will be plugged into either a PC or another device for the whole runtime, but not both at the same time.

earlephilhower, 2022-05-01:
There's `-DNO_USB` too which should completely remote USB support code.

JonnyHaystack, 2022-05-01:
> There's `-DNO_USB` too which should completely remote USB support code.

Yeah, disabling the USB stack is no problem. It's just that I needed to be able to disable it at runtime. Based on Max's advice above, I tried `irq_set_enabled(31, false)` and that seems to have done the trick. `NVIC_DisableIRQ(USBCTRL_IRQ_IRQn)` on its own didn't seem to help. While this solution works for me, it feels a bit hacky, and it would be nice if an API was provided to enable/disable the USB stack at runtime.

On a related note, it would be really nice to be able to use tinyusb directly as one would with the Pico SDK (calling `tud_task()` manually etc), rather than having to go through the Adafruit library (which notably doesn't have host support). Not sure if that's currently possible.

Also still very confused why running my time sensitive code on core1 didn't work either unless I disabled USB, but that's not blocking me for now.

Anyway, sorry if I've somewhat derailed this issue. I initially thought the performance problem might also be unique to PlatformIO because it is very noticeable.

NalcoDesigns, 2022-05-02:
> Have you tried adding additionally
> 
> ```
> lib_deps = https://github.com/adafruit/Adafruit_TinyUSB_Arduino/archive/refs/heads/master.zip
> ```
> 
> Or
> 
> ```
> build_flags =
>    -DCFG_TUD_MIDI=1
>    -DCFG_TUD_MIDI_RX_BUFSIZE=128
>    -DCFG_TUD_MIDI_TX_BUFSIZE=127
> ```

This fixes what's adressed by the issue,
Thank you for your help

closed 2022-05-02 by NalcoDesigns

maxgerhardt, 2022-05-02:
(I will try and engineer a fix that enables lib_archive = no for the Adafruit TinyUSB library soon so that ne special things need to be done -- along with a version push of the library itself if needed)

JonnyHaystack, 2022-05-17:
Just to follow up on my performance issue in case anyone else is having it, all I had to do was add `-O3` to build_flags. Performance instantly boosted by more than 10x and is about the same as raw Pico SDK now. Should've thought to try that earlier, but I completely forgot PlatformIO seems to not have those optimizations by default even though I had come across this problem before on AVR...