earlephilhower/arduino-pico #504: Serial1 on QT Py RP2040 not working

iKK001, 2022-03-03:
Not sure if I'm missing something, but I thought that the following code should write to the RX/TX UART of the `QT PY RP2040`:

```
void setup() {
    Serial1.begin(115200);
}

void loop() {
    Serial1.write("Hello World");
    delay(2000);
}
```

But it doesn't.

Then I tried to add the pins as follows:

```
    Serial1.setTX(6);   
    Serial1.setRX(7);     
    Serial1.begin(115200);
```

But this did not help.

Any idea what I am doing wrong ?

earlephilhower, 2022-03-03:
I don't have one, but according to the variant that Adafruit wrote:

https://github.com/earlephilhower/arduino-pico/blob/be21c31eab6dae50c0aa1860d795dfef7b444ad8/variants/adafruit_qtpy/pins_arduino.h#L13-L19

Serial1(UART0) comes out on *RP2040* pins 20 and 5.  You'll need to look at the schematic to figure out the actual PCB pin that's connected to.

iKK001, 2022-03-05:
Thank you Earle, it does not yet work - see my code here:

earlephilhower, 2022-03-05:
I think the board `pins.h` is wrong for your port.  Looking at the Fritzing from Adafruit:
[image]
those pins should be `Serial2` not `Serial1` (i.e. RX0 == UART0 == Serial1, RX1 == UART1 == Serial*2*).

I'd start by swapping `Serial2` for `Serial1` in your script.  That should work.  And then, please do a PR on the pins.txt and swap the serial1/2  pins...

iKK001, 2022-03-05:
Excellent, Earle - thank you so much. This was it.

Isn't the Adafruit pinout schematic confusing ? (i.e. with too many RX0/1's and TX0/1's in several colors making it diffucult to guess)...

I  tried to do a PR on the `pins_arduino.h` file.

However I do not have permission to push to your repo, for some reason:

```
remote: Permission to earlephilhower/arduino-pico.git denied to iKK001.
fatal: unable to access 'https://github.com/earlephilhower/arduino-pico.git/': The requested URL returned error: 403
```

Can you please give me permission to push to your repo ?
(my github handle:  iKK001)

----------------------------------------

And I realized the following:

BOTH Serials are existant on the `QT Py RP2040` - i.e. `Serial1` and `Serial2`.

Here are the two working examples:

Example with `Serial1`:

```
void setup() {
    Serial1.setTX(28u);   
    Serial1.setRX(29u);     
    Serial1.begin(115200);
}

void loop() {
    Serial1.write("Hello World 1");
    delay(2000);
}
```

Example with `Serial2`:

```
void setup() {
    Serial2.setTX(20u);   
    Serial2.setRX(5u);     
    Serial2.begin(115200);
}

void loop() {
    Serial2.write("Hello World 2");
    delay(2000);
}
```

My PR would consider both pinouts - if easier, you can fit it in yourself also - however you prefer.

```
// Serial1
#define PIN_SERIAL1_TX (28u)
#define PIN_SERIAL1_RX (29u)

// Serial2
#define PIN_SERIAL2_TX (20u)
#define PIN_SERIAL2_RX (5u)
```

earlephilhower, 2022-03-05:
> Can you please give me permission to push to your repo ?

The way this is done is you'd fork the repo to your own copy, clone that one locally (i.e. git clone https://github.com/<yourid>/arduino-pico), checkout a new branch (git checkout -b pr1), edit and push to your own repo.  You'd then be able to do a Pull Reguest when you opened this repo's page.

I'll just mod it myself since it's so minor, no problem.

closed 2022-03-05 by earlephilhower

iKK001, 2022-03-05:
Thank you Earle ! It's a joy to use your library. Thank you very much.

PR #505 (2022-03-05): Fix QT Py serial (Serial2 pinned out, not Serial1)
Fixes #504