earlephilhower/arduino-pico #508: Build failed

metanav, 2022-03-08:
When I am building using ./make-libpico.sh it has some errors at the end:

[100%] Linking CXX static library libpico.a
ar: stdio.c.obj: not found in archive
ar: stdio_usb.c.obj: not found in archive
ar: stdio_usb_descriptors.c.obj: not found in archive
make[2]: *** [libpico.a] Error 1
make[2]: *** Deleting file `libpico.a'
make[1]: *** [CMakeFiles/pico.dir/all] Error 2
make: *** [all] Error 2

earlephilhower, 2022-03-08:
Those files are not part of the build in the repo, and are incompatible with the core.  Did you change the `CMakeLists.txt` file?

earlephilhower, 2022-03-12:
Closing since no updates from the user.

closed 2022-03-12 by earlephilhower

krishnaguy, 2024-12-28:
I tried to build the library using ./make-libpico.sh today and had the same errors. I did not change anything in the CMakeLists.txt file.

earlephilhower, 2024-12-29:
Libpico is built at part of CI and not showing this nor are my own local builds.  https://github.com/earlephilhower/arduino-pico/actions/runs/12404721175

What OS, release, etc. are you running?  Can you attach full logs?

krishnaguy, 2024-12-29:
I am on an Apple M2 on Mac OS Sequoia 15.1.1

Thank you for the help!

I just followed the instructions in this repo, cloning it, updating the submodules etc first, running python3 get.py and then running the script.

This is the log, please let me know if you prefer a different formating. 

```
krishnavenkatraman@Krishnas-MacBook-Air libpico % ./make-libpico.sh 
++ cd ../../pico-sdk/
++ pwd
+ export PICO_SDK_PATH=/Users/krishnavenkatraman/development/robots/arduino-pico/pico-sdk
+ PICO_SDK_PATH=/Users/krishnavenkatraman/development/robots/arduino-pico/pico-sdk
++ cd ../../system/arm-none-eabi/bin
++ pwd
+ export PATH=/Users/krishnavenkatraman/development/robots/arduino-pico/system/arm-none-eabi/bin:/opt/homebrew/opt/openjdk@17/bin:/opt/homebrew/Cellar/pyenv-virtualenv/1.2.4/shims:/Users/krishnavenkatraman/.nvm/versions/node/v21.5.0/bin:/Users/krishnavenkatraman/.local/bin:/Users/krishnavenkatraman/.local/bin:/Users/krishnavenkatraman/.pyenv/shims:/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/nix/var/nix/profiles/default/bin/
+ PATH=/Users/krishnavenkatraman/development/robots/arduino-pico/system/arm-none-eabi/bin:/opt/homebrew/opt/openjdk@17/bin:/opt/homebrew/Cellar/pyenv-virtualenv/1.2.4/shims:/Users/krishnavenkatraman/.nvm/versions/node/v21.5.0/bin:/Users/krishnavenkatraman/.local/bin:/Users/krishnavenkatraman/.local/bin:/Users/krishnavenkatraman/.pyenv/shims:/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/nix/var/nix/profiles/default/bin/
++ cd ../../system/riscv32-unknown-elf/bin
++ pwd
+ export PATH=/Users/krishnavenkatraman/development/robots/arduino-pico/system/riscv32-unknown-elf/bin:/Users/krishnavenkatraman/development/robots/arduino-pico/system/arm-none-eabi/bin:/opt/homebrew/opt/openjdk@17/bin:/opt/homebrew/Cellar/pyenv-virtualenv/1.2.4/shims:/Users/krishnavenkatraman/.nvm/versions/node/v21.5.0/bin:/Users/krishnavenkatraman/.local/bin:/Users/krishnavenkatraman/.local/bin:/Users/krishnavenkatraman/.pyenv/shims:/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/nix/var/nix/profiles/default/bin/
+ PATH=/Users/krishnavenkatraman/development/robots/arduino-pico/system/riscv32-unknown-elf/bin:/Users/krishnavenkatraman/development/robots/arduino-pico/system/arm-none-eabi/bin:/opt/homebrew/opt/openjdk@17/bin:/opt/homebrew/Cellar/pyenv-virtualenv/1.2.4/shims:/Users/krishnavenkatraman/.nvm/versions/node/v21.5.0/bin:/Users/krishnavenkatraman/.local/bin:/Users/krishnavenkatraman/.local/bin:/Users/krishnavenkatraman/.pyenv/shims:/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/nix/var/nix/profiles/default/bin/
++ cd ../../system/picotool
++ pwd
+ export PATH=/Users/krishnavenkatraman/development/robots/arduino-pico/system/picotool:/Users/krishnavenkatraman/development/robots/arduino-pico/system/riscv32-unknown-elf/bin:/Users/krishnavenkatraman/development/robots/arduino-pico/system/arm-none-eabi/bin:/opt/homebrew/opt/openjdk@17/bin:/opt/homebrew/Cellar/pyenv-virtualenv/1.2.4/shims:/Users/krishnavenkatraman/.nvm/versions/node/v21.5.0/bin:/Users/krishnavenkatraman/.local/bin:/Users/krishnavenkatraman/.local/bin:/Users/krishnavenkatraman/.pyenv/shims:/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/nix/var/nix/profiles/default/bin/
+ PATH=/Users/krishnavenkatraman/development/robots/arduino-pico/system/picotool:/Users/krishnavenkatraman/development/robots/arduino-pico/system/riscv32-unknown-elf/bin:/Users/krishnavenkatraman/development/robots/arduino-pico/system/arm-none-eabi/bin:/opt/homebrew/opt/openjdk@17/bin:/opt/homebrew/Cellar/pyenv-virtualenv/1.2.4/shims:/Users/krishnavenkatraman/.nvm/versions/node/v21.5.0/bin:/Users/krishnavenkatraman/.local/bin:/Users/krishnavenkatraman/.local/bin:/Users/krishnavenkatraman/.pyenv/shims:/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/nix/var/nix/profiles/default/bin/
+ rm -rf build-rp2040
+ mkdir build-rp2040
+ cd build-rp2040
+ CPU=rp2040
+ cmake ..
Building for CPU rp2040
Using PICO_SDK_PATH from environment ('/Users/krishnavenkatraman/development/robots/arduino-pico/pico-sdk')
PICO_SDK_PATH is /Users/krishnavenkatraman/development/robots/arduino-pico/pico-sdk
Target board (PICO_BOARD) is 'pico_w'.
Using board configuration from /Users/krishnavenkatraman/development/robots/arduino-pico/pico-sdk/src/boards/include/boards/pico_w.h
Pico Platform (PICO_PLATFORM) is 'rp2040'.
-- Defaulting build type to 'Release' since not specified.
Defaulting compiler (PICO_COMPILER) to 'pico_arm_cortex_m0plus_gcc' since not specified.
Configuring toolchain based on PICO_COMPILER 'pico_arm_cortex_m0plus_gcc'
Defaulting PICO_GCC_TRIPLE to 'arm-none-eabi'
-- The C compiler identification is GNU 14.2.0
-- The CXX compiler identification is GNU 14.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/krishnavenkatraman/development/robots/arduino-pico/system/arm-none-eabi/bin/arm-none-eabi-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/krishnavenkatraman/development/robots/arduino-pico/system/arm-none-eabi/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
[... cut ...]
ar: pico_malloc.c.obj: not found in archive
ar: newlib_interface.c.obj: not found in archive
make[2]: *** [libpico-rp2040.a] Error 1
make[2]: *** Deleting file `libpico-rp2040.a'
make[1]: *** [CMakeFiles/pico-rp2040.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[100%] Built target pioasm
[  6%] Performing install step for 'pioasmBuild'
[100%] Built target pioasm
Install the project...
-- Install configuration: "Release"
[  6%] Completed 'pioasmBuild'
[  6%] Built target pioasmBuild
make: *** [all] Error 2

```

earlephilhower, 2024-12-29:
I think this is BSD `ar` vs Linux/GNU `ar` and trivially fixable.

See #2721 , just make sure we use GNU `ar`.

Can you give that PR a try and report back?

krishnaguy, 2024-12-29:
I tried the fix and it ran successfully, thanks so much!

PR #2721 (2024-12-29): Use the distributed ARM ar in libpico, not system
Fixes #508
commit bb682bbb (2024-12-29): Use the distributed ARM ar in libpico, not system (#2721)

Fixes #508