earlephilhower/arduino-pico #509: Board Definition

charles-d-burton, 2022-03-10:
Excuse my newness to all of this but I'm trying to update the RF69HCW module to support the Raspberry Pi Pico and I can't figure out where I'm supposed to find the definitions.  I've tried using the `boards.txt` file definition but that doesn't seem to work and I've also tried setting it to `RP2040` but that also doesn't seem to work.  For reference this is the line I'm trying to update:

https://github.com/charles-d-burton/RFM69/blob/2006deb76454dee5facc2241721d3434f0812b73/RFM69.h#L60

PontusO, 2022-03-10:
You should not have to change anything in the environment.

Simply instantiate the class using the following constructor and you should 
be good to go.

RFM69(uint8_t slaveSelectPin=RF69_SPI_CS, uint8_t 
interruptPin=RF69_IRQ_PIN, bool isRFM69HW_HCW=false, SPIClass *spi=nullptr);

Make sure you identify the correct SPI pins as well as the cs and irq pins 
when you connect your module.

charles-d-burton, 2022-03-11:
You mean rather than update the library just call the constructor directly?

PontusO, 2022-03-11:
You wouldn't call the constructor directly, it gets called when the class is instantiated.

For instance
`RFM69 rfm69(slave_select_pin, irq_pin, false, SPI);`
Would create rfm69 using the pins defined in slave_select_pin and irq_pin and the default SPI interface.

charles-d-burton, 2022-03-12:
Hmm, doesn't seem to like that.  The implementation of this library is different than what the other library is expecting.
```
/home/charles/Arduino/pico-test/pico-test.ino:25:34: error: no matching function for call to 'RFM69::RFM69(int, int, bool, SPIClassRP2040&)'
   25 | RFM69 radio(CSPIN,IRQPIN,true,SPI);
      |                                  ^
 ```
 
 Did a bunch of searching around and I'm finding precious little about how to coerce the types into the correct format.

charles-d-burton, 2022-03-12:
I did the definition like this
```
#define CSPIN         17
#define IRQPIN        1
// Create a library object for our RFM69HCW module:
RFM69 radio(CSPIN,IRQPIN,true,SPI);```

charles-d-burton, 2022-03-12:
Alright, got it to compile with this:
```
#define CSPIN         17
#define IRQPIN        1

// Create a library object for our RFM69HCW module:
RFM69 radio(CSPIN,IRQPIN,true,&SPI);```

Which is great but that didn't get me any further.  Still getting an RSSI of zero.

charles-d-burton, 2022-03-12:
For reference here's the whole thing
```cpp
// LowPowerLabs RFM69 Library
#include <RFM69.h>
// SPI Library
#include <SPI.h>
// Addresses for this node. CHANGE THESE FOR EACH NODE!

#define NETWORKID     0   // Must be the same for all nodes
#define MYNODEID      2   // My node ID
#define TONODEID      1   // Destination node ID

#define FREQUENCY     RF69_915MHZ

// AES encryption (or not):

#define ENCRYPT       true // Set to "true" to use encryption
#define ENCRYPTKEY    "passwordpassword" // Use the same 16-byte key on all nodes

// Use ACKnowledge when sending messages (or not):
#define USEACK        true // Request ACKs or not

#define CSPIN         17
#define IRQPIN        1

// Create a library object for our RFM69HCW module:
RFM69 radio(CSPIN,IRQPIN,true,&SPI);

void setup() {
  SPI.begin(); 
    // put your setup code here, to run once:
  Serial.begin(115200);
   // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.print("Node ");
  Serial.print(MYNODEID,DEC);
  Serial.println(" ready");
  // Initialize the RFM69HCW:
  radio.initialize(FREQUENCY, MYNODEID, NETWORKID);
  radio.setHighPower(); // Always use this for RFM69HCW

  // Turn on encryption if desired:

  if (ENCRYPT) {
    radio.encrypt(ENCRYPTKEY);
  }

  delay(1000);
}

void loop() {
  Serial.println(radio.canSend());
  Serial.print("RSSI: ");
  Serial.println(radio.RSSI);
  #if defined(RP2040)
  Serial.println("pico defined");
  #endif
  delay(500);
}
```

charles-d-burton, 2022-03-12:
Ok, progress.  Now I'm getting somewhere but for some reason it's hanging.  When I upload this sketch the serial console never comes back.  I think it doesn't like something about my SPI setup.

```cpp
// LowPowerLabs RFM69 Library
#include <RFM69.h>
// SPI Library
#include <SPI.h>
// Addresses for this node. CHANGE THESE FOR EACH NODE!

#define NETWORKID     0   // Must be the same for all nodes
#define MYNODEID      2   // My node ID
#define TONODEID      1   // Destination node ID

#define FREQUENCY     RF69_915MHZ

// AES encryption (or not):

#define ENCRYPT       true // Set to "true" to use encryption
#define ENCRYPTKEY    "passwordpassword" // Use the same 16-byte key on all nodes

// Use ACKnowledge when sending messages (or not):
#define USEACK        true // Request ACKs or not

#define CSPIN         17
#define IRQPIN        1

// Create a library object for our RFM69HCW module:
RFM69 radio(CSPIN,IRQPIN,true,&SPI);

void setup() {
  SPI.setRX(16);
  SPI.setTX(19);
  SPI.setCS(17);
  SPI.setSCK(18);
  SPI.begin(); 
    // put your setup code here, to run once:
  Serial.begin(115200);
   // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.print("Node ");
  Serial.print(MYNODEID,DEC);
  Serial.println(" ready");
  // Initialize the RFM69HCW:
  radio.initialize(FREQUENCY, MYNODEID, NETWORKID);
  radio.setHighPower(); // Always use this for RFM69HCW

  // Turn on encryption if desired:

  if (ENCRYPT) {
    radio.encrypt(ENCRYPTKEY);
  }

  delay(1000);
}

void loop() {
  //Serial.println(radio.canSend());
  Serial.print("RSSI: ");
  Serial.println(radio.readRSSI());
  delay(500);
}
```

PontusO, 2022-03-12:
Try putting a `while (!Serial) delay(10);` after the Serial.begin statement. This will ensure the USB link is up before printing to it.

charles-d-burton, 2022-03-12:
Awesome thanks, that definitely got the Serial connection working again.  Still nothing from the chip though.

charles-d-burton, 2022-03-12:
Forcing the IRQ doesn't do it either
```cpp

// LowPowerLabs RFM69 Library
#include <RFM69.h>
// SPI Library
#include <SPI.h>
// Addresses for this node. CHANGE THESE FOR EACH NODE!
volatile byte state = LOW;

#define NETWORKID     0   // Must be the same for all nodes
#define MYNODEID      2   // My node ID
#define TONODEID      1   // Destination node ID

#define FREQUENCY     RF69_915MHZ

// AES encryption (or not):

#define ENCRYPT       true // Set to "true" to use encryption
#define ENCRYPTKEY    "passwordpassword" // Use the same 16-byte key on all nodes

// Use ACKnowledge when sending messages (or not):
#define USEACK        true // Request ACKs or not

#define CSPIN         17
#define IRQPIN        1

// Create a library object for our RFM69HCW module:
RFM69 radio(CSPIN,IRQPIN,true,&SPI);

void setup() {
  SPI.setRX(16);
  SPI.setTX(19);
  SPI.setCS(17);
  SPI.setSCK(18);  
  SPI.begin(); 
    // put your setup code here, to run once:
  Serial.begin(115200);
  while (!Serial)delay(10);
  pinMode(IRQPIN, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(IRQPIN), blink, CHANGE);
   // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.print("Node ");
  Serial.print(MYNODEID,DEC);
  Serial.println(" ready");
  // Initialize the RFM69HCW:
  radio.initialize(FREQUENCY, MYNODEID, NETWORKID);
  radio.setHighPower(); // Always use this for RFM69HCW

  // Turn on encryption if desired:

  if (ENCRYPT) {
    radio.encrypt(ENCRYPTKEY);
  }

  delay(1000);
}

void loop() {
  Serial.println(radio.canSend());
  Serial.print("RSSI: ");
  Serial.println(radio.readRSSI());
  Serial.print("Temp: ");
  Serial.println(radio.readTemperature());
  digitalWrite(LED_BUILTIN, state);
  blink();
  delay(500);
}

void blink() {
  state = !state;
}

```

PontusO, 2022-03-12:
At this point I think you will get better support at the LowPowerLab github. Maybe there are some debug logs that can be enabled that will shed some light on your problems.

earlephilhower, 2022-03-12:
Thanks, @PontusO , for the assist.  I'll close this issue.   @charles-d-burton can refer to it when/if he opens one in the library repo.

closed 2022-03-12 by earlephilhower