earlephilhower/arduino-pico #510: TinyUSB library - Examples: msc_sdfat (and msc_sd) compiler error

madias123, 2022-03-11:
Hello,
I do not know who is responsable for this error (this repo or adafruit):
The example: TinyUSB -> mass storage -> msc_sdfat compiles with following error:
'SdCard' {aka 'class DedicatedSpiCard'} has no member named 'readBlocks'
It seems, that   the latest greiman sdfat library do not have implemented "readBlocks" and "writeBlocks" anymore. 
Sadly even the example "msc_sd" compiles with error: 'Sd2Card' does not name a type; did you mean 'SdCard'?
It looks like both examples are totally outdated?
Tried out with Arduino IDE 1.8.18

hathach, 2022-03-12:
you will need to install `SdFat (Adafruit's fork)` of the library https://github.com/adafruit/SdFat

earlephilhower, 2022-03-12:
Thanks, @hathach !  Closing.

closed 2022-03-12 by earlephilhower

madias123, 2022-03-12:
Hello,
thank you for your super fast reply! Now it works.
Dunno...I dream of a time only greiman sdfat library will still exits in Arduino land ;)