earlephilhower/arduino-pico #514: Is there a watchdog on the Adafruit QTPy RP2040 ?

iKK001, 2022-03-15:
Using the `Adafruit QTPy RP2040`, I can't get the following 2 watchdog possibilities to work:

Possibility 1 (normally runs on `Arduino RP2040 Connect`):
```
#include <mbed.h>
mbed::Watchdog &watchdog = mbed::Watchdog::get_instance();
```

Possibility 2 (normally runs on any Adafruit board):
```
#include <Adafruit_SleepyDog.h>
```

Is there another way to get a watchdog function on the `Adafruit QTPy RP2040` using your `arduino-pico` library ?

earlephilhower, 2022-03-15:
This core doesn't use the MBED OS, so your first example can't work.  The Adafruit library doesn't support RP2040, so the second is out.

You can use the Pico SDK native calls.  https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__watchdog.html  If they're not in the include path, please let me know and I can add them.

Good luck!

closed 2022-03-15 by earlephilhower

iKK001, 2022-03-15:
Wonderful - thank you very much Earle. I appreciate your support and ideas.

Here my working solution on the `Adafruit QTPy RP2040`:

i.e. In this example, the watchdog is updated every 5 sec - and the controller gets a reset if the watchdog is not updated within 30 sec.

```
#include <stdio.h>
#include "pico/stdlib.h"
#include "hardware/watchdog.h"

int PING_DELAY = 5000;

uint16_t nowy;
uint16_t watchdog_kicking_time;

void setup() {

    // set watchdog to 30 sec    
    watchdog_enable(30000, 1); // second arg is pause on debug which means the watchdog will pause when stepping through code
    watchdog_kicking_time = millis();

    // check if code was rebooted by watchdog or clean-boot    
    if (watchdog_caused_reboot()) {
        // rebooted by watchdog        
    } else {
        // clean boot
    }
}

void loop() {

    nowy = millis();
    
    // update watchdog
    if (nowy - watchdog_kicking_time > PING_DELAY) {
        watchdog_kicking_time = nowy;
        watchdog_update();
    }

    delay(10);
}