earlephilhower/arduino-pico #518: Arduino Alt-F8 gives error

Gavin-Perry, 2022-03-19:
Using Arduino IDE 2.0.0-rc5 when I press F8 to see errors it gives

In included file: "Compiler generates FPU instructions for a device without an FPU (check __FPU_PRESENT)"clang(pp_hash_error)
[core_cm0plus.h(90, 6): ]()Error occurred here

For the included file <Arduino.h>. And yet actually compiling doesn't give an error here!

earlephilhower, 2022-03-20:
I’m traveling so can’t try anything for a week or so, but can you explain what ALT-F8 is supposed to do in the beta IDE?  That’s a GCC (I think) internal header file and not part of the core, so I’m not sure exactly what they’re checking…

jandrassy, 2022-03-20:
I guess it is the presentation compiler

Gavin-Perry, 2022-03-21:
So that's an IDE bug I guess. How do I report that? I couldn't find the place where they take comments on RC releases.
I think I did see it once when installing but now I can't find it.

Gavin-Perry, 2022-03-21:
Speaking of the presentation compiler, it sometimes works OK giving you
help when hovering over a function.
At other times its busy indexing... compiling... (bottom left corner) and
just says loading... at the hover point for FAR too long.
It seems to restart after every character typed or cursor move.