earlephilhower/arduino-pico #520: Uploading to Raspberry Pi Pico from platformio gives error 1

Sciman101, 2022-03-20:
I'm trying to move my development from the Arduino IDE, which has worked fine with arduino-pico for me, to platformio, and followed the example platformio.ini given on the docs to try and upload a simple blink sketch, no dependencies. Every time I upload, it gets to this point

```
Use manually specified: COM9
Forcing reset using 1200bps open/close on port COM9
Uploading .pio\build\pico\firmware.elf
rp2040load 1.0.1 - compiled with go1.15.8
.....................
*** [upload] Error 1
```

and fails. There's no obvious error in the rest of the log, and in fact the code itself compiles fine. When I manually copy the compiled firmware.uf2 to the pico, it loads the blink sketch without issue. I've tried uploading in and out of BOOTSEL mode, though I'd previously uploaded arduino-pico code to this device so I'd assume I wouldn't need to use BOOTSEL. I've reinstalled my platformio extension, ran in verbose mode, used automatic port finding and manual port config, and force disabled USB mode, and nothing seems to work.

Has anyone else ran into this issue, and is there something I'm missing? My platformio.ini is exactly the same as the one listed [here](https://arduino-pico.readthedocs.io/en/latest/platformio.html)

Gavin-Perry, 2022-03-21:
Looks like your baud rate is far to low. Also be sure it matches at both
ends.

earlephilhower, 2022-03-21:
the 1200baud is correct.  It is just what is used to reset the chip as a doorbell knock.  There is no serial upload on this chip, just UF2 copy.

Sciman101, 2022-03-21:
So, do I just need to manually copy the file each time I compile?

Sciman101, 2022-03-22:
Tried that just now, doesn't seem to work - are the WinUSB drivers installed automatically or do I need to do that on my own?

maxgerhardt, 2022-03-22:
If you at least installed the Arduino-pico core in the Arduino IDE, they are installed. Otherwise, plug in your Pico in bootloader mode (BOOTSEL button pressed then plugged in) and use Zadig (https://zadig.akeo.ie/) to load WinUSB drivers for the "RP2 Boot2 (Interface 1)" device.

earlephilhower, 2022-03-22:
Is anything at all needed for UF2 upload, though?  In bootloader mode it presents itself as a MSD (mass storage device, aka USB stick) which should "just work," no?

maxgerhardt, 2022-03-22:
>In bootloader mode 

Exactly, in bootloader mode. To get back in bootloader mode, PlatformIO and the core uses a 1200bps UART reset mechanism, and after that, PlatformIO doesn't use the mass storage device, but instructs `picotool` to load the binary onto it, which requires WinUSB or libusb-win32 drivers to be loaded for it.

Sciman101, 2022-03-24:
Hm, I flashed a blink sketch with the Arduino IDE, then tried flashing the same sketch with different timing from platformio. Same error as before, it successfully autodetected the port and managed to enter bootloader mode so tbh I'm not sure what's going on

maxgerhardt, 2022-03-24:
And after PlatformIO has reset the device in the bootloader, Zadig shows that the RP2 Boot2 Interace 1 device has still WinUSB drivers loaded?

Sciman101, 2022-03-24:
I haven't tried Zadig yet, gonna give that a go now

Sciman101, 2022-03-24:
Should have just tried that from the start, works perfectly now. Thanks for the help!

closed 2022-03-24 by Sciman101