earlephilhower/arduino-pico #522: Please document the options on the Boot Stage 2 menu

technoblogy, 2022-03-23:
Would it be possible to add a bit of documentation for the **Boot Stage 2** menu on the **Generic RP2040** variant?

[image]

A good place would be here:

https://arduino-pico.readthedocs.io/en/latest/ide.html

I can see that the **Raspberry Pi Pico** variant uses the **W25Q080 QSPI /2** option, but what does "W25Q080" mean? The actual part used is W25Q16, and the recommended maximum size part is W25Q128, but there's no part called W25Q080 as far as I can see.

I assume '/2' and '/4' are the division of the clock speed? So when overclocking to 150MHz, which is beyond the rated W25Q16 flash speed of 133MHz, should one choose '/4'?

The W25X10CL is rated at 104MHz so are those options slower?

Thanks, David

Gavin-Perry, 2022-03-24:
I'm wondering too. I notice Generic SPI /2 is selected and don't dare change it.
There are other options when PICO is the selected board. Are they documented?

earlephilhower, 2022-03-25:
I’m still traveling so can’t go into detail, but changing the option you use won’t damage the chip.  It may only make it unbootable so you need to use the BOOTSEL method to upload a different selection.

The names and settings come directly from the Pico SDK so I can’t comment on the names chosen.  Basically for every flash chip the support a standard 1bit SPI mode at some frequency, but to get to the faster 4bit QSPI mode they all require different registers writes and bus timings. Hence the different boot2s.

PR #546 (2022-04-24): Document the Generic RP2040 option
Fixes #522
commit d1f9bce0 (2022-04-24): Document the Generic RP2040 option (#546)

Fixes #522
closed 2022-04-24 by earlephilhower