earlephilhower/arduino-pico #526: Pad drive strength

jlbirccyn, 2022-04-03:
Hello,

Thanks for this great work!

I looked at the code a bit and did some testing and the drive strength is the default value at reset (4mA), which is pretty low. Would it be possible to set it to 12mA in pinMode? Otherwise we can call the function of the sdk(gpio_set_drive_strength) but it is not the usual simplicity of the Arduino.

Kind regards.

earlephilhower, 2022-04-03:
I'm loathe to change the default value simply because the core is being used by folks for embedded, battery-powered apps.  Changing to a higher drive strength could change the behavior of their boards and reduce battery life.

Any ideas as to a way of allowing folks to override the default?

Gavin-Perry, 2022-04-03:
I'm happy with gpio_set_drive_strength() 
It clearly says what it does and does the job.

jlbirccyn, 2022-04-03:
I understand

Maybe add 4 constants to use with pinMode : OUTPUT_2, OUTPUT_4, OUTPUT_8 and OUTPUT_12? OUTPUT would default to OUTPUT4 for compatibility.

earlephilhower, 2022-04-07:
Interesting idea, especially since it seems the RP2040 can individually set the drive strengths on a per-pad basis.  Would you like to so a Pull Request?  This should be a pretty simple addition...

jlbirccyn, 2022-04-08:
Of course I do. I'll handle it.

earlephilhower, 2022-04-10:
Fixed by #532

closed 2022-04-10 by earlephilhower