earlephilhower/arduino-pico #527: excessive downloads of git repos when using "pio run"

skybrian, 2022-04-05:
I'm attempting to use platformio to build a small project that works fine in Arduino. Here is the ini file:

```
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower

; Flash Size: 2MB (No FS)
board_build.filesystem_size = 0m

; Adafruit TinyUSB
;build_flags = -DUSE_TINYUSB

platform_packages =
    maxgerhardt/framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git
    maxgerhardt/toolchain-pico@https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.3.3-a/x86_64-apple-darwin14.elf2uf2-2062372.220212.tar.gz
```

With VS Code, it hangs with "PlatformIO: Rebuilding IntelliSense Index". I believe that this is spurious; it's doing a lot of work in the background, but without giving the user any feedback.

Instead I tried running 'pio run' from the command line. This downloads many, many git repositories, eventually dying with an error:

```
Submodule path 'pico-sdk/lib/tinyusb/tools/uf2/hidapi': checked out 'a6a622ffb680c55da0de787ff93b80280498330f'
Submodule path 'tools/pyserial': checked out '0e7634747568547b8a7f9fd0c48ed74f16af4b23'
Failed to recurse into submodule path 'pico-extras'
Error: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', 'https://github.com/earlephilhower/arduino-pico.git', '/Users/skybrian/.platformio/.cache/tmp/pkg-installing-ebvkmrds']
```

I tried running "git clone" myself. It seems that the "--depth 1" argument is somewhat misleading (it's about how much history to check out). This will actually check out all submodules recursively.

It seems that today there is a certificate error which causes the git checkout to fail, but not until after it downloaded everything else:

```
Cloning into '/Users/skybrian/projects/midicontroller/tmp/arduino-pico/pico-extras/lib/lwip'...
fatal: unable to access 'https://git.savannah.nongnu.org/git/lwip.git/': SSL certificate problem: certificate has expired
fatal: clone of 'https://git.savannah.nongnu.org/git/lwip.git' into submodule path '/Users/skybrian/projects/midicontroller/tmp/arduino-pico/pico-extras/lib/lwip' failed
Failed to clone 'lib/lwip'. Retry scheduled
```

There is a bug for the certificate error [here](https://savannah.nongnu.org/support/?110546). It seems whether this works or not depends on the OS version?

I'll investigate that, but more generally, I'm wondering if there's some way to cut down on the recursion. The more dependencies included, the longer an initial install takes and the more likely there will be an error. It doesn't seem very friendly for people without good network connections.

Another example of an excessive dependency chain is arduino-pico -> pico-sdk -> tinyusb -> [many irrelevant device drivers](https://github.com/hathach/tinyusb/blob/master/.gitmodules).

skybrian, 2022-04-05:
For the record, the certificate error was due to a Let's Encrypt certificate expiring last year, which, oddly enough, only broke curl on my Mac, not anything else. (Git relies on curl.) The solution was to delete the expired certificate, as described [here](https://stackoverflow.com/questions/69387175/git-for-windows-ssl-certificate-problem-certificate-has-expired/69396425#69396425).

earlephilhower, 2022-04-07:
Closing this one as resolved w/the local cert issue fix.

closed 2022-04-07 by earlephilhower