earlephilhower/arduino-pico #528: "sh: arm-none-eabi-gcc: command not found" when using platformio

skybrian, 2022-04-05:
I'm attempting to using platformio for a project that works in Arduino. After resolving [a previous error](https://github.com/earlephilhower/arduino-pico/issues/527), I deleted ~/.platformio and .pio and started over.

I was able to reinstall PlatformIO itself, and installed the appropriate packages and libraries so that there are no errors in my main.cpp file. However, when I attempt to build, I get:

```
sh: arm-none-eabi-gcc: command not found
```

So it can't find the compiler.

I'm wondering if I can get more information about how it's supposed to work. Where should the compiler be installed when using PlatformIO? How does it get downloaded?

maxgerhardt, 2022-04-28:
This may be due to binary incompatibility. Can you share `file ~/.platformio/packages/toolchain-pico/bin/arm-none-eabi-gcc` and `uname -a` and `lscpu`?

skybrian, 2022-04-28:
It looks like gcc hasn't been downloaded at all. I'm trying to figure out why not.

```
$ ls ~/.platformio/packages/toolchain-pico
elf2uf2		package.json
```

I did add this to platformio.ini:

```
platform_packages =
    maxgerhardt/framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git
    maxgerhardt/toolchain-pico@https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.3.3-a/x86_64-apple-darwin14.elf2uf2-2062372.220212.tar.gz
```

```
$ uname -a
Darwin Macintosh.local 17.7.0 Darwin Kernel Version 17.7.0: Fri Oct 30 13:34:27 PDT 2020; root:xnu-4570.71.82.8~1/RELEASE_X86_64 x86_64
$ lscpu
-bash: lscpu: command not found
```

Yes, it's an old iMac; I can't update to a newer OS. Apparently this is how you get CPU info:
```
$ sysctl -a machdep.cpu
machdep.cpu.max_basic: 11
machdep.cpu.max_ext: 2147483656
machdep.cpu.vendor: GenuineIntel
machdep.cpu.brand_string: Intel(R) Core(TM) i7 CPU         870  @ 2.93GHz
machdep.cpu.family: 6
machdep.cpu.model: 30
machdep.cpu.extmodel: 1
machdep.cpu.extfamily: 0
machdep.cpu.stepping: 5
machdep.cpu.feature_bits: 43034875446164479
machdep.cpu.extfeature_bits: 4967106816
machdep.cpu.signature: 67301
machdep.cpu.brand: 0
machdep.cpu.features: FPU VME DE PSE TSC MSR PAE MCE CX8 APIC SEP MTRR PGE MCA CMOV PAT PSE36 CLFSH DS ACPI MMX FXSR SSE SSE2 SS HTT TM PBE SSE3 DTES64 MON DSCPL VMX SMX EST TM2 SSSE3 CX16 TPR PDCM SSE4.1 SSE4.2 POPCNT
machdep.cpu.extfeatures: SYSCALL XD EM64T LAHF RDTSCP TSCI
machdep.cpu.logical_per_package: 16
machdep.cpu.cores_per_package: 8
machdep.cpu.microcode_version: 10
machdep.cpu.processor_flag: 1
machdep.cpu.mwait.linesize_min: 64
machdep.cpu.mwait.linesize_max: 64
machdep.cpu.mwait.extensions: 3
machdep.cpu.mwait.sub_Cstates: 4384
machdep.cpu.thermal.sensor: 1
machdep.cpu.thermal.dynamic_acceleration: 1
machdep.cpu.thermal.invariant_APIC_timer: 0
machdep.cpu.thermal.thresholds: 2
machdep.cpu.thermal.ACNT_MCNT: 1
machdep.cpu.thermal.core_power_limits: 0
machdep.cpu.thermal.fine_grain_clock_mod: 0
machdep.cpu.thermal.package_thermal_intr: 0
machdep.cpu.thermal.hardware_feedback: 0
machdep.cpu.thermal.energy_policy: 0
machdep.cpu.arch_perf.version: 3
machdep.cpu.arch_perf.number: 4
machdep.cpu.arch_perf.width: 48
machdep.cpu.arch_perf.events_number: 7
machdep.cpu.arch_perf.events: 68
machdep.cpu.arch_perf.fixed_number: 3
machdep.cpu.arch_perf.fixed_width: 48
machdep.cpu.cache.linesize: 64
machdep.cpu.cache.L2_associativity: 8
machdep.cpu.cache.size: 256
machdep.cpu.tlb.inst.small: 64
machdep.cpu.tlb.inst.large: 7
machdep.cpu.tlb.data.small: 64
machdep.cpu.tlb.data.large: 32
machdep.cpu.tlb.shared: 512
machdep.cpu.address_bits.physical: 36
machdep.cpu.address_bits.virtual: 48
machdep.cpu.core_count: 4
machdep.cpu.thread_count: 8
machdep.cpu.tsc_ccc.numerator: 0
machdep.cpu.tsc_ccc.denominator: 0
```

maxgerhardt, 2022-04-29:
>    maxgerhardt/toolchain-pico@https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.3.3-a/x86_64-apple-darwin14.elf2uf2-2062372.220212.tar.gz

You have the wrong download link, elf2uf2 tools are downloaded when you need GCC.

maxgerhardt, 2022-04-29:
Try and use `https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.3.3-a/x86_64-apple-darwin14.arm-none-eabi-ed6d983.220212.tar.gz` as your download link. `rm -rf ~/.platformio/packages/toolchain-pico` if needed.

skybrian, 2022-04-29:
That fixes it, thanks!

maxgerhardt, 2022-04-29:
Great! Issue can be closed then?

skybrian, 2022-05-01:
Well, I was wondering if there was documentation to be improved. However, I can't remember where I picked up the bad platform_package setting from. Maybe I guessed?

Closing.

closed 2022-05-01 by skybrian

maxgerhardt, 2022-05-01:
Original docs https://arduino-pico.readthedocs.io/en/latest/platformio.html#current-state-of-development say

```ini
; note that download link for toolchain is specific for OS. see https://github.com/earlephilhower/pico-quick-toolchain/releases.
platform_packages =
    maxgerhardt/framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git
    maxgerhardt/toolchain-pico@https://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.3.1-a/x86_64-w64-mingw32.arm-none-eabi-7855b0c.210706.zip
```

Maybe I should add ot that that by "toolchain" we mean *GCC*, not any of the other packages.

skybrian, 2022-05-04:
I was having more conceptual difficulties; I didn't know which directory platformio was looking in for tools, where the missing file was supposed to come from, or where to find the source code to figure out how the platformio build process works. I just blindly copied the settings and modified them, and got it wrong somehow.

So, something like saying that platformio expects to find the compiler in ~/.platformio/packages/toolchain-pico/bin and it should be unpacked from the sources given in platform_packages would have been helpful for figuring out the screwup myself. Or, maybe a reference to somewhere to learn about the platformio internals? It's working well, but still mostly a black box to me.